visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #637: MQTT Home Assistant Temp shows occasional spikes

Zetto12, 2023-09-25:
**Describe the bug**
The Actual Temperature (Both target and current) looks fine but the history has strange anomalies where the temp is either pulled to 0 or to 100. Anyone know why this happens and how to fix it?

**To Reproduce**
Steps to reproduce the behavior:
Integrate Wifi controller into Home Assistant with MQTT

**Expected behavior**
The temperature should be displayed without the spikes.

**Screenshots**
[image]
[image]

**Hardware (please complete the following information):**
 - Pump model S100103
 - 6 wires
 - PCB type Custom (D1_mini)
 - Level shifter type. No level shifter used

**Software (please complete the following information):**
 - V4.2.1

visualapproach, 2023-09-27:
You have an "out of spec" device (No LLC) which I'm not fond of troubleshooting but I'll be nice since you used the bug template :-)
[start debug by clicking here](http://layzspa.local/debug-on/)
Let it run until the file `debug.txt` has grown to a few kB.
[then stop debug by clicking here](http://layzspa.local/debug-off/)

Now, post the log.txt file here

Zetto12, 2023-09-27:
> You have an "out of spec" device (No LLC) which I'm not fond of troubleshooting but I'll be nice since you used the bug template :-) [start debug by clicking here](http://layzspa.local/debug-on/) [...]

Thanks for helping out anyways. I've ordered LLC's and will put one in when they arrive.
Had it running for a few hours. Here is the content of Log.txt

1695840855:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840861:C0,1,FE,9F,FE,DF,FE,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840861:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840870:C0,1,FE,9F,FE,DF,FE,61,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840875:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840883:C0,1,FE,9F,FF,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840884:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840885:C0,0,FE,9F,FE,DF,FE,61,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840887:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840890:E0,1,FE,9F,FF,DF,FE,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840891:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840903:C0,1,FE,9F,FF,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840905:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840909:C0,1,FE,9F,FE,DF,FF,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840910:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840911:C0,1,FE,9F,FE,DF,FE,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840914:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840926:C0,1,FE,9F,FF,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840931:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840936:C0,1,FF,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840937:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840953:C0,0,FE,9F,FF,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840957:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840959:E0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695840963:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841006:C0,1,FF,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841006:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841043:C0,1,FF,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841049:C0,1,FE,9F,FE,DF,FE,41,FF,FF,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841051:C0,1,FF,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841053:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841060:C0,1,FE,9F,FE,DF,FE,60,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841061:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841061:E0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841063:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841076:C0,1,FE,9F,FF,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841077:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841100:C0,1,FE,9F,FE,CF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841100:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841105:C0,1,FE,8F,FE,CF,FE,61,FF,FF,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841107:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841115:C0,1,FE,9F,FE,DF,FE,1,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841119:C0,1,FE,9F,FE,DF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841198:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841220:C0,0,FF,8F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841221:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841273:C0,1,FE,9F,FE,FF,FE,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841275:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841300:C0,1,FE,DF,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841302:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841315:E0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841317:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841341:C0,0,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841342:C0,1,FE,9F,FE,FF,FF,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841344:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841350:C0,1,FF,9F,FE,FF,FE,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841353:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841373:C0,1,FE,9F,FE,FF,FF,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841375:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841385:C0,1,FE,9F,FE,FF,FF,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841386:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841394:C0,0,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841395:C0,1,FE,9F,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841424:C0,1,FE,DF,FE,FF,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695841426:C0,0,FE,9F,FE,FF,FF,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
[... 910 lines cut ...]
1695848777:C0,1,FE,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848805:C0,1,FE,9F,FF,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848806:C0,1,FE,9F,FE,F,FF,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848806:C0,1,FE,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848813:C0,1,FE,DF,FE,F,FE,1,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848814:C0,1,FE,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848825:C0,1,FE,9F,FF,F,FE,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848828:C0,1,FE,9F,FF,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848829:C0,1,FE,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848842:C0,1,FE,9F,FF,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
[... cut ...]
1695848914:C0,1,FF,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848914:C0,1,FE,9F,FE,F,FE,61,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848919:C0,1,FE,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848928:C0,1,FE,8F,FF,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848930:C0,1,FE,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848955:C0,1,FF,9F,FE,F,FE,40,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848955:C0,1,FE,9F,FE,F,FE,41,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0
1695848985:C0,1,FE,8F,FE,7,FE,61,FF,F7,FF	,1B,1B,0,0,0,0,0,0,0,0,0

closed 2023-11-23 by visualapproach