visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #640: [BUG] Energy Today value not correct in Home Asistant MQTT Entity

MattSpeke, 2023-10-01:
**Describe the bug**
A clear and concise description of what the bug is.

Energy Today Entity not correct in Home Assistant via MQTT. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Energy Today should be reset every day. It looks like the value does not get reset. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

[image]

**Hardware (please complete the following information):**
 - Pump model, for example 54123
 - 4 or 6 wires
 - PCB type
 - Level shifter type,

**Software (please complete the following information):**
 - Version

Via MQTT Mosquito Broker.

**Additional context**
Add any other context about the problem here.

visualapproach, 2023-10-01:
[image]

MattSpeke, 2023-10-01:
[image]

Thanks.

visualapproach, 2023-10-01:
You're welcome 😉

closed 2023-11-23 by visualapproach