visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #651: [BUG] Hawaii Hydrojet Pro 2021 with PCB V2B: display only makes chirp sound when power pressed, no lights, no reaction.

petekn, 2023-11-07:
First off all, great work with this project.  I think I read almost every thread on these github pages, and there has been an enormous amount of work and info here.

**Describe the bug**
I installed the PCB_V2B Board with exact config for my 2021 S200102 pump (Hawaii Hydrojet Pro 2021).  After installing wifi, I can access web interface but none of the buttons do something, they all turn back off when i try to toggle them.  
On the display there is no light, only thing when i press the power button is a chirp sound.

**Screenshots**
If applicable, add screenshots to help explain your problem.
https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/assets/115432187/604db6f2-ef7a-4c75-9567-b842d37e89f9

[image]

**Hardware (please complete the following information):**
 - Pump model, for example 54123 : S200102
 - 4 or 6 wires : 6
 - PCB type : PCB V2B
 - Level shifter type : Transistor based LLC 

**Software (please complete the following information):**
 - Version Firmware: 2023-08-19-2340

**Additional context**
HW test succeeded on all cables.

debug log when pressing random buttons:
1699385188:0,0,0,0,0,0,0,0,0,0,0	,FF,FF,0,0,0,0,0,0,0,0,0
1699386122:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386127:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386127:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386128:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386128:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386128:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386128:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386129:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386130:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386130:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386130:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386131:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386132:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386132:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386133:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386133:0,0,0,0,0,0,0,0,0,0,0	,2,0,0,0,0,0,0,0,0,0,0
1699386133:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386134:0,0,0,0,0,0,0,0,0,0,0	,8,9,0,0,0,0,0,0,0,0,0
1699386134:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386134:0,0,0,0,0,0,0,0,0,0,0	,13,12,0,0,0,0,0,0,0,0,0
1699386134:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386135:0,0,0,0,0,0,0,0,0,0,0	,11,12,0,0,0,0,0,0,0,0,0
1699386135:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386135:0,0,0,0,0,0,0,0,0,0,0	,12,12,0,0,0,0,0,0,0,0,0
1699386135:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386136:0,0,0,0,0,0,0,0,0,0,0	,10,12,0,0,0,0,0,0,0,0,0
1699386136:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386138:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386138:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0
1699386138:0,0,0,0,0,0,0,0,0,0,0	,0,0,0,0,0,0,0,0,0,0,0
1699386138:0,0,0,0,0,0,0,0,0,0,0	,1B,1B,0,0,0,0,0,0,0,0,0

Does anyone have any idea what might be wrong?

visualapproach, 2023-11-07:
Thanks! The ESP is not receiving any signal from the pump(CIO). Indicated by all the zeroes:
1699386138: ----0,0,0,0,0,0,0,0,0,0,0 ,-------1B,1B,0,0,0,0,0,0,0,0,0
   ^ _______________^ _________________________^
timestamp-------from CIO---------------from DSP

Reason? Don't know. Did you select PCB_V2B in the config page? Maybe filters are needed? Bent pins?

petekn, 2023-11-07:
Hi @visualapproach, thank you for your reply. 

- PCB_V2B is selected. 
- Pins are ok since the HW test is ok, and the display connected directly works i think?
- What kind of filters would be needed where?

visualapproach, 2023-11-07:
You're welcome. The (low pass) filters are between the A and B ports. But TBH I'm not convinced they will help. They might do but I their purpose is to suppress noise. And you have flat zeroes. Can you see what layzspa.local/inputs/ reveals? I think you need the dev branch version for that.

petekn, 2023-11-07:
indeed, file not found.
Is there an easy way i can switch to dev branch version?

visualapproach, 2023-11-07:
Well, no one click solution anymore. Download and build as you did with the master branch

petekn, 2023-11-07:
Well, i might sound like a noob (which i am for sure in this matter :) ), but when i try to upload build and upload firmware i get:
Retrieving maximum program size .pio/build/nodemcuv2/firmware.elf
Checking size .pio/build/nodemcuv2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  49.7% (used 40692 bytes from 81920 bytes)
Flash: [=====     ]  53.7% (used 560823 bytes from 1044464 bytes)
Building .pio/build/nodemcuv2/firmware.bin
/Users/peterweets/.platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/Users/peterweets/.platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
Creating BIN file ".pio/build/nodemcuv2/firmware.bin" using "/Users/peterweets/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and ".pio/build/nodemcuv2/firmware.elf"
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbserial-110
Uploading .pio/build/nodemcuv2/firmware.bin
esptool.py v3.0
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:f5:20:20:cd:60
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 564976 bytes to 380965...

A fatal error occurred: Invalid head of packet (0xE0)
*** [upload] Error 2

visualapproach, 2023-11-07:
I'm a noob on Macs. Try clean build (there is a button somewhere in platformio) and redo.

petekn, 2023-11-07:
I managed to fix it with my Windows pc :)

the inputs file gives me:
Edges received on pin D1: 2288
Edges received on pin D2: 1
Edges received on pin D5: 782
Edges received on pin D6: 1
Edges received on pin D4: 1
Edges received on pin D3: 1
Edges received on pin D7: 1
On 6-w pump the highest number is CLK, next is DATA and third is CS. On 4-wires the highest is CIO or DSP TX to ESP.

visualapproach, 2023-11-07:
Good. But poor result. Do the hwtestinfo again and follow the new instructions on that page

petekn, 2023-11-07:
Ok I did it, but after 10 seconds only the message to plugin the cables, and it seems the test is already finished before i get the message:
Connect the cables now!
Starting test in 10 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 9 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 8 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 7 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 6 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 5 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 4 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 3 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 2 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 1 seconds...
                                                                                                                                                                                                          [... cut ...]
-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0 // 50 errors out of 100
Sending on D3, receiving on D5
-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0 // 50 errors out of 100
End of test!
Errors indicated by 1 or 0 depending on test state. - is good.
Switching cio pins 5s HIGH -> 5s LOW -> input
then DSP pins 5s HIGH -> 5s LOW -> input, repeating
Disconnect cables then reset chip when done!

petekn, 2023-11-08:
Does this have to do with the fact i’m getting power over the regular usb connection?
If i need to power it seperately, the pcb_v2b says 3v? So not regular usb?

petekn, 2023-11-09:
Hi @visualapproach 

I managed to do the test now:

Connect the cables now!
Starting test in 10 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 9 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 8 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 7 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 6 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 5 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 4 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 3 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 2 seconds...
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
Connect the cables now!
Starting test in 1 seconds...
                                                                                                                                                                                                                                                                                                   [... cut ...]
Sending on D4, receiving on D2
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D3, receiving on D5
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
End of test!
Errors indicated by 1 or 0 depending on test state. - is good.
Switching cio pins 5s HIGH -> 5s LOW -> input
then DSP pins 5s HIGH -> 5s LOW -> input, repeating
Disconnect cables then reset chip when done!

petekn, 2023-11-15:
Hi @visualapproach, any feedback on the latest test? Is there something else i can do to try to get it working?
sorry for asking again, but just want to know if i should give up or not. :(
thanks!

visualapproach, 2023-11-16:
Sorry, I've been away for a while. So judging from your tests the clock signal (D2) is not getting from the pump to the esp. The esp and LLC appears to work so I'd say there is a bad connection somewhere between the LLC and the pump. Check with a multimeter (ohms) between D2 (solder point on red wire to cio) and if you can find an exposed part to the corresponding wire on the pump connector. Obviously you need to have the connector connected and power off.
It should read close to zero ohms.  You can also have everything connected and powered on and measure voltage between D2 and GND on the pcb. It should read closer to 5V. It's a square wave signal oscillating between 0 and 5V so the DMM will read something in between.
I don't think you should give up yet ;)

petekn, 2023-11-28:
Hi, sorry for the delay but I had some issues opening the pump while it's connected to an operating spa :-)
Anyways, the Ohms between the D2 solder point and connector is 0 (i couldn't find any exposed part closer to the pump).
When powered on the voltage between D2 and GND is indeed +/-4,81-4,84 somewhere.
So I think connection-wise it should be ok?

visualapproach, 2023-11-28:
Well it's peculiar. Unfortunately the voltage reading is not conclusive. There are pull up resistors on the inputs so they will read almost 5V even when disconnected. I made a test myself. It's virtually impossible to judge with a DMM. I can't tell what you need to do. Maybe resolder the cables, or switch some of the wires and run tests. Or if you have an oscilloscope or a logic analyzer to see what's going on. You want the red wire from the CIO to show bursts of ones and zeroes.

msggsm, 2023-12-06:
Yes, I have the same problem:
Had a pump P05504 type 204002002770 (Hawaii HydroJet Pro MY 2021, 6 wires) and my PCB V2B module with level shifter worked perfectly. Due to a suddenly not working heater the pump was replaced by LayZ, and I got the newer model
P05504 but type 302029000278 (seems to be the last one before the wifi integrated version).
Now my module doesn't communicate with pump or display anymore, tried all settings, tried older firm like 2022-12-20 and all newer like 2023-05-19 and further to the latest 2023-09-06-2103.
I do only get few information from board, different in older and newer firmware; pic 1 is with older firm connected to pump, pic 2 is with latest firmware since 2023; you can see slightly different values received.
With old firmware no function at all, with newer !sometimes! when I activate pump it starts but after aprox. 30-60 seconds pump starts for seconds and turns off again (no reaction in app).
I did even get a complete new module; but same behaviour.
What can I do or what information, logs etc. can I provide?
[image]
[image]

visualapproach, 2023-12-06:
@msggsm You could do the `/inputs` test, and `/hwtestinfo.html`. Maybe the new pump is not compatible?

petekn, 2023-12-06:
> Well it's peculiar. Unfortunately the voltage reading is not conclusive. There are pull up resistors on the inputs so they will read almost 5V even when disconnected. I made a test myself. It's [...]

I have asked a family member to check this with an oscilloscope, I will let you know once I know more. 

@msggsm, glad to hear i'm not alone :)

visualapproach, 2023-12-06:
> I have asked a family member to check this with an oscilloscope, I will let you know once I know more.

Cool, thanks for the info. Will be interesting

PatrickDeTomasi, 2023-12-15:
I am having the exact same symtpoms as the original post
but I have a S200103 for the Maldives model
I am using the D1 Mini and followed all instructions and setup as a 6 wire Air + Jet and with a custom pin configuration (as described by [this guide](https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control/) and as well as many posts in the discussion.
[image]

My solder joints seem fine, shortened cables, passed the hwtestinfo.html test. I've got 2 modules so I tried both with exactly the same result. I have also tried the latest development branch of the project.

I will update this bug if I find any solutions and am very interested in if any of you are able to solve your issue :)

petekn, 2023-12-19:
Ok @PatrickDeTomasi , will keep you updated if I know more.  Keep us updated as well :)
Great that I'm not the only one experiencing this...

petekn, 2024-02-19:
Hi Guys, just wanted to let you know that the oscilloscope appointment didn't happen yet.  I hope in 2 weeks to know more.... anyone else has news?

petekn, 2024-03-19:
Hi @visualapproach 
So i finally managed to do some testing with an oscilloscope. So...
The 3 signal cables, green, red and blue are all giving some 0's and 1s. 

[image]

This is red:

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/assets/115432187/141a8a06-4bbd-4a60-8045-7252d6f0b906

this is green:

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/assets/115432187/2ee58e7e-1d47-4549-b49d-dab0d94e2b73

this is blue:

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/assets/115432187/94098904-fc37-4e88-b6a6-68aba69a7d3c

So i did another test, I bypassed the red, green and blue cable connecting it directly from CIO to DSP without the PCB in between, and kept the black, white & yellow cable through the PCB.  The result of this is that the display works but without sound.  
As soon as i route 1 extra signal cable through the PCB again the display does weird things and nothing reacts.
I also tried switching almost every color combination of the 3  signal cables with the board in between but without result.

Just thinking out loud:
Could it be there is some test running at startup between CIO and DSP, that is not working yet because the PCB needs to startup first which is then failing other functionality?

Are there any other things i can test?

Thanks upfront @visualapproach

svenm75, 2024-03-23:
Hello to @petekn and @visualapproach

I’ve been struggling the last two weeks from version 3.5.1 up to 4.2.5  on the same issue. Already tried three different NodeMCU’s. It’s the same —> The Webinterface is working but no functional display or buttons. Even the Webinterface shows random values.

 But if you connect the NodeMCU with USB and start monitoring with the function in the code via VSCode the display fires up and everything starts to work.

Is there a lack of power? Or does anyone knows about this ?

visualapproach, 2024-03-25:
Sorry @petekn but I can't do much with those movies. Other than confirm some wires has activity at some time. And it's a bit too complicated to try to describe what to do. Preferably logging every wire from start up until a few minutes with a logic analyzer.

IIRC someone tried to start up with the display disconnected and connecting it later. Maybe this will work if @svenm75 's idea about power is true.

Did you try while monitoring as @svenm75 suggested?

svenm75, 2024-03-28:
Finally all the things are working. I'm now using the D1 Wemos mini with Softwareversion 2023-11-19-1330. Just followed the description and hints from https://antanaitis.lt/2022/09/07/wi-fi-on-your-lay-z-spa-make-it-easy-to-remote-control/
@petekn --> give it a try, perhaps your succesful. btw: my pump is a S100101

greets and happy easter!

rh2600, 2024-03-29:
@svenm75 I had only ever been using the instructions that you link to, and still have the chirp-only bug. The last time I tried was 24th Of December. What hints in the article do you think are most relevant?

svenm75, 2024-03-29:
@rh2600 As mentioned earlier I guess it has something to with the powersouce of the display. It is now connected directly to the 5 V powersource of the CIO (pump-connector). If you‘re using the D1 Wemos mini be aware of the different pin assignment.

closed 2024-06-06 by github-actions[bot]

pwilms, 2024-06-17:
Hey @svenm75 @rh2600,

did you find anything out? I've got the same issue. 

- Chirp on the On/Off Button. 
- HW-Test fine
- 5V Passthrough 
- Monitoring d1 while Connected -> nothing

When I disconnect the wifiremote and connect the display to the pump while power on the display works immediately.

PatrickDeTomasi, 2024-06-17:
Unfortunately did not reach a resolution 😔 
I am adamant to do so though. I was thinking of trying the original mentioned hardware (PCB V2 with an LLC) and see how that goes...
From things I've read, I got a feeling it's something to do with d1 mini signalling at 3.3v (though I've not tested myself) and that it wasn't enough, needs to be closer to 5v and that's where the LLC comes in.

Keep in mine I have very limited experience with hardware like this and don't know what I'm talking about

svenm75, 2024-06-22:
@pwilms @PatrickDeTomasi
As mentioned above i‘ve connected the display directly to the 5V powersource of the CIO. NO PASSTHROUGH over the Wemos D1 Mini. So this one worked for me.

pwilms, 2024-06-23:
> @pwilms @PatrickDeTomasi
> 
> As mentioned above i‘ve connected the display directly to the 5V powersource of the CIO. NO PASSTHROUGH over the Wemos D1 Mini. So this one worked for me. 

I've did that, yellow. And didn't work. (Ignore the colors ;-) ) 

Weird bug. 

[image]

petekn, 2026-11-05:
Hi @visualapproach , an update after a lot of new testing — I finally have logs that I hope are useful.

**Setup**
- Pump S200102 (Hawaii Hydrojet Pro 2021), 6 wire
- PCB_V2B with NodeMCU and 8-ch transistor LLC
- Firmware 2025-07-28-1200 (built from master, firmware + littlefs uploaded via OTA)
- Hardware config: CIO + DSP "6 wire, 2021 (air+jet)", PCB V2B, pins CIO 1/2/5, DSP 6/4/3/7

**Symptom (unchanged since 2023):** display dark, power button chirps, no control.
If I insert the module while the pump is running with the display on, the display keeps working until the ESP finishes booting — then it goes blank.

**`/hwtestinfo.html` loopback:** all six lines pass, 0 errors in both directions (attached). All cables checked/redone.

**Debug log** (`/debug-on/` → `log.txt`):
- DSP msg count 1577, quality 100% — button presses arrive and are forwarded (ESP-CIO shows e.g. `A,9`)
- **CIO msg count 0**, packet errors 0, CIO-ESP all zeros

**`/inputs/` recording with the pump connected** (attached), changes per GPIO over 1024 edges:
- GPIO14 / D5 (CIO CS): 259 changes ✅
- GPIO5 / D1 (CIO DATA): 764 changes ✅
- GPIO4 / D2 (CIO CLK): **0 changes — stuck HIGH the whole time** ❌

Same result in two separate recordings, before and after redoing all cables.

**So:** the board passes every loopback test and the DSP side works perfectly, but with the pump connected nothing ever arrives on the CIO clock pin, while CS and DATA from the pump do. Earlier, with an oscilloscope, I saw activity on all three signal wires coming from the pump. Could this pump put its clock on a different pin position than the board expects, or is there something else I should check?

Already tried: several firmware versions, DSP model settings, display powered directly from CIO 5V, series resistors on the DSP lines (removed again), swapping CIO signal wires, redoing all cables.

Thanks for any pointers!