visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #655: [BUG]

acolod, 2023-12-06:
**Describe the bug**
When on the firmware update page of the web UI, it shows the current firmware date, but Master and Beta, it just says reload page.  Reloading doesn't do anything, and clicking either the Master or Beta update button just brings me back to the main UI page (not updated).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Firmware page
2. See "reload page" next to Master and Beta
3. Click on "MASTER: Update data files, then firmware" or "BETA: Update data files, then firmware"
4. See that nothing is updated and redirected to layzspa.local

**Expected behavior**
I expect that the available updates show up next to Master and Beta for each, and clicking the update button for either would update the firmware.

**Screenshots**
[image]

**Hardware (please complete the following information):**
 - Helsinki - S100105
 - 6 wires, 2021 (air)
 - D1mini

**Software (please complete the following information):**
 - 2023-09-06-2103

msggsm, 2023-12-06:
Yes, confirmed.
But I have this issue since at least 2 months,
tried it every week since it occured!

visualapproach, 2023-12-06:
No worries, they are not overly exposed. It's a pity we can't have that feature but there just isn't enough memory to do it.

acolod, 2023-12-06:
> No worries, they are not overly exposed. It's a pity we can't have that feature but there just isn't enough memory to do it.

Thanks for the response, understood - I remember this being discussed in another thread but didn't realize it was removed.  I will update the old fashioned way!