visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #659: [OTHER] License of the project

GSzabados, 2023-12-25:
**Hardware (please complete the following information):**
 - Pump model, for example 54123
 - 4 or 6 wires
 - PCB type
 - Level shifter type,

**Software (please complete the following information):**
 - Version

@visualapproach, what license does your project follows?

Somebody posted this on the HA community forum recently:

https://community.home-assistant.io/t/lay-z-spa-hot-tub-wi-fi-pump-automation/229006/266

Including a link to a product as well:

https://www.kleinanzeigen.de/s-anzeige/wifiwhirl-modul-fuer-bestway-lay-z-spa-wlan-wifi-platine/2549532529-87-1311

There is more on the German interwebs under the name WiFiWhirl from multiple different sellers.

https://www.ebay.de/itm/186155088694?mkrid=707-53477-19255-0&siteid=77&mkcid=1&campid=5337690175&toolid=10001&mkevt=1

visualapproach, 2023-12-25:
Hi @GSzabados Thanks for your concern.
It's GPL-3.0, you can read about it on github but it's kind of wordy. Basically it says no one is allowed to make this not open source and there is rules about modifying it and so on.
Happy holidays

GSzabados, 2023-12-25:
Happy Holidays to You as well.

My concern is really about changing the name to WiFiWhirl and not mentioning anywhere the source of the project. The images of the device and webpage clearly looks like this project.

visualapproach, 2023-12-25:
Thank you! 
There is this text:
_Bitte beachte, dass das WLAN-Modul auf dem Open-Source-Projekt von Visual Approach "Wifi-remote-for-Bestway-Lay-Z-SPA" basiert und kein Originalteil von Bestway ist. Die Platine ist handgelötet._
Maybe added after your comment. A link and description of the changes would be fair but at least he isn't hiding the origin.

closed 2024-02-24 by GSzabados

GSzabados, 2025-03-21:
@visualapproach, you might want to have a look at this project as well. Advertising itself on the Home Assistant community forum.

https://www.mk-innovations.de/

Nice and ready made, but re-branded with obvious similarities, and it is using the same project as the site links to a topic on iobroker, which is about your code and device.

And has this written in only one of the pages:

> Hinweis:
> Es handelt sich nicht um ein Originalteil der Marke Bestway®. Das WLAN-Modul basiert auf dem Open-Source Projekt [WiFi-remote-for-Bestway-Lay-Z-SPA](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA) von [visualapproach](https://github.com/visualapproach) ([Thomas Landahl](http://visualapproach.se/)).
> 

I just want to note for you, because as per the EU's CRA, manufacturers monetising open source code, has to contribute into the code. 😄 

https://github.blog/open-source/maintainers/what-the-eus-new-software-legislation-means-for-developers/

visualapproach, 2025-03-21:
Thank you! Yes I have also seen that page. You'd expect people to ask before doing this, but I'm not really surprised.

GSzabados, 2025-03-21:
I ended up on the page, because he self advertised himself on the HA Community Forum. It looks nice, but why cannot show that it is not his code. I am even concerned that the casing is his own work or not...