visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #661: visual studio build error

huzsiroc, 2024-01-11:
i can't translate it with the visual studio please share the firmware file separately which i can flash to the esp module if possible thank you very much 
[image]

undone37, 2024-01-11:
Hi huzsiroc,

this is a problem in plerup/EspSoftwareSerial (see https://github.com/plerup/espsoftwareserial/issues/307) and can be fixed by adding the `dok-net/ghostl` as dependency in platformio.ini:

```ini
;[...]
[env:nodemcuv2]
platform = espressif8266@^4
board = nodemcuv2
framework = arduino
lib_deps = 
	bblanchon/ArduinoJson
	; mcxiaoke/ESPDateTime
	links2004/WebSockets
	knolleary/PubSubClient
	me-no-dev/ESPAsyncTCP
	; khoih-prog/ESP_WifiManager@^1.12.1
        tzapu/WifiManager
	plerup/EspSoftwareSerial
	milesburton/DallasTemperature
        https://github.com/dok-net/ghostl ; << Fix missing circular_queue.h
board_build.filesystem = littlefs
extra_scripts = gzip_littlefs.py
;[...]
```

Hope this helps :)

huzsiroc, 2024-01-11:
thank you very much for your help!!! So it is really solved!!! 
[image]

closed 2024-01-11 by huzsiroc

visualapproach, 2024-01-13:
Thanks @huzsiroc , the .ini-file is updated with this and I locked ArduinoJSON to v6 since v7 gave a lot of deprecation warnings that I don't want to hassle with now.

huzsiroc, 2024-01-13:
With great respect, thank you very much!