visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #670: ideas

starsx2016, 2024-01-31:
Hi!

any plan for a DARK theme mode for the interface?

thanks for all your hard work

Matt

**Hardware (please complete the following information):**
 - Pump model, for example 54123
 - 4 or 6 wires
 - PCB type
 - Level shifter type,

**Software (please complete the following information):**
 - Version

**Your message**

visualapproach, 2024-02-04:
Yes, it's pulled to the dev branch from @colinpeddle

colinpeddle, 2024-02-04:
Yeah it’s a very rudimentary implementation but it does work for saving your eyes :-). I’ve been testing a correction for the occasional flashing as the page loads (especially if the connection is slow or your device is slow), along with several other visual advancements for the app, which hopefully I’ll be able to do a pull for soon!