visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #677: Wifi connection issues

xraitekx, 2024-03-05:
**Hardware**
 - Pump model: S100101
 - 6 wires
 - PCB type: Selfmade (Based on V2B schematics)
 - Level shifter type: TXS0108E (with 10k resistor - without filter)

**Software**
 - Version: Current - d3cef1b - merge #674 

Hello!
I have a problem with the initial wifi connection.
I tried to connect to two different wifi´s and always get an ERROR as shown in the serial log output below.
Installation on another identical NodeMCU v3 doesn´t work either.
Do you have any idea what the problem might be? Thanks a lot!

**Serial Log - Fresh installation (FS Image + Code) with first connection try on ssid "lc":**

>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3424, room 16
> tail 0
> chksum 0x2e
> load 0x3fff20b8, len 40, room 8
> tail 0
> chksum 0x2b
> csum 0x2b
> v0008a540
> ~ld
> rf cal sector: 1020
> freq trace enable 0
> rf[112] : 0�
> Start
> WiFi > using WiFi configuration with SSID ""
> WiFi > Trying to connect ...WiFi > Using WiFiManager Config Portal
> *WM:
> *WM: AutoConnect
> *WM: Connecting as wifi client...
> *WM: Status:
> *WM: 0
> *WM: No saved credentials
> *WM: Connection result: 
> *WM: 0
> *WM:
> *WM: Configuring access point... 
> *WM: Lay-Z-Spa Module
> *WM: layzspam0dule
> *WM: AP IP address:     
> *WM: 192.168.4.1        
> *WM: HTTP server started
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Handle root
> *WM: Scan done      
> *WM: DUP AP: lc
> *WM: DUP AP: lc-gast
> *WM: lc
> *WM: -65
> *WM: lc-gast        
> *WM: -65
> *WM: Sent config page
> *WM: Request redirected to captive portal
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Handle root
> *WM: Scan done
> *WM: DUP AP: lc
> *WM: DUP AP: lc-gast
> *WM: lc
> *WM: -66
> *WM: lc-gast
> *WM: -67
> *WM: Sent config page
> *WM: WiFi save
> *WM: Sent wifi save page
> *WM: Connecting to new AP
> *WM: Connecting as wifi client...
> *WM: Status:
> *WM: 0
> *WM: [ERROR] WiFi.begin res:
> *WM: 7
> disconnected
> disconnected
> disconnected
> disconnected
> *WM: Connection result:
> *WM: 4
> *WM: Failed to connect.
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected
> disconnected

**Manual reset**

>  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> 
> load 0x4010f000, len 3424, room 16
> tail 0
> chksum 0x2e
> load 0x3fff20b8, len 40, room 8
> tail 0
> chksum 0x2b
> csum 0x2b
> v0008a540
> ~ld
> rf cal sector: 1020
> freq trace enable 0
> rf[112] : 0�
> Start
> WiFi > using WiFi configuration with SSID ""
> WiFi > Trying to connect ...WiFi > Using WiFiManager Config Portal
> *WM:
> *WM: AutoConnect
> *WM: Connecting as wifi client...
> *WM: Status:
> *WM: 0
> *WM: Using last saved values, should be faster
> *WM: Connection result: 
> *WM: 255
> disconnected
> *WM: 
> *WM: Configuring access point...
> *WM: Lay-Z-Spa Module
> *WM: layzspam0dule
> *WM: AP IP address: 
> *WM: 192.168.4.1
> *WM: HTTP server started

visualapproach, 2024-03-07:
Don't know, but it says `WiFi > using WiFi configuration with SSID ""` when it should say "lc" or "lc-gast". Did you enter and save the credentials properly in the config portal?

xraitekx, 2024-03-15:
Thanks for your answer!

Yes i saved the credentials (See in video). I tried it also with an different SSID.
While saving the credentials the SW is running in an ERROR:
***WM: [ERROR] WiFi.begin res:
*WM: 7**

Can that cause the issue, that it is not saving the credentials?

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/assets/25708606/9038d0d0-27a1-4e4a-b91b-2d19e3de41f0

Log during configuration of wifi (again with fresh installation)

> WM: Configuring access point...
> *WM: Lay-Z-Spa Module
> *WM: layzspam0dule
> *WM: AP IP address: 
> *WM: 192.168.4.1
> *WM: HTTP server started
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Scan done
> *WM: DUP AP: lc-gast
> *WM: DUP AP: lancom
> *WM: lc-gast
> *WM: -65
> *WM: lancom
> *WM: -67
> *WM: Sent config page
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Scan done
> *WM: DUP AP: lc-gast
> *WM: lancom
> *WM: -65
> *WM: lc-gast
> *WM: -65
> *WM: Sent config page
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: WiFi save
> *WM: Sent wifi save page
> *WM: Connecting to new AP
> *WM: Connecting as wifi client...
> *WM: Status:
> *WM: 0
> *WM: [ERROR] WiFi.begin res:
> *WM: 7
> disconnected
> disconnected
> disconnected
> disconnected
> *WM: Connection result:
> *WM: 4
> *WM: Failed to connect.
> *WM: Request redirected to captive portal
> *WM: Request redirected to captive portal
> *WM: Handle root
> *WM: Handle root

visualapproach, 2024-03-15:
It seems the WiFi Manager is throwing the error. It's a third party library, not written by me so I can't tell right now what it means. I can dig into it when I have time. I'm out of country at the moment.

xraitekx, 2024-03-15:
Have a nice trip!

Btw.: I discovered, that maybe the FS Image is not installed correctly on the ESP.

There is no favicon, as well the /dir or /info is not loading.
Just the basic /wifi, /0wifi, /i and /r are loading.

Maybe it can not save the credentials, because the wifi.json is also not available?

[image]

[image]

However, the build and upload should be fine?

- The Image will be built and created correctly:

> [...]
> Building file system image from 'datazip' directory to .pio\build\nodemcuv2\littlefs.bin
> /calib.html.gz
> /chkupdatefw.html.gz
> /config.html.gz
> /darkmode.js.gz
> /favicon.ico.gz
> /favicon.png
> /filelist.txt
> /function.js.gz
> /furelise.mel
> /hwconfig.html.gz
> /hwtest1.jpg
> /hwtest2.jpg
> /hwtestinfo.html.gz
> /index.html.gz
> /index.js.gz
> /main.css.gz
> /manifest.json
> /menu.png
> /mqtt.html.gz
> /popcorn.mel
> /remove.html.gz
> /success.html.gz
> /upload.html.gz
> /visualapproach.png
> webconfig.html.gz
> /weekender
> /wifi.html.gz
> del_gzip_data([".pio\build\nodemcuv2\littlefs.bin"], ["datazip"])
> Clearing zipped files
> [...]

- The upload seems to be correct as well
> [...]
> /weekender
> /wifi.html.gz
> del_gzip_data([".pio\build\nodemcuv2\littlefs.bin"], ["datazip"])
> Clearing zipped files
> Looking for upload port...
> Auto-detected: COM4
> Uploading .pio\build\nodemcuv2\littlefs.bin
> esptool.py v3.0
> Serial port COM4
> Connecting....
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Configuring flash size...
> Compressed 1024000 bytes to 57430...
> Writing at 0x00300000... (25 %)
> Writing at 0x00304000... (50 %)
> Writing at 0x00308000... (75 %)
> Writing at 0x0030c000... (100 %)
> Wrote 1024000 bytes (57430 compressed) at 0x00300000 in 5.4 seconds (effective 1516.3 kbit/s)...
> Hash of data verified.

visualapproach, 2024-03-15:
Thanks :) 
I think it's '/dir/' with a trailing slash.

xraitekx, 2024-03-15:
Also with both slashes "File not found" :(

visualapproach, 2024-03-17:
I  made a fresh upload to a module and got this:
[image]
So I guess the code is okay.
Are you sure you switched WiFi back to your home network after credentials (to that network) was saved? The /0wifi and such you were referring to are addresses only valid when the WM AP is running.

visualapproach, 2024-03-17:
And these are the code definitions:
```
const char * const WIFI_STA_STATUS[] PROGMEM
{
  "WL_IDLE_STATUS",     // 0 STATION_IDLE
  "WL_NO_SSID_AVAIL",   // 1 STATION_NO_AP_FOUND
  "WL_SCAN_COMPLETED",  // 2
  "WL_CONNECTED",       // 3 STATION_GOT_IP
  "WL_CONNECT_FAILED",  // 4 STATION_CONNECT_FAIL, STATION_WRONG_PASSWORD(NI)
  "WL_CONNECTION_LOST", // 5
  "WL_DISCONNECTED",    // 6
  "WL_STATION_WRONG_PASSWORD" // 7 KLUDGE
};
```

closed 2024-05-24 by github-actions[bot]