visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #679: Software installation fails

mr-mmc, 2024-03-17:
- Pump model  54123
 - 4 wires
 - PCB type  V2B
 - Level shifter type txs0108e

**Software (please complete the following information):**
 - Version 4.0?? Sorry newbie…
 
 Cannot connect esp

first time going through, but after minute make some error 🤷‍♂️
Second try and…
[image]

visualapproach, 2024-03-17:
Tried on a pc?

mr-mmc, 2024-03-17:
> Tried on a pc? (Windows)

Yes, (win 11) devices not found, I think something gone broken 😔

visualapproach, 2024-03-17:
Some modules require a reset while holding the flash button pressed, but may be something else. Try disconnecting everything and reboot pc. Also some modules have a counterfeit uart chip that doesn't work with the driver.

visualapproach, 2024-03-17:
Just tried a fresh install on one of my modules and it works fine.

simondean, 2024-04-01:
Lowering the upload speed fixed a similar issue for me on my Mac.  I changed this line in `platformio.ini` from this:

```
upload_speed = 921600
```

to this:

```
upload_speed = 460800
```

mr-mmc, 2024-04-06:
new esp module arrived and, works fine, something broken ch340 chip.

closed 2024-06-13 by github-actions[bot]