visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #683: [FEATURE RQ] Easier integration to external sensors. IE Temp and Power

PaleFlyer, 2024-03-19:
I just got this up and running a couple days ago, and was looking into the ambient temp sensor input from Home Assistant, as I have a Tempest weather station on my fence, ~200 ft from the hot tub, but with HA trying to be "more" user friendly now, I'm struggling to tie the sensor into the Spa controller. The main issue I ran into is HomeAssistant in the GUI editor didn't show a device for the Spa controller to import the temp. I was looking for something like the sample YAML had, but found nothing similar.

In fact, the Spa controller seems to read more like a thermostat than anything else, considering I have an "HVAC" mode option as one of the possible commands. The only "set value" I see is the display brightness. I am assuming the issue is that HA doesn't want to have a "sensor" be both a sensor, and a setpoint?

HA is up to date, running as the "Supervisor" install (I don't remember exactly what version anymore), and running virtualized on an x86 box.

Also, on the power usage, it's neat that you are able to guesstimate it, but I'd rather use the power monitor I have using actual clamps in my breaker box, and I'm not seeing ANY way to import that data into the Spa controller (even if I were to move from circuit level to the actual device.)

For anyone trying to do the same,
alias: Lay-z-spa Ambient Temp
description: Sends the current Ambient Temp from the Tempest to the Laz-z-spa controller
trigger:
  - platform: state
    entity_id:
      - sensor.tempest_temperature
condition: []
action:
  - service: mqtt.publish
    metadata: {}
    data:
      topic: layzspa/command
      payload_template: |-
        {"CMD":14,"VALUE":{{ states("sensor.tempest_temperature")|round
         }},"XTIME":0,"INTERVAL":0}
mode: single

If you scroll down in the "Action/Then Do", and hit "Other Action", you can then select MQTT. I had to dig out the exact "topic" from the MQTT data to get it to work. The device originally registered as "layzspa WiFi Controller", and I shortened it to layzspa, as it was too long for LoveLace to let me read all the entries.

visualapproach, 2024-03-19:
Yes, it's not super easy to set the ambient temp. I guess it should be possible to auto-add a number entity or something. I'll have a look at it.
Regarding the name length - I agree. There was a change in HA some time ago where they started adding device name as a prefix. Not sure how to motigate that but I may have a look at that too.

PaleFlyer, 2024-03-19:
I wasn't trying to imply the name length was totally on you, I do know HA did something, and now the entity names are stupid long.  They also have the size of the drop down capped so you can't really deal with it.

visualapproach, 2024-03-22:
Hey @PaleFlyer 
I have now edited the HA autodiscovery file. 1500 lines, yuck. But I'm happy with the result. Shorter entity names, and the ambient temp (C and F) are now a number entity that you can set, and it will send the data to the ESP. Thanks for your FR!
I will upload shortly to the dev branch.
[image]

visualapproach, 2024-03-22:
BTW, best if you remove the old device from MQTT in HA:
[image]

PaleFlyer, 2024-03-25:
Can we push this, and other changes, over the air? I know the "firmware update" button in the project doesn't work because of a GIT-ESP issue, but just like on the local network, so I don't have to open the pump/put a USB extension through the pump body and drag a laptop out to the side of the hot tub?

I was looking into ESP OTA programming, and it said it needed some library I don't see included, but then I do see OTA credentials in the main programs.

closed 2024-04-06 by visualapproach