visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #690: 54123 : shows 04 and no connection.

mr-mmc, 2024-04-06:
**Hardware V3:**
 - Pump model  54123
 - 4  wires
 - PCB type V2B
 - Level shifter type red LLC

**Software (please complete the following information):**
 - Fw Version 2023-11-19-1330

sorry I'm newbie ...
DSL shows only 04 , no alive , goes on and off but no button effect
-i change txs0108e to red LLC, no effect.
-try set D1 / 2 /4 /6 settings, if use v2b setting cannot change right pin tx of rx.
-cio an dsp selected  4wire, 54123

[image]
as this show, test ok?

[image]

maybe change wires H1A and H2A ? and places where hwconfig says 2 5 4 and 3

visualapproach, 2024-04-06:
Yes, the wires should go in the white slots, or the equivalent on the B side. So move red+green one step down.

visualapproach, 2024-04-06:
It may not be the only problem since you had the hardware test okay, but best chance anyway.

mr-mmc, 2024-04-06:
i don't know what I do but now almost everything works... XD. ..only showing 04 dls
but can start and set temp...

I try change wires, maybe that is problem

mr-mmc, 2024-04-06:
yes! now its working! Thanks!

dasmith83, 2024-05-20:
> yes! now its working! Thanks!

What was the final fix for you?

I’ve also got a 4 wire model - and everything works fine except for 2 things:

1. DSP water temp shows 04 - but if I increase or decrease the target temp the numbers change as expected but then reverts back to 04
2. It’s like the egg doesn’t know what temp the water is flowing at. When I enable the heater it heats and just keeps on going way past how the target temp would feel to touch. If I remove the WiFi module the dsp works fine again. 

Interestingly I previously used a similar module sold by Andy who makes and sells “hot tub manager” - it’s basically the same thing but his own code and pcb with a d1 mini. That worked great for a year or two and end of last season I started having this exact problem where it couldn’t read the egg water temp. 

Why would these WiFi modules prevent the egg reading temp properly?

visualapproach, 2024-05-22:
> Why would these WiFi modules prevent the egg reading temp properly?

Temp is sent from the pump via the ESP module to the display. So if the ESP misreads the temp it will pass it on to the display. The ESP does not write anything on the display. It just sends some data. The display chooses what to show, for instance E04.
Does the display show 4 degrees when TAKE CTRL is OFF?

closed 2024-07-29 by github-actions[bot]