visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #692: Web login not loading correctly

danc89, 2024-04-17:
Any help with this one guys? I had it all working then changed the pin assignment and this happened.

[image]

dietmar-1, 2024-04-19:
I have the same problem, after saving the wifi settings and restarting the esp, I got the following output in the serial monitor:
[image]

The (web-)files are not loading correct.

Is Maybe this error the reason for it ?:
[image]

VSC an PlatformIO have are updatet.

dietmar-1, 2024-04-19:
@danc89 : Yes, you are right, you can certainly reproduce the error if you change the  pin assignment.

ZoXx, 2024-04-27:
unfortunately, i seem to have the same problem.

I have flashed the latest version. If the ESP is connected via USB, everything works perfectly. If the ESP is installed in the pump, I get the same picture or the one in the attachment.

What can I do to make it work again?

[image]

[image]

visualapproach, 2024-04-28:
If it worked with an earlier version you can always downgrade again. You can choose version on github.
It seems to be a power issue. The power consumption may drop a little if you change cpu speed to 80 MHz by commenting out the relevant line in platformio.ini.

ZoXx, 2024-04-28:
but is it with all new versions like this?

inrepublica, 2024-04-29:
I have the same issue:
`
WiFi > Using WiFiManager Config Portal
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: No saved credentials
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: Lay-Z-Spa Module
*WM: layzspam0dule
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Scan done
*WM: ****MASKED****
*WM: -63
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: [ERROR] WiFi.begin res:
*WM: 7
disconnected
got IP: 192.168.1.105
*WM: Connection result: 
*WM: 3
*WM: freeing allocated params!
start NTP
............
WS IRamheap 10912
IRamheap 9664
startmqtt
Failed to read mqtt.json. Using defaults.
192.168.1.105
End of setup()
31512
start NTP
............
File size: 2865
HTTP > file sent: /index.html.gz (1608 bytes)
********* File not completed *******
File size: 3088
HTTP > file sent: /main.css.gz (1072 bytes)
********* File not completed *******
File size: 3455
HTTP > file sent: /index.js.gz (950 bytes)
********* File not completed *******
File size: 614
HTTP > file sent: /function.js.gz (614 bytes)
File size: 2425
HTTP > file sent: /favicon.png (1072 bytes)
********* File not completed *******
File size: 2425
HTTP > file sent: /favicon.png (986 bytes)
********* File not completed *******
File size: 2953
HTTP > file sent: /visualapproach.png (2027 bytes)
********* File not completed *******
`
I have the same problem, after saving the wifi settings and restarting the esp....

inrepublica, 2024-04-29:
I try to set CPU speed to 80Mhz in platformio.ini:
`board_build.f_cpu = 80000000L`
But the reseult is the same:
```
got IP: 192.168.1.105
start NTP
............WS IRamheap 10888
IRamheap 9640
startmqtt
Failed to read mqtt.json. Using defaults.
192.168.1.105
End of setup()
36744
File size: 2865
HTTP > file sent: /index.html.gz (1608 bytes)
********* File not completed *******
File size: 3088
HTTP > file sent: /main.css.gz (1072 bytes)
********* File not completed *******
File size: 614
HTTP > file sent: /function.js.gz (614 bytes)
File size: 3455
HTTP > file sent: /index.js.gz (1072 bytes)
********* File not completed *******
File size: 2953
HTTP > file sent: /visualapproach.png (955 bytes)
********* File not completed *******
File size: 2425
HTTP > file sent: /favicon.png (986 bytes)
********* File not completed *******
File size: 2425
HTTP > file sent: /favicon.png (2425 bytes)
```

ZoXx, 2024-04-29:
I have just flashed 4.2.4. Everything works without problems and immediately. It definitely seems to be due to 4.2.5.

inrepublica, 2024-04-30:
Just flashed with 4.2.4 too and it's ok for me!

PaleFlyer, 2024-05-01:
I'm running 4.2.5, and have no issues. I do have the "filter" set up if that matters. (IE the caps/resistors, and the non-egg pump.)

kecksk0enig, 2024-05-14:
i installed my v1 modul yesterday and everything worked fine (website /homeassistant/mqtt integration) however today the webfrontend wont load (or load only partially, control via homeassistant works)

visualapproach, 2024-05-22:
It may be out of memory problems or wifi problems. @kecksk0enig 4 or 6 wire?

kecksk0enig, 2024-05-22:
6 wire, wifi is strong how do i look for OOms?

visualapproach, 2024-05-22:
It's difficult. You can turn on OOM debug in the .ini file but it will take more memory...
Anyhow, maybe it's the flash drive that's corrupted. It's hosting the html files and the /info is generated by code IIRC. Try uploading a new filesystem image and see if that helps

visualapproach, 2024-06-06:
Try latest dev branch. It should load the pages better.

fbeek, 2024-06-24:
Hello,

i flashed the latest release after i good the same error on 4.2.5 and get the same error, tried 3 separate ESPs, check power rail with oscilloscope but there everything looks normal. After initial upload all looks fine, but when i change some setting and reboot the error is back immediately until I reflash the file system image.

visualapproach, 2024-06-24:
V4.3.1 is the latest. Have you tried it? Don't forget to upload the file system also.

fbeek, 2024-06-30:
Yes, sorry i meant 4.3.1

visualapproach, 2024-06-30:
This may sound weird but does it help if you in platformio.ini uncomment the line that says DDEBUG_ESP_PORT=Serial and install firmware again?

Another problem can be the WiFi channel. Some ppl says that ch8 is better on some routers

fbeek, 2024-06-30:
Wifi Channel 8 is already the used channel, i have Unifi Wifi Setup here and the RSSI is very good. A few minutes ago i tried another ESP from another source. This one works perfect, I can change the settings without any problem an after a reboot it connects flawlessly. BUT sometime i can see the above mentioned "File not complete" message for the index.html.gz. But not always. The file in the flash is ok . I think the first 3 from vendor A have some fault with the flash chips or some other issues so we can ignore these, and the error has nothing todo with your software. But the error in the fourth one can have some other source. Eventually a buffer overflow or something?  Because only mqtt is important for me I think we can close this issue, as the "File not complete" error goes away after a reload.

visualapproach, 2024-07-01:
Thx, I suspect it's a bug or something in the server library. I've been googling a lot on this topic and it seems to be an issue for others too, outside of this project.

visualapproach, 2024-07-08:
v4.4.0 should be better!

closed 2024-07-23 by visualapproach