visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #694: Option to display current time scrolling on display.

davebuk, 2024-04-19:
Could there be an option to scroll the current time on the display using the display TEXT option? 

I have a D1 mini running and being controlled with openHAB/MQTT and thought I may be able to send once per minute, or once per 5 minutes, the current time as text to the controller via MQTT. Would this work?

The other option would be to be able to set one of the built in commands running on the 'node MCU/D1 mini' to display the current time on a per minute basis. That way its running locally rather than needing the MQTT connection.

visualapproach, 2024-05-01:
Yes, sending the time as text in an MQTT message should work.
`{CMD: 19, VALUE: 1, XTIME: 1714589520, INTERVAL: 0, TXT: 'test'}`

davebuk, 2024-05-20:
I've been busy with other things but will give it a go in the next few weeks. Thanks.

davebuk, 2024-05-23:
I've had a play around but can't get it to display any text. Out of interest, when viewing the webpage for the spa module, would the digits in RED in RED brackets show the text if it was being processed?

From openHAB, I have a rule that if the temp selected is 30, send the command. Using another mqtt client I can see the following is published by openHAB to the broker:
`{"CMD":19,"VALUE":0,"XTIME":0,"INTERVAL":0,"TXT":"20.22"}`
20.22 was the time the code was sent. Its a `STRING` from the current time.

The other command `{"CMD":0,"VALUE":"31","XTIME":0,"INTERVAL":0}` that goes and sets the required temp works fine.

Even sending `{CMD: 19, VALUE: 1, XTIME: 0, INTERVAL: 0, TXT: 'test'}` from an mqtt client doesn't seem to work.

visualapproach, 2024-05-23:
I should've mentioned that text only works for the 6-wire displays. Do you have that?

davebuk, 2024-05-23:
Hi, yes. I've have a D1mini setup as 6 wire,2021 air+jet. Its a square pump LayZspa Hawaii.

visualapproach, 2024-05-23:
Ok I'll do some tests. If you send text from the web ui (config) it works?

davebuk, 2024-05-23:
Should it show in the RED brackets on the webUI status page?

davebuk, 2024-05-23:
It works on the pump by setting a test message to print from the webUI every 5 seconds.

visualapproach, 2024-05-23:
No. At least, it doesn't :)

visualapproach, 2024-05-23:
I tried to send this string `{CMD: 19, VALUE: 1, XTIME: 1716493800, INTERVAL: 0, TXT: 'test'}` to layzspa/command and it worked. Text scrolled by on the display

visualapproach, 2024-05-23:
I'm on Firmware: 2024-05-09-1955 but it's not changed since Febuary 2023.

davebuk, 2024-05-23:
Yep, that works. I'll have another play around. Is there a difference between `XTIME: 1716493800` and `XTIME: 0`

davebuk, 2024-05-23:
Firmware: 2023-11-19-1330.

visualapproach, 2024-05-23:
In practice - no. It's when to execute the command, counting seconds from 1971. So a '0' makes the ESP think it's 53 years behind and do it immediately.  1716493800 was the present time a few minutes ago.

davebuk, 2024-05-23:
Works ;-)
openHAB sent `{"CMD":19,"VALUE":0,"XTIME":1716493800,"INTERVAL":0,"TXT":"21.11"}` and it scrolled `21 11`.

visualapproach, 2024-05-23:
Nice! You can send text with every command if you don't want to make an extra call :)

davebuk, 2024-05-23:
Then I'd have to work out how to send two variables via openHAB to mqtt :-)

This seems to work really. I just turned the temp down to 30C on the pump and then it displayed the current time. Good work. I can add some additional openHAB examples if you like to the write up?

closed 2024-05-23 by davebuk

visualapproach, 2024-05-23:
Yeah why not! I can link to it or copy to the manual.