visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #697: [BUG] Unable to upload with newer NodeMCU

metaljay, 2024-04-26:
Ive recently bought a newer set of NodeMCUs, when I try to flash them I get error 2.

>CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbserial-210
Uploading .pio/build/nodemcuv2/firmware.bin
esptool.py v3.0
Serial port  **/dev/cu.usbserial-210**
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 08:f9:e0:75:8e:e5
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 566576 bytes to 381668...
A fatal error occurred: Invalid head of packet (0xE0)
*** [upload] Error 2
============================================================================ [FAILED] Took 2.68 seconds ============================================================================
`

If I compare it to my older boards I get this (this proceeds to install correctly)

> CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbserial-FTB6SPL3
Uploading .pio/build/nodemcuv2/firmware.bin
esptool.py v3.0
Serial port **/dev/cu.usbserial-FTB6SPL3**
Connecting........_
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 60:01:94:3c:fc:57
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 566576 bytes to 381668...
Writing at 0x00000000... (4 %)

Any ideas why the boards are being flagged via a different serial port? (see bold) - or can I add compatibility somehow?
I am running on a Mac with a micro USB

visualapproach, 2024-04-26:
Hi @metaljay and thank you for your kindness! Can you still upload to old boards? I'm thinking it's either some update in platformio, or it's something wrong with the new boards. If it's platformio you should have the same problem regardless of the board. If it's the board then maybe it's one of those fake cp21somethingsomething UART chips. Then you need to hassle with special drivers (pc experience)

metaljay, 2024-04-26:
The old boards do upload as normal, certainly feels like a knockoff ESP, I’ll reach out to the seller and ask for info on drivers. Have you come across anything I can read up on?

visualapproach, 2024-04-26:
Google cp2102 vs ch340
Try baudrate 115200

visualapproach, 2024-04-26:
Or just comment out the Andrade line in platformio.ini

metaljay, 2024-04-27:
if I comment out the 'upload speed' I can get it to flash (albeit slower) - but it works successfully

On future releases is it possible to account for both chips? What I mean is, I didn't specify CP2102, I just ordered a ESP8266, so I imagine in the future others may have the same error

> [env:nodemcuv2]
platform = espressif8266@^4
board = nodemcuv2
framework = arduino
lib_deps = 
	bblanchon/ArduinoJson@6.21.2
	; mcxiaoke/ESPDateTime
	links2004/WebSockets
	knolleary/PubSubClient
	me-no-dev/ESPAsyncTCP
	; khoih-prog/ESP_WifiManager@^1.12.1
    tzapu/WifiManager
    https://github.com/dok-net/ghostl ; << Fix missing circular_queue.h in plerup/EspSoftwareSerial@8.2.0
	plerup/EspSoftwareSerial
	milesburton/DallasTemperature
board_build.filesystem = littlefs
extra_scripts = gzip_littlefs.py
monitor_speed = 76800
board_build.f_cpu = 160000000L
; board_build.f_flash = 80000000L;
board_upload.resetmethod = nodemcu
;upload_speed = 921600

visualapproach, 2024-04-28:
I'll try to remember to lower the upload speed next time :)