visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #699: [FEATURE RQ] Add a second filter timer so that we can track how often to wash the filter, v replace it.

PaleFlyer, 2024-05-01:
Basically, I have been using the same filter for over 41 days, but after 7 days I have to wash it out, and then about every 5 days I have to wash the filter again, or I get the [E02} error. So I want to have both the "replace" filter timer, AND a wash timer, so I can track if I need to wash the filter, before I get an E02, as if the E02 triggers, I might lose a lot of heat for the pump being off, and I need to look into a better way to clear the E02 than pulling power, versus just taking a couple min to wash the filter out.

And maybe even a 3rd timer for a "deep" filter clean (Haven't had to do this one yet, but I have some filter cleaner to use if the rinse doesn't work, which I'm getting close on.)

visualapproach, 2024-05-01:
Should be doable. I'll save this FR for later use :-)

visualapproach, 2024-05-09:
Feature added in development branch

visualapproach, 2024-05-10:
@PaleFlyer Just a reminder to also upload the new data files to the ESP flash drive. You can make a new filesystem image but it will erase any settings you have saved. So the best way is to upload files from the menu 'upload files' in the UI. Upload index.html and .js and config.html

closed 2024-07-17 by github-actions[bot]