visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #700: [BUG] mqtt server restart switches on pump and heating

ZoXx, 2024-05-03:
Good morning everyone
After having the problem for about a week now (since the Lay-Z-Spa is back up) I was able to narrow it down and find the problem.

My iobroker (docker) is shut down at night and backed up. The docker is then restarted. At that moment, the MQTT server (in iobroker as an adapter) is of course also restarted. 

shortly afterwards, the lay-z-spa connects via mqtt and switches on the pump and the heating! There are no scripts running in iobroker that trigger this, nor is there a routine etc. What could be the problem here? Why is the pump and heating started by the lay-z-spa after every mqtt restart?

visualapproach, 2024-05-04:
Hi, My MQTT broker is restarted every week and I don't have that problem. Are there any messages sent from your broker at start-up? You can use something like mqtt explorer to monitor

ZoXx, 2024-05-05:
Found it. :) THANKS!

ZoXx, 2024-05-15:
Not really. Same problem again. No other mqtt device got the same problem.

visualapproach, 2024-05-15:
Did you check if the broker sent any messages at restart?

ZoXx, 2024-06-22:
I tested everything and couldn't find anything. the 50 or so other devices never caused any problems either.
i have updated to the latest version and esd now works without any problems. there is no restart anymore

ZoXx, 2024-07-10:
Good morning there ar some news.
With version v4.3.1 the error was gone. For two weeks, there was no problem with the heating being switched on at night and the pump accordingly.
Yesterday I updated to v4.4.1 and in the night to today the problem was immediately present again

ZoXx, 2024-07-10:
No reboot this night

visualapproach, 2024-07-10:
I can't replicate the problem and I can't figure out why it would change the heater status.
Pump Model?

visualapproach, 2024-07-10:
you can post a screenshot of your mqtt page also

ZoXx, 2024-07-10:
pump model: 6 wire, 2021 (air)
i changed today from iobroker mqtt to homeassistant mqtt.