visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #702: Code issues?

1NewC, 2024-05-04:
Hello,

I am hoping to make this project work for my spa, but am having issues with the code. When try to connect to the lazyspa.local url using the provide credentials, my board goes into some sort of rapid reset loop.

When I debug the code, I receive the following errors... I am very new to coding and was wondering if someone can help me out?  I have not modified the provided code.

[{
	"resource": "/c:/......................./Our Spa/WiFi-remote-for-Bestway-Lay-Z-SPA-4.2.5/Code/.pio/libdeps/nodemcuv2/PubSubClient/src/PubSubClient.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]",
	"startLineNumber": 523,
	"startColumn": 16,
	"endLineNumber": 523,
	"endColumn": 16
},{
	"resource": "/c:/.....................................s/Our Spa/WiFi-remote-for-Bestway-Lay-Z-SPA-4.2.5/Code/.pio/libdeps/nodemcuv2/WebSockets/src/SocketIOclient.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'void WebSocketsClient::loop()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 175,
	"startColumn": 23,
	"endLineNumber": 175,
	"endColumn": 23
},{
	"resource": "/c:/....................................../Our Spa/WiFi-remote-for-Bestway-Lay-Z-SPA-4.2.5/Code/.pio/libdeps/nodemcuv2/WebSockets/src/SocketIOclient.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'void WebSocketsClient::loop()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 175,
	"startColumn": 28,
	"endLineNumber": 175,
	"endColumn": 28
},{
	"resource": "/c:/................................../Our Spa/WiFi-remote-for-Bestway-Lay-Z-SPA-4.2.5/Code/src/main.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "deleting object of polymorphic class type 'PubSubClient' which has non-virtual destructor might cause undefined behavior [-Wdelete-non-virtual-dtor]",
	"startLineNumber": 512,
	"startColumn": 5,
	"endLineNumber": 512,
	"endColumn": 5
}]

visualapproach, 2024-05-04:
Hi, those logs are compile time messages and they are expected and nothing to worry about. Not much else to go on from your post I'm afraid. The more details about your setup the higher chance you will get help. Like what version are you using? People often states "The latest one" but it is not a proper answer since there may have been several updates after they tried, and there is two branches so which one is referred to? Also, are you running on a bench or in the tub? What hardware? And so on.

1NewC, 2024-05-04:
Hello, thank you for your response, I apologize for the lack of information. I am new to Arduino and Virtual Studio Code, but am trying to learn.  I am running on a bench and working with version 4.2.5.  I can flash to the ESP8266 without an issue and without errors.  The "Laszyspamodule" access point is created and I can connect to it, however I am not sure the UI looks the way it is supposed.  I get this (capture):

When I click on configure WiFi, I can locate my network and entire my credentials without issue.  However, after doing so I get "Credentials Saved Trying to connect ESP to network. If it fails reconnect to AP to try again".  Then the blue led on the SP8266 starts flashing erratically and I can no longer see the "Laszyspamodule" access point.  The only way for me to recover the board is to reflash it... 

I feel like I am missing something simple, just no sure what it is...  The only UI I am presented with, looks different than with the instructions.

Thank you very much for your help and for putting this together!

1NewC, 2024-05-04:
[image]

1NewC, 2024-05-04:
[image]

choumarin, 2024-05-04:
Once you enter your home Wifi credentials, the module will connect to your wifi, and will not be an access point anymore.
Reconnect to your home wifi and connect to its IP, either from the logs if you have the module connected via usb, or via your router's interfance.
eg. http://192.168.0.32
If your network supports it, you might also be able to get via mdns: http://layzspa.local

1NewC, 2024-05-04:
This is the point at which I get stuck.  When I try to connect to its IP, I get a "took too long to connect" error.

When I connect to a serial monitor, I see the following:

2:52:27.103 -> start NTP
12:52:27.266 -> ............WS IRamheap 10912
12:52:32.775 -> IRamheap 9664
12:52:32.775 -> startmqtt
12:52:32.775 -> Failed to read mqtt.json. Using defaults.
12:52:32.809 -> 192.168.1.xxx
12:52:32.809 -> End of setup()
12:52:32.809 -> 36608

The LED light on the board is flashing extremely fast.  Unless I am missing something, I don't believe the LED should be flashing.

choumarin, 2024-05-04:
Actually the led flashing is normal.
Make sure you're connecting to http and not https, some browser try to be too smart

1NewC, 2024-05-05:
OK, I was able to get it to work. Problem was that my 2.4G network was down and I keep unknowingly connecting to 5G.  

Problem is now that my web interface does not look correct, any thoughs?

[image]

1NewC, 2024-05-05:
Never mind, It seems to be an issue with 4.2.5, it works with the code in the development branch.  Thank you everyone for your help.

closed 2024-05-05 by 1NewC