visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #706: [BUG] The reported power usage is wrong (for US devices?)

choumarin, 2024-05-13:
**Describe the bug**
The values for power usages are hardcoded to :

 HEATERPOWER = 1900
 PUMPPOWER = 40
 AIRPOWER = 800
 IDLEPOWER = 2
 JETPOWER = 400

However, my observation is that heater + pump is more around 1200W
This actually matches with the user manual, eg. https://p360.bestwaycorp.com/Manual/D064DFC3-8013-4CEA-BB9B-79DC97341B33.pdf page 15

I'm guessing give the international nature of this project, the values above are EU / rest of the world that knows not to use 110V.
Given the values on my user manual, the values for a US system would be:

 HEATERPOWER = 1243
 PUMPPOWER = 77
 AIRPOWER = 715
 IDLEPOWER = 2
 JETPOWER = ?? (my model doesn't have that feature, but I can try to lookup another model which has)

Also, there should be a logic to consider that the max is 12A even if everything is turned on.

I'll need to revalidate all those numbers by running only those modes.

**To Reproduce**
Check power usage with "smart plug"

**Expected behavior**
I would expect the reported numbers to be correct.
The current hardcoded values could be replaced by either:
- a selector between 110V / 220V systems and hardcoded values
- an override for each values to be configured by the user

**Screenshots**
N/A

**Hardware (please complete the following information):**
 - S100105
 - 6 wires
 - PCB V2B
 - "red one"

**Software (please complete the following information):**
 - Version 2023-11-19-1330

**Additional context**
N/A

choumarin, 2024-05-13:
I can volunteer for implementation if we agree on the logic to implement,

visualapproach, 2024-05-14:
I think user defined values would be the best route because there's probably quite a lot of variation between the pumps. The values are already customizable in code but I guess most ppl don't wanna touch the code. It makes updates more tedious. You are welcome to make a PR to development branch if you want. I'm thinking maybe hardware config page could host a few more input fields.

choumarin, 2024-05-16:
Ran more tests, here are the numbers I get:
HEATERPOWER = 1225
PUMPPOWER = 35
AIRPOWER = 765
IDLEPOWER = 2

choumarin, 2024-05-21:
I've started working on the implementation

choumarin, 2024-06-10:
closed with https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/pull/713

closed 2024-06-10 by choumarin