visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #711: Monaco 4 wire model DSP 04 on display and gui only. Not detecting water temp.

dasmith83, 2024-05-21:
- Pump model - Monaco 54113
 - 4 wire
 - PCB type D1 mini only
 - Level shifter type N\A

 - SW Version 4.2.6

Hi all.

Firstly, great work @visualapproach! I love these little ESP boards and really wish I had the ability to create such code from scratch myself. I take my hat off to you sir!

I've got a Monaco 4 wire pump (and actually a few 6 wire pumps I've grabbed for spares too lol).

Trying to get the 4 wire 54113 Monaco egg working, I've come across the below issue...

All works buttons and GUI control works fine, except the DSP and Wi-Fi module, doesn't seem to be "seeing" the water temp. So when the heater is turned on, either from the DSP, or from the web GUI, it just continues to heat until the egg overheats (E08 I think?). 

The DSP constantly displays "04" rather than the water temp, even before the pump or heater is turned on.

When I remove the module and connect the egg to the DSP as original again, all works fine.

Any ideas what might be causing this?

I've configured it as below:
D7 DSP CLK/RX
D6 DSP TD/RX
D4 CIO CLK/RX
D3 CIO TD/RX

I also tested with one of the 6-wire eggs and it all worked fine - which is great! But it heated really slow, so I suspect that egg has a faulty heater (same behaviour without the module in-line). One thing I noticed when testing the 6 wire egg, was the 04 shows on the egg DSP until the esp has fully booted, then it displays water temp as you'd expect. So something odd is happening on the 4 wire model for sure.

Thanks in advance for any suggestions.

Dave

dasmith83, 2024-05-22:
> Thanks for your kind words! I'd love to help you get it sorted, but since you're out of spec (no level converter) there is too many unknowns. I don't say it's the reason but the fact that it might be is enough for me. Also see #690

Thanks for the quick reply :)

Completely understand your position - I figured it was worth a try as I already had some D1s lying around from previous projects.

Oddly enough, I went back to it early today and to my surprise the temp. had started coming through. Upon more testing, it seems the temp is read and sent to the DSP, it just takes a few minutes post boot to come through. Apparently I was being too impatient when initially testing.

So further tests will continue, but for now all aspects are working wonderfully!

Now for the automation fun with HA to heat when solar is in abundance and battery is full lol.

Thanks again - I've also sent you some ££ for a beer.

visualapproach, 2024-05-23:
Thanks!! I will convert it to liquid ASAP 😃 
Good to know about the startup delay.

visualapproach, 2024-05-24:
It could very well be the case. No promises but I wouldn't be surprised if it works better.

visualapproach, 2024-06-05:
I still think it's signal problems. We can start with some logging:
[click here for debug on](http://layzspa.local/debug-on/)
[click here to turn off debug](http://layzspa.local/debug-off/)

Run the debug for a few minutes while pressing some buttons on display. Post the debug log file here.

dasmith83, 2024-06-06:
Thanks for this. OK, so i turned debug on and then pressed the DSP buttons as below:
Bubbles - on/off/on/off/on/off
Pump - on/off/on/off
Temp - increase to 24degC
Heater - on/off, then Pump off

Log.txt file appeared as below when then turning debug off:

1717662009SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18059
1717662034SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18260
1717662036SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18281
1717662040SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18311
1717662042SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18331
1717662047SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18372
1717662053SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18413
1717662058SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18458
1717662061SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18481
1717662069SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18544
1717662074SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18589
1717662087SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18690
1717662088SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,11,0,1,15,FD,0,0,0,0 ESP-CIO:BB,3,11,0,1,15,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18704
1717662090SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,13,0,1,17,FD,0,0,0,0 ESP-CIO:BB,3,13,0,1,17,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18720
1717662096SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18767
1717662098SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18783
1717662103SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18827

When i then re-enabled debug and use the web interface to enable "take control" it turned off straight away after a few presses, but eventually remained on, and the pump then remained on too...
I then turned bubbles on, from the GUI, but it didn't turn on until the 3rd or 4th press, and really delayed, then turned off but also after a few seconds delay too.
I then set the temp on GUI to 26 - I then turned on the heater, but both of these attempts failed and the sliders went back to default positions. A few seconds later, they then magically started working and remained on.

The log.txt doesn't seem to have updated though after the gui button presses? unless i'm looking at the wrong file?

1717662009SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18059
1717662034SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18260
1717662036SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18281
1717662040SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18311
1717662042SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18331
1717662047SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18372
[... cut ...]
1717662096SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18767
1717662098SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18783
1717662103SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 18827

All through the above, the DSP shows 04 and the GUI [ ]

dasmith83, 2024-06-06:
I just deleted the log.txt, turned debug on again, and tried the buttons in the GUI - it didn't recreate the log file.

But then when i press the DSP buttons, a new log file was created.
So i turned on the pump, then the bubbles, then turned bubbles off and then turned the pump off. Log as below...

1717664848SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 41041
1717664858SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,30,0,1,34,FD,0,0,0,0 ESP-CIO:BB,3,30,0,1,34,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 41116
1717664859SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 41131
1717664861SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 41139
1717664866SW:2023-11-19-1330 CIO-ESP:0,0,0,0,0,0,0,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
cio msg count: 0 dsp msg count: 41186

DSP still showing 04.

visualapproach, 2024-06-06:
I guess you've already tried with the CIO cables connected to the A-port as well?

dasmith83, 2024-06-06:
I've just tried moving to the pins on H1A and H2A...

a) If I move the CIO pins to H1A and leave the DSP pins on H2B, I still see 04 on DSP and [ ] on the web gui.
b) If I move the CIO pins to H1A and the DSP pins to H2A, I see 04 on the DSP and [25] on the web gui (temp works on the gui! :)
c) If I do the opposite of a) and move the CIO pins to H1B and the DSP pins to H2A, I see 04 on DSP and [ ] on web gui again.

So it seems using the HA pins for both CIO and DSP at least shows the temp on the web gui, but doesn't pass it on to the actual DSP?

So...debug turned on again, with the HA pins being used instead shows the below.
DSP attempt to turn on pump initially worked, but then the pump turned off and all buttons on pump didn't have any actual impact as such (web gui take control was still disabled). A few seconds later the web gui then showed E02, and the DSP remained at 04.

1717667018SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1268
1717667042SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1462
1717667074SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1697
1717667076SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1711
1717667079SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,30,0,1,34,FD,0,0,0,0 ESP-CIO:BB,3,30,0,1,34,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1740
1717667080SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1746
1717667081SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1757
1717667083SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1772
1717667086SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1798
1717667092SW:2023-11-19-1330 CIO-ESP:BC,3,19,0,1,1D,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,19,0,1,1D,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1846

Restarting the ESP, leaving pins on the HA placement and trying again...
Pump on, temp to 26, heater...then pump off when gui showed temp had reached 30 (I'm using a descaling unit to test away from the spa liner, so not surprising it overshot the temp.). DSP remained 04.

1717667362SW:2023-11-19-1330 CIO-ESP:BC,3,14,0,1,18,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,14,0,1,18,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1123
1717667383SW:2023-11-19-1330 CIO-ESP:BC,3,14,0,1,18,FD,0,0,0,0	 DSP-ESP:BB,3,11,0,1,15,FD,0,0,0,0 ESP-CIO:BB,3,11,0,1,15,FD,0,0,0,0	 ESP-DSP:BC,3,14,0,1,18,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1294
1717667385SW:2023-11-19-1330 CIO-ESP:BC,3,14,0,1,18,FD,0,0,0,0	 DSP-ESP:BB,3,13,0,1,17,FD,0,0,0,0 ESP-CIO:BB,3,13,0,1,17,FD,0,0,0,0	 ESP-DSP:BC,3,14,0,1,18,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1311
1717667415SW:2023-11-19-1330 CIO-ESP:BC,3,14,0,1,18,FD,0,0,0,0	 DSP-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0 ESP-CIO:BB,3,1B,0,1,1F,FD,0,0,0,0	 ESP-DSP:BC,3,14,0,1,18,FD,0,0,0,0
cio msg count: 4 dsp msg count: 1556
1717667459SW:2023-11-19-1330 CIO-ESP:BC,3,1E,0,1,22,FD,0,0,0,0	 DSP-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0 ESP-CIO:BB,3,1B,0,1,1F,FD,0,0,0,0	 ESP-DSP:BC,3,1E,0,1,22,FD,0,0,0,0
cio msg count: 5 dsp msg count: 1900
1717667546SW:2023-11-19-1330 CIO-ESP:BC,3,30,0,1,34,FD,0,0,0,0	 DSP-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0 ESP-CIO:BB,3,1B,0,1,1F,FD,0,0,0,0	 ESP-DSP:BC,3,30,0,1,34,FD,0,0,0,0
cio msg count: 6 dsp msg count: 2592
1717667547SW:2023-11-19-1330 CIO-ESP:BC,3,30,0,1,34,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,30,0,1,34,FD,0,0,0,0
cio msg count: 6 dsp msg count: 2602
1717667553SW:2023-11-19-1330 CIO-ESP:BC,3,30,0,1,34,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,30,0,1,34,FD,0,0,0,0
cio msg count: 6 dsp msg count: 2648

I'll do the continuity tests suggested shortly :)
I used dupont's purely whilst working out which cable should go where as it gave me flexibility. I've spare boards and components to hard wire, once POC successful :)

dasmith83, 2024-06-06:
Also...i've found with HA pins in use, when the gui is "in control" and pump running...the DSP buttons for the bubbles still work? is that expected? the heater/pump DSP buttons beep and lights change, but have no impact as i'd expect. Log for this scenario below...

1717667885SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 11 dsp msg count: 5316
1717667890SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 11 dsp msg count: 5356
1717667894SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 11 dsp msg count: 5389
1717667899SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 11 dsp msg count: 5430
1717667903SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,30,0,0,33,FD,0,0,0,0 ESP-CIO:BB,3,30,0,0,33,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 11 dsp msg count: 5468
1717667904SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 11 dsp msg count: 5477
1717667905SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 11 dsp msg count: 5479
1717667905SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 12 dsp msg count: 5481
1717667907SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 12 dsp msg count: 5498
1717667908SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 12 dsp msg count: 5509
1717667909SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 12 dsp msg count: 5519
1717667911SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 12 dsp msg count: 5532
1717667916SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 12 dsp msg count: 5576

visualapproach, 2024-06-06:
So, now it receives a few msg from the CIO. But maybe not all of'em. Regarding what happens when pressing dsp when "in control" I don't remember right now to be honest. I've played with trying to enable control from both UI and DSP, but not sure I implemented it. Best not to fiddle with the DSP whilst in control.
You can do the hardware test to see if the device can connect to itself. Broken cables should show as errors in this test. It doesn't account for signal loss in the CIO/DSP though. [to the test](http://layzspa.local/hwtestinfo.html)

dasmith83, 2024-06-06:
Thanks, yea the hardware test came back fine before I plugged it into the egg...running it again now however, as I'm now using pins 2&5 and 4&3, the HW test seems to fail as it's trying to use D5 and D7 instead of 5 and 3?

Does this suggest the ESP might not be using pin 3 as it's configured to do? I wonder if i need to move the  DSP from pin 3 to 7...

Log output...
Start test. Seq begins with HIGH, then alters.

Sending on D2, receiving on D4
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D5, receiving on D7
-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0 // 50 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100
Sending on D4, receiving on D2
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D7, receiving on D5
-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0 // 50 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100
End of test!
Errors indicated by 1 or 0 depending on test state. - is good.
Switching cio pins 5s HIGH -> 5s LOW -> input
then DSP pins 5s HIGH -> 5s LOW -> input, repeating

dasmith83, 2024-06-06:
Moving DSP pin 3 to 7 makes the hwtest now work...

Sending on D2, receiving on D4
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D5, receiving on D7
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100
Sending on D4, receiving on D2
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D7, receiving on D5
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100

visualapproach, 2024-06-06:
Hm I need to check if the test uses the right pins. Doing other stuff now but will try to do it later today

dasmith83, 2024-06-06:
OK...so testing with using the pins as per the hwtest instead of what i'd defined in the hwconfig page (i.e. moving DSP pin from 3 to 7). Seems to work better all around...

DSP shows temp and so does the web gui. When turning pump on and bubbles and heater etc. on DSP, below is the debug log output...

1717672429SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 10889
1717672441SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 10991
1717672443SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,30,0,1,34,FD,0,0,0,0 ESP-CIO:BB,3,30,0,1,34,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 11008
1717672445SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 11026
1717672448SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,30,0,1,34,FD,0,0,0,0 ESP-CIO:BB,3,30,0,1,34,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 11048
1717672450SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 11064
1717672457SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,11,0,1,15,FD,0,0,0,0 ESP-CIO:BB,3,11,0,1,15,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 11123
1717672459SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,13,0,1,17,FD,0,0,0,0 ESP-CIO:BB,3,13,0,1,17,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 11136
1717672489SW:2023-11-19-1330 CIO-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0	 DSP-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0 ESP-CIO:BB,3,1B,0,1,1F,FD,0,0,0,0	 ESP-DSP:BC,3,1B,0,1,1F,FD,0,0,0,0
cio msg count: 535 dsp msg count: 11384
1717672553SW:2023-11-19-1330 CIO-ESP:BC,3,2B,0,1,2F,FD,0,0,0,0	 DSP-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0 ESP-CIO:BB,3,1B,0,1,1F,FD,0,0,0,0	 ESP-DSP:BC,3,2B,0,1,2F,FD,0,0,0,0
cio msg count: 536 dsp msg count: 11904
1717672553SW:2023-11-19-1330 CIO-ESP:BC,3,2B,0,1,2F,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0 ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0	 ESP-DSP:BC,3,2B,0,1,2F,FD,0,0,0,0
cio msg count: 536 dsp msg count: 11905
1717672565SW:2023-11-19-1330 CIO-ESP:BC,3,2B,0,1,2F,FD,0,0,0,0	 DSP-ESP:BB,3,10,0,0,13,FD,0,0,0,0 ESP-CIO:BB,3,10,0,0,13,FD,0,0,0,0	 ESP-DSP:BC,3,2B,0,1,2F,FD,0,0,0,0
cio msg count: 536 dsp msg count: 12010
1717672571SW:2023-11-19-1330 CIO-ESP:BC,3,2B,0,1,2F,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2B,0,1,2F,FD,0,0,0,0
cio msg count: 536 dsp msg count: 12052

When I turn it off from the DSP, and enable control on the gui, enable the pump, and try and use the DSP buttons for the bubbles, it (on one occasion) "blips the blower on and then off with every press". But when i enabled logging, typically it didn't do this again, but instead turned the blower on/off with every DSP press, even though the gui was in control. Then when i enabled the blower on the gui, and pressed the bubbles DSP button, it would turn it off even though the dsp light said it was on. but the gui would update and show it as off.

Certainly seems to be the best result so far though!

1717672791SW:2023-11-19-1330 CIO-ESP:BC,3,30,0,1,34,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,30,0,1,34,FD,0,0,0,0
cio msg count: 537 dsp msg count: 13828
1717672809SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 538 dsp msg count: 13979
1717672827SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 538 dsp msg count: 14121
1717672833SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 538 dsp msg count: 14171
1717672839SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 538 dsp msg count: 14223
1717672841SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 538 dsp msg count: 14240
1717672854SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 541 dsp msg count: 14341
1717672860SW:2023-11-19-1330 CIO-ESP:BC,3,2F,0,1,33,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2F,0,1,33,FD,0,0,0,0
cio msg count: 543 dsp msg count: 14388
1717672862SW:2023-11-19-1330 CIO-ESP:BC,3,2E,0,1,32,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2E,0,1,32,FD,0,0,0,0
cio msg count: 544 dsp msg count: 14408
[... cut ...]
1717673049SW:2023-11-19-1330 CIO-ESP:BC,3,2D,0,1,31,FD,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:BC,3,2D,0,1,31,FD,0,0,0,0
cio msg count: 616 dsp msg count: 15902
1717673055SW:2023-11-19-1330 CIO-ESP:BC,3,2D,0,1,31,FD,0,0,0,0	 DSP-ESP:BB,3,0,0,0,3,FD,0,0,0,0 ESP-CIO:BB,3,0,0,0,3,FD,0,0,0,0	 ESP-DSP:BC,3,2D,0,1,31,FD,0,0,0,0
cio msg count: 617 dsp msg count: 15948
1717673058SW:2023-11-19-1330 CIO-ESP:BC,3,2D,0,1,31,FD,0,0,0,0	 DSP-ESP:BB,3,20,0,0,23,FD,0,0,0,0 ESP-CIO:BB,3,20,0,0,23,FD,0,0,0,0	 ESP-DSP:BC,3,2D,0,1,31,FD,0,0,0,0
cio msg count: 617 dsp msg count: 15976

visualapproach, 2024-06-06:
can you post the file `hwcfg.json` please?

visualapproach, 2024-06-06:
My bet is that it says 
`pins":["2","5","","4","7","","",""]` instead of 
`pins":["2","5","","4","3","","",""]`
In that case select your pump model and pcb v2b again, and remember to hit SAVE. Then move back to pin D3. Default pins *should* have highest success rate. Check the file after saving to confirm it is saved properly, before restarting.

dasmith83, 2024-06-06:
Would it matter though? As long as the pins it “thinks” it should use are the ones the wires are connected to?

{"cio":"5","dsp":"8","pcb":"v2b","hasTempSensor":"0","pins":["2","5","","4","7","","",""]}

visualapproach, 2024-06-07:
If it works you can use any pins you like. Someone with more knowledge about electronics than me figured out that some pins are better suited for particular circuits than other pins. I'm not sure it applies when LLC is used. IIRC it's described in #390

closed 2024-08-14 by github-actions[bot]