visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #714: Unable to install

markusschmitz81, 2024-05-26:
**Hardware (please complete the following information):**
 - Lazy Spa St. Lucia/ Rio S100101
 - 6 wires
 - BWCPCB _V2B
 - Level shifter type 1

**Software (please complete the following information):**
 - Visual Code Version: 1.89.1 (Universal) and newest PlatformIo

I am unable to connect to the ESP8266, I build the board like it is describe in the manual. After that I downloaded VSC and installed PlatformIo. 

I build the file system and then I tried to upload the file system but I am unable to get a connection to the Esp. 
I have two options cu.Bluetooth-incoming-Port and cu.BLTH, okay and auto. But nothing works.

the blue LED flashs sometimes but not more. 

Can somebody give me an andvise?

Thank you 

Messages from the VSC Terminal see PDF...

[terminal upload filesystem.pdf](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/15447851/terminal.upload.filesystem.pdf)

markusschmitz81, 2024-06-01:
Problem fixed: Wrong Usb cable and windows deactived the usb ports after an update

visualapproach, 2024-06-01:
Lovely 😊 Thanks for the report

closed 2024-08-09 by github-actions[bot]