visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #716: [BUG] can't read all of calibration pop up text

brucebuchan89, 2024-05-28:
**Describe the bug**
[image]

See screenshot, positioning of the pop up text for calibration makes it difficult to read; have tried on computer and mobile browsers

**To Reproduce**
hover over question mark

**Expected behavior**
pop up should be centralised on screen

**Screenshots**
see screenshot

**Software (please complete the following information):**
 - Firmware: 2023-11-19-1330

Thanks

visualapproach, 2024-05-28:
It's fixed in dev branch

brucebuchan89, 2024-05-28:
thanks @visualapproach; for the time being, what is the text meant to say? thanks

visualapproach, 2024-05-28:
[image]

closed 2024-08-04 by github-actions[bot]