visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #729: [OTHER] Miami 4 wires 04 on screen

piltous, 2024-06-14:
**Hardware (please complete the following information):**
 - Pump model: old miami 54123 (sticker outside)
 - 4  wires
 - PCB type: v2B
 - Level shifter type: LLC red one, no cap or resistor

**Software (please complete the following information):**
 - Version: Firmware: 2024-06-08-1820

Hello @visualapproach 

I have my setup in place but communication issue, it seems with the DSP

```
2024-06-14 18:37:01UTC.  SW:2024-06-08-1820 
CIO-ESP:BB,3,10,0,1,14,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,10,0,1,14,FD,0,0,0,0
CIO msg count: 1519 DSP msg count: 0
CIO msg quality: 100.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)
```
The panel is working but the screen shows "04". I saw some similar issues but not sure if the fix would be the same.

[image]

So I have a value coming for the temperature which looks good, target is not and I have no control...
But if if I take control and switch C to F, to pump stop but the led panel keep on...

[image]

What should I start to try to investigate? My skills are a bit limited....

Thanks in advance,

Julien

piltous, 2024-06-15:
adding the inputs

```
Edges received on pin D2: 1
Edges received on pin D5: 1
Edges received on pin D0: 0
Edges received on pin D4: 1
Edges received on pin D3: 1
Edges received on pin D0: 0
Edges received on pin D0: 0
On 6-w pump the highest number is CLK, next is DATA and third is CS. On 4-wires the highest is CIO or DSP TX to ESP.
```

and hwtest using using a usb cable on the chip

```
Start test. Seq begins with HIGH, then alters.

Sending on D2, receiving on D4
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D5, receiving on D3
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100
Sending on D4, receiving on D2
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D3, receiving on D5
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100
End of test!
"1" or "0" indicates ERROR, depending on test state. "-" is good.
Switching cio pins 5s HIGH -> 5s LOW -> input
then DSP pins 5s HIGH -> 5s LOW -> input, repeating
Disconnect cables then reset chip when done!
```

visualapproach, 2024-06-15:
Hi, see if latest development branch helps.

piltous, 2024-06-15:
thanks for your reply!

it seems a bit better for the logs but still the same issue

```
2024-06-15 09:46:40UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,10,0,1,14,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,10,0,1,14,FD,0,0,0,0
CIO msg count: 218 DSP msg count: 0
CIO msg quality: 100.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)
```

```
Edges received on pin D2: 1
Edges received on pin D5: 1213
Edges received on pin D0: 1
Edges received on pin D4: 1
Edges received on pin D3: 1
Edges received on pin D0: 1
Edges received on pin D0: 1
On 6-w pump the highest number is CLK, next is DATA and third is CS. On 4-wires the highest is CIO or DSP TX to ESP.
```

I'm still not sure if it could be an hardware issue or not...

Also, how the esp can remind my wireless setup in memory? should I clear something?

visualapproach, 2024-06-15:
You should have data (edges) on D4 on your input test, but it may be that the DSP has just given up after some timeout. Maybe try the inputs right after power up. The ESP/PCB  device itself seems to be fine judging from the test.
Only thing I can see as a hardware failure point is the connector to the dsp not making contact.

The ESP stores the WiFi credentials in "EEPROM" and my program also saves it in a file on the flash memory. If you want it to forget all the WiFi settings you can click the resetwifi link in the links.html file, found in the directory. But I don't think this has anything to do with your issue.

piltous, 2024-06-15:
I build a new hardware (new PCB, new TTC) but still the same...
I got a second nodemcu but it seems it has issue, I can reach the files like logs but not the main interface and it blinks.
maybe a got defective nodemcu, I should try with an other one maybe...

visualapproach, 2024-06-15:
I think you forgot to upload the file system image

piltous, 2024-06-15:
oh... I don't even know that and how I did the 1st time, thank you!

So I'm now with a functional firmware dev branch, full new hardware but still same :/

```
Edges received on pin D2: 1
Edges received on pin D5: 1195
Edges received on pin D0: 1
Edges received on pin D4: 1
Edges received on pin D3: 1
Edges received on pin D0: 1
Edges received on pin D0: 1
On 6-w pump the highest number is CLK, next is DATA and third is CS. On 4-wires the highest is CIO or DSP TX to ESP.
```

2024-06-15 15:59:24UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,10,0,1,14,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,10,0,1,14,FD,0,0,0,0
CIO msg count: 817 DSP msg count: 0
CIO msg quality: 100.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)

2024-06-15 16:04:31UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,10,0,1,14,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,10,0,1,14,FD,0,0,0,0
CIO msg count: 38 DSP msg count: 0
CIO msg quality: 100.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)

Here is a pic of my hardware in case of...

[image]

visualapproach, 2024-06-15:
Your devices are probably fine. The question is why the DSP doesn't communicate. You could try the filter caps+resistors on the DSP port B but I'm not sure it'll help. Hmm...

visualapproach, 2024-06-15:
What happens if you briefly power the esp with a usb power bank in the hot tub?

piltous, 2024-06-15:
thanks! I'll give a try for the cap and resistor on DSP but not sure of the reference, I'll search in the issues.
I tried the powerbank but it has not effect.
I though my model was quite generic but it seems not so easy.
I will give a try to a old firmware maybe just to see what happens.

Many thanks again for your help!!! and you work indeed, it's impressive!

visualapproach, 2024-06-15:
Thank you @piltous 
Good ideas

visualapproach, 2024-06-15:
Try v3.4 or 3.5

piltous, 2024-06-15:
so possibly something with my DST itself....

I changes the model to 
```
#define NO54123 //not tested. ref https://github.com/mrQ000/layz-rc link to presentation slides
```
and pin to this for PCB V2B
```
#ifdef PCB_V2
const int CIO_RX = D2;
const int CIO_TX = D5;
const int DSP_TX = D4;
const int DSP_RX = D3;
```
[image]

I hope cap and res will do the trick :)
I'll let you know.

Cheers

piltous, 2024-06-16:
Hi there! :)

Something I didn't get...

Logs:
```
CIO-ESP:BB,3,10,0,1,14,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,10,0,1,14,FD,0,0,0,0
CIO msg count: 817 DSP msg count: 0
```

Inputs:
```
Edges received on pin D2: 1
Edges received on pin D5: 1195
Edges received on pin D4: 1
Edges received on pin D3: 1
```

Status (on old firmware 3.4.0):
```
CSIO TX Active
CSIO RX ?
DSP TX Dead
DSP RX ?
```

Pin for PCBV2B:
```
const int CIO_RX = D2;
const int CIO_TX = D5;
const int DSP_TX = D4;
const int DSP_RX = D3;
```

CSIO-ESP is CSIO TX? so D5?
and ESP-DSP is DSP RX? so D3?

For me D2, D4, D3 is still "1" as input so that would mean I only have CSI TX which is working on D5....

Have a nice Sunday! ;)

visualapproach, 2024-06-17:
Thx, yes that sounds right.

piltous, 2024-07-08:
Hi @visualapproach,
I'm losing hope to get it works...
I added the cap and res and still down for the DSP TX with the old firmware :/
Still 04 and the pump screen, I can control pump, bubbles, heat from the panel but nothing from the app.
Any suggestion to troubleshoot?
I'm convaince some people suceed to get it works. It would be nice a kind of knowledge DB.
Thanks!

visualapproach, 2024-07-08:
Hi @piltous 
Take a look at this https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/742#issuecomment-2212501402
Maybe that will do the trick?

Yeah a KDB would be nice, if someone managed it... I have tried to put a few things in the FAQ in the manual, but other than that I don't have the time to do it. I think of this forum as the data base. Admittently a lot to search through.

piltous, 2024-07-08:
Many thanks for the prompt reply!
I'm right now on the new firmware (dev branch) and I have the pump fully functional without 04 anymore, it’s a good progress for me :)
So the pin 7 is better for me (I missed one step before my last post certaines...).
If it’s really matter of pin changes I'll try d1 and d6 as well.
Many the composent also may differ one from other constructor.
In knowledge DB, I was not meaning more works for you, you did enough. More a place where  users can give feedback on their setup and as contribution.
Cheers! ;)

piltous, 2024-07-09:
just to log my last progress... :)

```
2024-07-09 17:36:18UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,0,0,1,4,FD,0,0,0,0
DSP-ESP:9E,3,23,2,1,29,FD,0,0,0,0
ESP-CIO:9E,3,23,2,1,29,FD,0,0,0,0
ESP-DSP:BB,3,0,0,1,4,FD,0,0,0,0
CIO msg count: 271 DSP msg count: 105
CIO msg quality: 100.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-07-09 17:36:18UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,0,0,1,4,FD,0,0,0,0
DSP-ESP:BC,3,23,2,1,29,FD,0,0,0,0
ESP-CIO:BC,3,23,2,1,29,FD,0,0,0,0
ESP-DSP:BB,3,0,0,1,4,FD,0,0,0,0
CIO msg count: 273 DSP msg count: 106
CIO msg quality: 100.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-07-09 17:40:37UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,0,0,1,4,FD,0,0,0,0
DSP-ESP:BC,3,23,2,1,29,FD,0,0,0,0
ESP-CIO:BC,3,23,2,1,29,FD,0,0,0,0
ESP-DSP:BB,3,0,0,1,4,FD,0,0,0,0
CIO msg count: 110 DSP msg count: 37
CIO msg quality: 100.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

```

```
Edges received on pin D2: 1
Edges received on pin D5: 1225
Edges received on pin D0: 0
Edges received on pin D4: 1
Edges received on pin D7: 1
Edges received on pin D0: 0
Edges received on pin D0: 0
On 6-w pump the highest number is CLK, next is DATA and third is CS. On 4-wires the highest is CIO or DSP TX to ESP.
```

It looks now good for the communication I think. (I'm still missing is this logs covered the RX or TX for both CIO/DSP...)

Now E03 issue after couple of minutes. I will search in the issues reported.

piltous, 2024-07-09:
[image]

No more E03 right now :/ my hardware seems unstable

visualapproach, 2024-07-09:
> It looks now good for the communication I think. (I'm still missing is this logs covered the RX or TX for both CIO/DSP...)

If I interpret this right, you mean that you'd want more edges  received from the DSP? Well, it might be the case that the DSP only sends messages after receiving something from the CIO. In the input test scenario we don't retransmit anything so it may be normal to see what you see.

So is it working now?

piltous, 2024-07-11:
Hello,

For now, still the same state:
- from DSP screen/control OK
- from App screen/control NOK. If I take control and change temp unit from C to F, the spa stops working with an error (E03 if I remember well...)

I made this new hardware check which looks good:

```
Start test. Seq begins with HIGH, then alters.

Sending on D1, receiving on D4
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D5, receiving on D7
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100
Sending on D4, receiving on D1
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D7, receiving on D5
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D0, receiving on D0
1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1-1- // 50 errors out of 100
End of test!
```

And from the logs, it looks good too

```
2024-07-10 19:26:59UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,10,0,1,14,FD,0,0,0,0
DSP-ESP:BC,3,1D,0,1,21,FD,0,0,0,0
ESP-CIO:BC,3,1D,0,1,21,FD,0,0,0,0
ESP-DSP:BB,3,10,0,1,14,FD,0,0,0,0
CIO msg count: 425 DSP msg count: 22
CIO msg quality: 100.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

```

So I don't understand why the app doesn't receive the right values (16 temp, 20 target by default) and why I can't take any control to the pump or change target temp.

I will certainly need to monitor the app logs from the serial to understand what is going on.

I'll let you know.

Thanks

visualapproach, 2024-07-11:
There are two target temperatures: one internal to the display which the ESP can not know, and one internal to the ESP which the display cannot know. When taking control, the ESP decides whether to heat or not based on its internal target.
You should not get errors though. E03 is an error that says too low ambient temperature or something along those lines. What if you just take control and don't switch units. Can you start the pump and such from the ui?

piltous, 2024-07-11:
thanks for your reply.
when I take control from the app, nothing works for me (pump, bubble, etc...)

visualapproach, 2024-07-11:
Are the clocks in the esp and your browser device synced?

piltous, 2024-07-11:
it looks ok

[image]

piltous, 2024-07-11:
additional tests..
if I enable the heating on the spa, the value goes from 16 to 27 on the app.
if I refreshed to much the app in the browser, the pump restart and the app crash and it cuts the pump.
my mqtt keeps disconnecting (when I have others mqtt devices fully stable)

dsp msg seems still unstable when I didn't change anything except a reboot

```
2024-07-11 12:20:50UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0
DSP-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0
ESP-CIO:BC,3,1B,0,1,1F,FD,0,0,0,0
ESP-DSP:BB,3,1B,0,1,1F,FD,0,0,0,0
CIO msg count: 7618 DSP msg count: 23
CIO msg quality: 99.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-07-11 12:20:52UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,1B,0,1,1F,F9,0,0,0,0
DSP-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0
ESP-CIO:BC,3,1B,0,1,1F,FD,0,0,0,0
ESP-DSP:BB,3,1B,0,1,1F,F9,0,0,0,0
CIO msg count: 7632 DSP msg count: 23
CIO msg quality: 99.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-07-11 12:20:52UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0
DSP-ESP:BC,3,1B,0,1,1F,FD,0,0,0,0
ESP-CIO:BC,3,1B,0,1,1F,FD,0,0,0,0
ESP-DSP:BB,3,1B,0,1,1F,FD,0,0,0,0
CIO msg count: 7633 DSP msg count: 23
CIO msg quality: 99.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

```

There was a reboot here

```
2024-07-11 12:44:36UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,1B,0,1,1F,FD,0,0,0,0
CIO msg count: 4635 DSP msg count: 0
CIO msg quality: 99.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)

2024-07-11 12:44:59UTC.  SW:2024-06-10-2016 
CIO-ESP:FB,3,1B,0,1,1F,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:FB,3,1B,0,1,1F,FD,0,0,0,0
CIO msg count: 4804 DSP msg count: 0
CIO msg quality: 99.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)

2024-07-11 12:44:59UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,1B,0,1,1F,FD,0,0,0,0
CIO msg count: 4805 DSP msg count: 0
CIO msg quality: 99.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)

2024-07-11 12:45:13UTC.  SW:2024-06-10-2016 
CIO-ESP:3B,3,1B,0,1,1F,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:3B,3,1B,0,1,1F,FD,0,0,0,0
CIO msg count: 4919 DSP msg count: 0
CIO msg quality: 99.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)

2024-07-11 12:45:13UTC.  SW:2024-06-10-2016 
CIO-ESP:BB,3,1B,0,1,1F,FD,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:BB,3,1B,0,1,1F,FD,0,0,0,0
CIO msg count: 4920 DSP msg count: 0
CIO msg quality: 99.000000% DSP msg quality: 0.000000% (Only useful in 4 wire pumps)

```

visualapproach, 2024-07-11:
SW:2024-06-10-2016 

That is not the latest firmware

visualapproach, 2024-07-11:
2024-07-09-2306 is the latest as of now

visualapproach, 2024-07-11:
I don't know what you mean. Bubbles shows 33 - where?

piltous, 2024-07-11:
When I take control from the app and enable bubble from it, the SPA screen shows 33 when it was 04 just before. 
That would mean a message is sent to the DSP instead the CIO from the app? Wrong wire?

piltous, 2024-07-14:
Hi there,

I have some quite good progress.

1st I mixed up CIO and DSP on the board... it explained the strange behavior I got. You can post me on the wall of fame and sorry for the wasted time...

So now, I'm on these pins:
CIO RX 5
CIO TX 2
DSP RX 4
DSP TX 7

Everything is on H1A (PCBv2) since it didn't work beside H2A, maybe some cap that I need to revert of polarity.

I can get the current water temp on the DSP and as well on the app.

The pump stop working after couple of minutes and then I get a E03 message.

I think I still have a communication issue but I don't know which pin and if I need to pass back the res/cap. There is a lot of combination possible so not easy...

```
2024-07-14 11:00:22UTC.  SW:2024-07-07-2211 
CIO-ESP:BC,3,16,0,1,1A,FD,0,0,0,0
DSP-ESP:BB,3,10,0,1,14,FD,0,0,0,0
ESP-CIO:BB,3,10,0,1,14,FD,0,0,0,0
ESP-DSP:BC,3,16,0,1,1A,FD,0,0,0,0
CIO msg count: 3 DSP msg count: 713
CIO msg quality: 100.000000% DSP msg quality: 99.000000%

Packet errors:0
bit0-bit error, bit1-too many bytes, bit2: too few bytes.

CIO CMD byte  0: 0
CIO CMD byte  1: 0
CIO CMD byte  2: 0
CIO CMD byte  3: 0
CIO CMD byte  4: 0
CIO CMD byte  5: 0
CIO CMD byte  6: 0
CIO CMD byte  7: 0
CIO CMD byte  8: 0
CIO CMD byte  9: 0
CIO CMD byte 10: 0
CIO CMD byte 11: 0
CIO CMD byte 12: 0
CIO CMD byte 13: 0
CIO CMD byte 14: 0
CIO CMD byte 15: 0
CIO CMD byte 16: 0

```

Any advances still welcome! :)

Have a nice day!

piltous, 2024-07-14:
I forgot to mention that I'm on the main branch firmware for now. I'll to switch to the dev one and 3.5.1 to see...

piltous, 2024-07-14:
I have almost everything working!!!

I just need, I think, now to understand some features for the take control.

This is my setup, for now all for now without cap/res
[image]

That was a bit a challenge (many hours of test... :p) and I still don't understand this pins layout...

Rgds

piltous, 2024-07-14:
Could maybe be added to the faq (but not sure...):
- current water temp working in app and display: full cio tx > dsp rx ok
- take control working: esp to cio rx ok

visualapproach, 2024-07-15:
>  I still don't understand this pins layout...

How can I help?

piltous, 2024-07-15:
> > I still don't understand this pins layout...
> 
> How can I help?

For me, all is ok, it works fine :) I just did not understand why some pins works and some others not. Issue is solved! Many thanks!!!

Maybe one more question not related... when I take control and start the pump, it works but the pump led on the egg remains off. Is it normal?

visualapproach, 2024-07-15:
Okay, good that it works!

Yes, it's normal. Unlike the 6-wire models, we can't control what the display is showing.

piltous, 2024-07-15:
Many thanks for your great project and support!
If you want me to share my setup somewhere, let me know ;)
The drink will come soon ;)

closed 2024-07-15 by piltous

visualapproach, 2024-07-15:
> Many thanks for your great project and support! If you want me to share my setup somewhere, let me know ;) The drink will come soon ;)

Sharing is caring - I think a new discussion would be a good place. Tagged "show and tell".
Looking forward to a drink! Thank you very much!

Fox49, 2024-07-17:
Hi @piltous 

I have the exact same problem on a 54154 pump. I tried the pins you mention but no luck.

How did you find out which pins were the good ones ?

Ty

piltous, 2024-07-17:
well @Fox49 , that was a bit painfull for me, I agree... I kept in mind the ESP in the middle of the info transfert. I mentionned some things above about that.
Exemple, If you can control the pump, that mens the esp cio tx is ok.
If you pump shows the right temp that means csio tx > esp > dsp rx is ok 
I also try old firmware without so much succes but that was nice to see "tx alive".

Also res/cap was in the loop when for me it was not required at the end...

Ping me in you tread you I could give some advice in this logic maybe....