visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #733: S100202 Pump

woodwai, 2024-06-17:
**Hardware (please complete the following information):**
 - S100202
 - 4 or 6 wires - Both
 - PCB type - V2B

**Software (please complete the following information):**
 - Firmware: 2023-11-19-1330

**Your message**

I opened up my Pump and it has 6 wires to the board, but only 4 to the display. I have the board working, and can control the pump, but I cannot get the display working. Does anyone have any ideas?

visualapproach, 2024-06-17:
Is it a model with WiFi already built in?

woodwai, 2024-06-17:
> Is it a model with WiFi already built in?

Yes it is.

visualapproach, 2024-06-17:
Never got it to work with those. Someone switched the control boards to an ordinary but other than that I think the only way is to tap into their api. You can google it. I think it's in a home assistant forum.

visualapproach, 2024-06-17:
What model did you select to make it work with the pump?

woodwai, 2024-06-17:
> What model did you select to make it work with the pump?

6 wire, 2021 (air)

I cant get the BestWay app to work. Guess ill try switching the control boards as well then.

visualapproach, 2024-06-17:
where do you live? (country)

woodwai, 2024-06-17:
> where do you live? (country)

UK

visualapproach, 2024-06-17:
Ok, I have some spare parts. Was thinking that you could have a display but I think the postage is quite expensive to UK and it's the old round version anyway, and I guess you have the square one.
If you can find a new display it would be interesting to see if you get it to work with different model setting on the display and pump. It *should* work

woodwai, 2024-06-18:
Ah thanks - I’ve ordered another control unit and will see what happens. Most bizzare.

IW

closed 2024-08-25 by github-actions[bot]

jrote1, 2024-11-01:
@woodwai did you ever get this to work?

comm, 2025-06-08:
@woodwai: to confirm: all pump controls worked fine but the display doesn't?

I'd also be interested to understand if you got it resolved and how?

Thanks.

woodwai, 2025-06-08:
Yes that was my finding with a WiFi control board model. I just purchased a non WiFi control model and it worked perfectly.