visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #738: [OTHER] Any success to get working with latest P05504

msggsm, 2024-06-27:
- Pump model HydroJet S200102, P05504, late 2022, so last version before WiFi-version, 302029000278
 - 6 wires
 - PCB V2
 - LLC red

Tried nearly all versions since 2021, including latest 4.3.1 from 06/2024

Neither got module nor display to work,
tried lot of versions and settings,
tried to bypass 5V directly to display (own power source for module),
with or without LLC,
resistors,
short cable.

Module never gets correct info,
display no lights (only little with C and F) and not responding to pressing buttons (with some versions just a chirp,
pressing buttons on webgui on reaction but sometimes „bubbles „ starts for some seconds but turms off immediately, even changing C and F dies not work.

So I read nearly every post here,
does anybody have an idea what to do or how to fix with this special pump?
(I had 3 modules, all working with predecessor pump but not with this version)

Any help appreciated,
regards Mark

visualapproach, 2024-06-27:
Post images of hardware settings page and the result of /inputs/ test. Maybe we can see if the wires is correct

msggsm, 2024-06-27:
Thank you very much,
here the requested images:
[image]
[image]

visualapproach, 2024-06-27:
Ok it seems to be right. Can you run the debug-on a while and post log.txt

msggsm, 2024-06-28:
2024-06-28 09:22:14UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 833
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

msggsm, 2024-06-28:
2024-06-28 10:09:59UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 457
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

visualapproach, 2024-06-28:
Run the debug mode for a few minutes, and press some buttons on the display meanwhile.

msggsm, 2024-06-28:
Ok, first two logfiles I only pressed buttons in webgui,
now I tried every button on display, but only power reacts in a chirp; nothing else, no lights.

2024-06-28 13:04:01UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115914
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:03UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115933
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:03UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115938
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:04UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115941
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:04UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:2,0,0,0,0,0,0,0,0,0,0
ESP-CIO:2,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115945
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:05UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115950
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:05UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:8,9,0,0,0,0,0,0,0,0,0
ESP-CIO:8,9,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115951
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:05UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 115955
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:06UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1,0,0,0,0,0,0,0,0,0,0
[... 256 lines cut ...]
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 116065
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:21UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1,0,0,0,0,0,0,0,0,0,0
ESP-CIO:1,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 116066
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:24UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 116102
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 13:04:24UTC.  SW:2024-06-10-2016 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP

visualapproach, 2024-06-28:
Ok, that's weird. Your "inputs" test shows activities from the CIO, but the debug logs shows that no messages at all is accepted by the ESP. So, The ESP is (on purpose) throwing away all messages that is not equal to preceeding message. This was made to avoid "spikes" in the data, where one bit moved or flipped for some reason. Maybe your CIO sends only one message for each state, and then is silent until something changes. That must be it! So, if that's the case you should replace these files and upload again:
(attached in the next post)

msggsm, 2024-06-28:
Ok, great to hear we could finally have a solution.
Only one file?
Can I upload via "file uploader"?

visualapproach, 2024-06-28:
There was only that one file I needed to change. Please report back how it goes.

visualapproach, 2024-06-28:
No, you need to copy the deflated file into your folder/code/lib/cio/... then recompile the code and upload again. Via OTA if you want. You only need to comment/uncomment a few lines in platformio.ini

msggsm, 2024-06-28:
Thanks a lot.
But I get this message:

Uploading: [============================================================] 100% Done...

15:50:46 [INFO]: Waiting for result...
15:51:46 [ERROR]: No Result!
*** [uploadfsota] Error 1

msggsm, 2024-06-28:
Restarted esp before and after upload, still same no reaction from display.

visualapproach, 2024-06-28:
You wanna upload the firmware, not the file system. The small right arrow at the bottom

msggsm, 2024-06-28:
Oh, sorry.
Now I get:

C:\Users\MSG\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\mmu_iram.cpp:29:34: note: '#pragma message: ICACHE size 16K'
   29 | #pragma message("ICACHE size 16K")
      |                                  ^
src\main.cpp:2074:10: fatal error: prometheus.txt: No such file or directory
 2074 | #include "prometheus.txt"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\nodemcuv2\src\main.cpp.o] Error 1

visualapproach, 2024-06-28:
Ok you have an older code base. That line is 2078 in my file. Here - extract all somewhere and open folder from vscode/platformio
[Code.zip](https://github.com/user-attachments/files/16031908/Code.zip)
[image]

edit as follow for OTA

[image]

Works fine here

[image]

visualapproach, 2024-06-28:
If it works, you shall see the FW version "test" on the home page.

visualapproach, 2024-06-28:
you must save platformio.ini and wait a few seconds before uploading, or you will get an error message sometime. VSCode needs to load something I guess.

msggsm, 2024-06-28:
Thank you very much, but unfortunately still the same.
Firmware is TEST, no reactions on button press.
[image]

visualapproach, 2024-06-28:
That's boring! If you want you can build and upload the file system also. It won't make it work but you'll have the latest version with more timers.
Anyhow, what if you make the debug session again?

msggsm, 2024-06-28:
Here it is:

2024-06-28 14:42:39UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8053
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:04UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8327
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:04UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8329
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:05UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8336
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:08UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8370
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:09UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1,0,0,0,0,0,0,0,0,0,0
ESP-CIO:1,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8371
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:09UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8374
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:09UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1,0,0,0,0,0,0,0,0,0,0
ESP-CIO:1,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8375
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:10UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8380
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:10UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8381
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:11UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8382
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:11UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8383
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 14:43:11UTC.  SW:test 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 8384
CIO msg qualit

visualapproach, 2024-06-28:
That's ridiculous. What's in the file hwcfg.json ?

msggsm, 2024-06-28:
At least from pins it should be ok?

{"cio":"2","dsp":"2","pcb":"custom","hasTempSensor":"0","pins":["1","2","5","6","4","3","7",""]}

visualapproach, 2024-06-28:
Ok hang on, I'll try to do another mod. Want to see what (th) the cio sends.

msggsm, 2024-06-28:
ok, very appreciated.
Will have to go to a customer now, will be back in aprox. 1h.

visualapproach, 2024-06-28:
Here we go. Throw away the last folder and unpack this instead. Then same procedure with uploading this firmware.
IT WILL NOT WORK - but enable /debug-on/  a while and post log.txt. I hope we can see the raw data from cio now (not all zeroes)

[Code.zip](https://github.com/user-attachments/files/16032762/Code.zip)

visualapproach, 2024-06-28:
FW version should now be "test2". Don't tell me I'm not creative 😄

msggsm, 2024-06-28:
Did that, here the logfile:

2024-06-28 18:08:10UTC.  SW:test2 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 1473
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

2024-06-28 18:08:14UTC.  SW:test2 
CIO-ESP:C0,0,0,E,0,DA,0,0,0,18,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:C0,0,0,E,0,DA,0,0,0,18,0
CIO msg count: 0 DSP msg count: 1514
CIO msg quality: 0.000000% DSP msg quality: 100.000000% (Only useful in 4 wire pumps)

visualapproach, 2024-06-28:
Good, so something is coming from cio at least. If you want to test more we can do these mods until we solve it or fail. 
For a start can you capture a longer log with some up down clicks

msggsm, 2024-06-28:
Thank you so much.
Yes, I would like to do more testing, if it's ok for you (even another day if you prefer).
I spent so much time and money into this (3 modules, LLC, resistors) and the pump is disassembled in our livingroom for months now; so my wife wants to kill me for that; and not being able to use the pool  ;-)

visualapproach, 2024-06-28:
Oh dear, then we´d better continue ;)
I'll think about it over some wine tonight, and will post here when I have something more to test. Cheers

msggsm, 2024-06-28:
Ok, very good idea.
Bottle of wine will be on me.
Just post here whenever you have something to test.
Cheers and have a nice evening.

visualapproach, 2024-06-29:
[Code.zip](https://github.com/user-attachments/files/16042068/Code.zip)

Cobbled together something here. Lets see if I can do anything with this. Install and run /inputs/
Follow the instructions :-)

msggsm, 2024-06-30:
Here comes the result:
waitmicros, gpio registers
378205170,35035
378205578,37035
378213235,33035
378213272,33005
378213288,33015
378213309,33025
378213329,33035
378213345,33005
378213366,33015
378213386,33005
378213403,33015
378213423,33005
378213440,33015
378213460,33005
378213476,33015
378213497,33025
378213513,33035
378213534,33005
378213550,33015
378213570,33035
378213591,33025
378213607,33035
378213628,33025
378213644,33035
378213660,33025
378213681,33035
378213697,33025
378213718,33035
378213734,33025
378213755,33035
378213771,33025
378213791,33035
378213808,33025
378213872,33035
378213893,33025
378213909,33035
378213966,33025
378213982,33035
378214003,33025
378214019,33035
378214040,33025
378214056,33035
378214076,33025
378214093,33035
378214113,33025
378214130,33035
378214146,33025
378214166,33035
378214183,33025
378214203,33035
378214220,33025
378214240,33035
378214277,37035
378223226,33035
378223263,33005
378223283,33015
378223299,33035
378223304,33025
378223320,33035
378223336,33015
378223341,33005
378223357,33015
378223378,33005
378223394,33015
378223414,33005
378223431,33015
378223451,33005
378223468,33015
[... 932 lines cut ...]
378345883,37015
378345919,37005
378345924,37015
378345940,33015
378345977,33005
378345993,33015
378346013,33005
378346034,33015
378346050,33005
378346071,33015
378346087,33005
378346108,33015
378346128,33005
378346144,33015
378346165,33005
378346181,33015
378346202,33025
378346218,33035
378346238,33025
378346255,33035
378346295,33005
378346312,33015
378346332,33005
378346349,33015
@msggsm: Cut and paste all above. Post on forum

visualapproach, 2024-06-30:
Good! First look tells me that it seems fine. Almost the same as my pump except it is faster. The CLK edges are on average 20 us apart in your pump, while they are 25 us apart in my pump. Maybe too fast for the device? I hope not. Maybe it actually works but I sent you a non working code for testing, can't remember right now. Have to investigate more, but now I need to assemble some IKEA furniture at my daughter's place. Call you back when I have something new for you to try.

msggsm, 2024-06-30:
Ok great, thank you.
Good luck with the IKEA furniture.
Will keep following here.

visualapproach, 2024-07-01:
[Code.zip](https://github.com/user-attachments/files/16051005/Code.zip)

Try this. Not too hopeful but we can squeeze out a bit more info on what is going on maybe.
Do the debug-on-off procedure please.
I sped up the overall speed of the program at the cost of memory but it seems okay.

msggsm, 2024-07-01:
Ok, here is log.txt.
Last 4 actions were 2x up and 2x down

2024-07-01 10:36:17UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 450
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:36:51UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 824
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:36:51UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 827
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:36:52UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 831
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:36:52UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 834
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:36:53UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:0,0,0,0,0,0,0,0,0,0,0
ESP-CIO:0,0,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 839
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:36:55UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 867
CIO msg quality: 0.000000% DSP msg quality: 100.000000%
[... 387 lines cut ...]
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 1056
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:37:17UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:8,9,0,0,0,0,0,0,0,0,0
ESP-CIO:8,9,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 1061
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

2024-07-01 10:37:18UTC.  SW:test3 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 1064
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

msggsm, 2024-07-01:
Do you need /inputs/ as well?

visualapproach, 2024-07-01:
Might as well.

msggsm, 2024-07-01:
waitmicros, gpio registers
765544308,35035
765544497,37035
765550357,33035
765550393,33005
765550411,33015
765550430,33025
765550448,33035
765550467,33015
765550470,33005
765550485,33015
765550507,33005
765550522,33015
765550543,33005
765550562,33015
765550580,33005
765550598,33015
765550617,33025
765550635,33035
765550654,33005
765550672,33015
765550693,33035
765550709,33025
765550727,33035
765550745,33025
765550764,33035
765550782,33025
765550801,33035
765550819,33025
765550837,33035
765550856,33025
765550874,33035
765550892,33025
765550911,33035
765550974,33025
765550993,33035
765551011,33025
765551030,33035
765551087,33025
765551102,33035
765551124,33025
765551139,33035
765551161,33025
765551176,33035
765551194,33025
765551213,33035
765551231,33025
765551249,33035
765551268,33025
765551286,33035
765551305,33025
765551323,33035
765551341,33025
765551360,33035
765551396,37035
765560341,33035
765560377,33005
765560395,33015
765560414,33025
765560432,33035
765560450,33015
765560454,33005
765560469,33015
765560491,33005
765560506,33015
765560527,33005
765560543,33015
765560564,33005
765560582,33015
765560601,33025
[... 931 lines cut ...]
765690580,33035
765690599,33025
765690617,33035
765690672,33025
765690690,33035
765690708,33025
765690727,33035
765690745,33025
765690764,33035
765690782,33025
765690800,33035
765690819,33025
765690837,33035
765690855,33025
765690874,33035
765690892,33025
765690911,33035
765690974,33025
765690993,33035
765691029,37035
765700349,33035
765700386,33005
765700404,33015
765700422,33025
@msggsm: Cut and paste all above. Post on forum

visualapproach, 2024-07-01:
Hm, I have another idea but things don't make sense to me. Now it says no bit/byte errors but it doesn't get any messages... I will make it dump raw data. Hang on a few hours (AFK)

visualapproach, 2024-07-01:
[Code.zip](https://github.com/user-attachments/files/16054179/Code.zip)

test4. same procedure again :-)

visualapproach, 2024-07-01:
but I don't need inputs test. Only debug. I hope

msggsm, 2024-07-01:
Ok, that gave a lot of feedback;
beginning from when I installed it to the buttonpresses right now:

2024-07-01 15:05:44UTC.  SW:test4 
CIO-ESP:0,0,0,0,0,0,0,0,0,0,0
DSP-ESP:1B,1B,0,0,0,0,0,0,0,0,0
ESP-CIO:1B,1B,0,0,0,0,0,0,0,0,0
ESP-DSP:0,0,0,0,0,0,0,0,0,0,0
CIO msg count: 0 DSP msg count: 266
CIO msg quality: 0.000000% DSP msg quality: 100.000000%

Packet error:0 bit0-bit error, bit1-too many bytes, bit2: too few bytes.

CIO CMD byte  0: C0
CIO CMD byte  1: 0
CIO CMD byte  2: 0
CIO CMD byte  3: B6
CIO CMD byte  4: 0
CIO CMD byte  5: CC
CIO CMD byte  6: 0
CIO CMD byte  7: 0
CIO CMD byte  8: 0
CIO CMD byte  9: 14
CIO CMD byte 10: 0
CIO CMD byte 11: 8F
CIO CMD byte 12: 42
CIO CMD byte 13: 42
CIO CMD byte 14: 42
CIO CMD byte 15: 42
CIO CMD byte 16: 42
CIO CMD byte 17: 42
CIO CMD byte 18: 42
CIO CMD byte 19: 42
CIO CMD byte 20: 42
CIO CMD byte 21: 42
CIO CMD byte 22: 5
CIO CMD byte 23: 40
CIO CMD byte 24: C0
CIO CMD byte 25: 0
CIO CMD byte 26: 0
CIO CMD byte 27: B6
CIO CMD byte 28: 0
CIO CMD byte 29: CC
CIO CMD byte 30: 0
CIO CMD byte 31: 0
CIO CMD byte 32: 0
CIO CMD byte 33: 14
CIO CMD byte 34: 0
CIO CMD byte 35: 8F
CIO CMD byte 36: 42
CIO CMD byte 37: 42
CIO CMD byte 38: 42
CIO CMD byte 39: 40
CIO CMD byte 40: C0
CIO CMD byte 41: 0
CIO CMD byte 42: 0
CIO CMD byte 43: B6
CIO CMD byte 44: 0
CIO CMD byte 45: CC
CIO CMD byte 46: 0
CIO CMD byte 47: 0
CIO CMD byte 48: 0
CIO CMD byte 49: 14
CIO CMD byte 50: 0
CIO CMD byte 51: 8F
CIO CMD byte 52: 42
CIO CMD byte 53: 42
CIO CMD byte 54: 42
CIO CMD byte 55: 42
CIO CMD byte 56: 42
[... 2276 lines cut ...]
CIO CMD byte 39: B6
CIO CMD byte 40: 0
CIO CMD byte 41: CC
CIO CMD byte 42: 0
CIO CMD byte 43: 0
CIO CMD byte 44: 0
CIO CMD byte 45: 4
CIO CMD byte 46: 0
CIO CMD byte 47: 88
CIO CMD byte 48: 42
CIO CMD byte 49: 42
CIO CMD byte 50: 42
CIO CMD byte 51: 42
CIO CMD byte 52: 42
CIO CMD byte 53: 42
CIO CMD byte 54: 42
CIO CMD byte 55: 0
CIO CMD byte 56: B6
CIO CMD byte 57: 0
CIO CMD byte 58: CC
CIO CMD byte 59: 0
CIO CMD byte 60: 0
CIO CMD byte 61: 0
CIO CMD byte 62: 4
CIO CMD byte 63: 0

visualapproach, 2024-07-01:
Ha! Now edit this line in lib/cio/cio_type1.h (from my memory, you see the exact name in vsc)

Static to work in switch/case.*/
        static const uint8_t DSP_CMD1_MODE6_11_7 = 0x01;

to 0x05 instead of 0x01

I think it will work after that

msggsm, 2024-07-01:
You're the man!
Displaybuttons and webgui-buttons are now -mostly- working.
I'm not sure if this 'test4' was a fully working code?
Temperature display stays off, powerbutton (long or short) press leads to a restart of whole display,
C and F indicators work as well as the bubbles button and it's light (as pump is still standalone didn't dare to test other buttons),
up and down give sound but I do have no feedback (no display and nothing in webgui).
But I think now these shall be small issues, hopefully?

visualapproach, 2024-07-01:
Yay 😀 
That's nice to hear! The code is a full version, just tweaked for debugging/hacking. Now I guess that since they (bestway) changed one thing in the data headers (maybe due to new chips) they may as well changed the data itself. It's not difficult to find out but somewhat tedious. So I will give you more instructions to sort it out. Maybe tomorrow. I think it will be easier if we switch to email conversation. Send me a mail at info *a*t* visualapproach.se if you want.

visualapproach, 2024-07-02:
> Temperature display stays off, powerbutton (long or short) press leads to a restart of whole display,

Is the ESP restarted too when you press POWER button?
Does the display show scrolling text at ESP start up?

visualapproach, 2024-07-03:
[Code.zip](https://github.com/user-attachments/files/16082232/Code.zip)
FW version "test5"

david47800, 2024-07-03:
Hello @visualapproach I hope you can help me too,
I've read everything on the subject here, but I can't get any further.

I have a pump as described here with S200102 P05504
a V1 circuit board 
with the firmware: 2024-06-17-1334

I have had the V1 board for 1-2 years and had it running with a 4 wire pump.

But now I have the pump listed above and I can't get it to work.
I'm having trouble with the wiring and don't know if the V1 will even work with the pump?

In addition, my English is not the best :-) 

please, can you help me.

Here are also photos of the settings and the circuit board 
[image]
[image]
[image]
[image]
[image]

visualapproach, 2024-07-03:
@msggsm and I have worked out a solution to his model. I will release a new version asap so stay tuned.

visualapproach, 2024-07-04:
New code available in development branch (v4.4.0-beta)

david47800, 2024-07-04:
Could the new version also be suitable for me? Does it also fit the V1?

visualapproach, 2024-07-04:
Yes there is a few changes. I'd try it

david47800, 2024-07-04:
WOW  it works 💯  , @visualapproach you are a genius. many many thanks

visualapproach, 2024-07-04:
Nice❗️

closed 2024-09-10 by github-actions[bot]