visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #739: [BUG] Screen not

fungio, 2024-06-28:
**Hardware (please complete the following information):**
- Pump model S100101
- 6 wires
- PCB - V2B - picture added
- Level shifter type, - the blue one

**Software (please complete the following information):**
- Version - the latest

[image]
[image]

Hi all, when I plug the PCB the display does not work, only *F light is on and no buttons are working. It makes a light bird chirp sound. As you can see in the picture I tried H1A & H1B but none worked. 

Any idea ?

visualapproach, 2024-06-30:
Have you set the hardware config? saved and restarted?

fungio, 2024-06-30:
I got it working. I don't know what I made to make it works but it does now. Thank you

fungio, 2024-07-04:
I close the issue since everything is working perfectly now

closed 2024-07-04 by fungio