visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #742: 4-Wire 54138 Watchdog timer reboot causing E13

DeanozUK, 2024-06-30:
**Describe the bug**
After install board seems to be stable with no issues until I select 4wire 54138 on CIO and DSP and reboot. While the board is idle (not even connected to pump) it will periodically reboot and when checking reason it says watchdog

**To Reproduce**
Steps to reproduce the behavior:
1. After install goto hardware config and select 54138 for CIO and DSP and also select v2b
2. Reboot ESP and setup constant ping to device
3. Goto web page home page (maybe change temp)
4. page might drop out. I left it idle for sometime while on monitor via usb and I will post the log

**Expected behavior**
Expect the unit to be stable when idle and when connected to unit

**Screenshots**
[ESP WDT Error.txt](https://github.com/user-attachments/files/16046331/ESP.WDT.Error.txt)

**Hardware (please complete the following information):**
 - Pump model, 54138
 - 4 wire
 - PCB type v2b
 - Level shifter type, Red one

**Software (please complete the following information):**
 - Firmware: 2024-06-17-1334

**Additional context**
When its connected to the pump after startup I will just leave it without touching any buttons. only load the web page and after short amount of time it will E13.
Also tried 2 other ESPs and the beta version to no effect.

Below is the result from USB Monitor.

Writing cmdq.json (I was changing temp on web page)
Done!
Writing cmdq.json
Done!
Writing cmdq.json
Done!
Writing cmdq.json
Done!
Writing cmdq.json
Done!
Writing cmdq.json
Done!

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x40234f9c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register       
  epc1=0x40234f9c in AsyncTCPbuffer::write(unsigned char const*, unsigned int) at ??:?

>>>stack>>>

ctx: cont
sp: 3fff1d40 end: 3fff2040 offset: 0160
3fff1ea0:  00000000 00000066 89374bc6 07f64dd6
3fff1eb0:  4010a85b 0006e2db 00000000 40109bdc
3fff1ec0:  4010a85b 00000066 0000012f 4021f402
3fff1ed0:  0006da50 0000012f 40284928 00000004
3fff1ee0:  40284930 00000004 40284938 40109bac
3fff1ef0:  4010a72c 40109bdc 00000195 4021f6d1
3fff1f00:  00000001 4028467c 00000005 40284684
3fff1f10:  95017e81 00000000 00000000 00000000  
3fff1f20:  00000000 40284660 00000004 402846a4
3fff1f30:  00000000 00000004 00000001 00000001
3fff1f40:  00000001 00000001 00000000 402846cc
3fff1f50:  00000006 402846d4 00000004 4010a72c
[... 224 lines cut ...]
Millis: 91 @ line: 41
startWiFi() @ millis: 340
Setting static IP
WiFi > using static IP t(�?␏ on gateway 10.10.30.1
WiFi > Using WiFiManager Config Portal
*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Using last saved values, should be faster
got IP: 10.10.30.97
start NTP
WS IRamheap 9296
IRamheap 8048
startmqtt
Failed to read mqtt.json. Using defaults.
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 10.10.30.97
*WM: freeing allocated params!
End of setup()
Millis: 3494 @ line: 73
33176

visualapproach, 2024-07-01:
Ok, haven't had time to look at this yet. Call you back

visualapproach, 2024-07-04:
I can't reprocude, but see if the new beta version helps.

DeanozUK, 2024-07-04:
Ok, Ill give it ago now. Do the errors mean anything to you?

visualapproach, 2024-07-04:
Oh, now I get web socket disconnections. With newest code too, so I will have too investigate more

DeanozUK, 2024-07-04:
happens quicker if you are on the home page and turn the temp up from default 20, then refresh and its rebooted and the temps back at default 20 again. (ESP is on bench and not connected to pump atm).

DeanozUK, 2024-07-04:
I sent you a coffee to help while to look into it ;)

visualapproach, 2024-07-04:
That very nice of you! Thanks!!
I uploaded a new version to development branch. It's kind of a band aid but it seems to work here. Plz report how it goes

DeanozUK, 2024-07-04:
Well, I've tried on 2x ESPs (I also completely erased them and started from scratch to rule out any oddities) and all seems fine when setting up via AP etc. get web page nice. 

Once on my Wifi I then configure in hardware to 54138 on DSP and CIO and then select v2b and reboot. It then it don't come back online. I'm constantly pinging the unit and all I see after the reboot is 1 high ping of say 380ms then times out forever.

Below is a serial monitor from when the web page loads as (default) 6wire and I got into menu and setup 4 wire 54138 etc then reboot, seems to look ok on monitor but nothing comes in after the line 'pm open,type:2 0'

Need anymore info? 

[WS][0][sendFrame] ------- send message frame -------
[WS][0][sendFrame] fin: 1 opCode: 1 mask: 0 length: 296 headerToPayload: 0
[WS][0][sendFrame] text: {"CONTENT":"STATES","TIME":1720111476,"LCK":0,"PWR":0,"UNT":0,"AIR":0,"GRN":0,"RED":0,"FLT":0,"CH1":97,"CH2":98,"CH3":99,"HJT":0,"BRT":7,"ERR":0,"GOD":0,"TGT":0,"TMP":25,"VTMC":-2.997020721,"VTMF":26.60536194,"AMBC":20,"AMBF":68,"AMB":68,"VTM":26.60536194,"TGTF":0,"TMPF":25,"TGTC":-18,"TMPC":-4}
[WS][0][sendFrame] pack to one TCP package...
[write] n: 300 t: 48724
[WS][0][sendFrame] sending Frame Done (4732us).
[WS][0][sendFrame] ------- send message frame -------
[WS][0][sendFrame] fin: 1 opCode: 1 mask: 0 length: 343 headerToPayload: 0
[WS][0][sendFrame] text: {"CONTENT":"TIMES","TIME":1720111476,"CLTIME":0,"FREP":0,"FRIN":0,"FCLE":0,"UPTIME":48,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"COST":0.00002573,"FREPI":60,"FRINI":7,"FCLEI":20,"CLINT":14,"KWH":0.000026667,"KWHD":0.000026951,"WATT":2,"T2R":-2,"RS":"Ready","DBG":" 0 0 0 0 0 0 0 0 0 0 0 cio msgs:0 ||  255 255 0 0 0 0 0dsp msgs:0"}
[WS][0][sendFrame] pack to one TCP package...
[write] n: 347 t: 48811
[WS][0][sendFrame] sending Frame Done (3408us).
[WS][0][sendFrame] ------- send message frame -------
[WS][0][sendFrame] fin: 1 opCode: 1 mask: 0 length: 165 headerToPayload: 0
[WS][0][sendFrame] text: {"CONTENT":"OTHER","MQTT":-1,"HASJETS":false,"HASGOD":false,"MODEL":"PRE2021","RSSI":-39,"IP":"10.10.30.18","SSID":"Matrix_IoT","FW":"2024-07-04-1647","loopfq":2073}
[WS][0][sendFrame] pack to one TCP package...
[write] n: 169 t: 48871
[WS][0][sendFrame] sending Frame Done (3654us).
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 8
[WS][0][handleWebsocket] mask: 1 payloadLen: 2
[WS][0][handleWebsocketWaitFor] size: 6 cWsRXsize: 2
[WS][0][handleWebsocketWaitFor][readCb] size: 6 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 8
[WS][0][handleWebsocket] mask: 1 payloadLen: 2
[WS][0][handleWebsocket] get ask for close. Code: 1001

[WS][0][handleWebsocket] clientDisconnect code: 1000
[WS][0][sendFrame] ------- send message frame -------
[WS][0][sendFrame] fin: 1 opCode: 8 mask: 0 length: 2 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[write] n: 4 t: 50697
[WS][0][sendFrame] sending Frame Done (3464us).
[WS-Server][0] client disconnected.
[WS-Server][0] Disconnect client
File size: 3796
HTTP > file sent: /hwconfig.html.gz (3796 bytes)
File size: 3088
HTTP > file sent: /main.css.gz (3088 bytes)
File size: 562
HTTP > file sent: /darkmode.js.gz (562 bytes)
File size: 2953
HTTP > file sent: /visualapproach.png (2953 bytes)
Free mem before stop: 20296
stopping cio
del cio
stopping dsp
del dsp
detaching
stopping mqtt
stopping server
stopping ws
stopping FS
end stopall
Free mem after stop: 28288
ESP restart ...
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
disconnected
4#-a|(␕␗�P␚v␛X  �~h~u�Ph 4�r(␑,+,*#A␑␑@�*␕4nzAl )z�~,j �r(␗%␑␙@#A.*�*,4nzAl )z�~,PZ8�~,PZ�Pl␄4/'�rf cal sector: 1020
freq trace enable 0
rf[112] : 0�
SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386

Start
Millis: 76 @ line: 42
startWiFi() @ millis: 183
WiFi > Using WiFiManager Config Portal
*WM:
*WM: AutoConnect
fpm close 1
mode : sta(a8:48:fa:c1:74:62)
add if0
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Using last saved values, should be faster
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with Matrix_IoT, channel 6
dhcp client start...
ip:10.10.30.18,mask:255.255.255.0,gw:10.10.30.1
got IP: 10.10.30.18
start NTP
WS IRamheap 24960
[WS-Server] Websocket Version: 2.4.2
[WS-Server] Server Started.
IRamheap 23632
startmqtt
Failed to read mqtt.json. Using defaults.
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 10.10.30.18
*WM: freeing allocated params!
End of setup()
Millis: 3291 @ line: 74
22104
pm open,type:2 0

visualapproach, 2024-07-04:
It looks normal. I don't know how well it reacts to pinging but you will (should) get more serial output when reloading a page. It may also be a problem if using several devices with the same hostname. If so edit config.h

DeanozUK, 2024-07-04:
Normally it pings fine, no issues up until I select 4wire and reboot. On the last version it would ping all day until I changed a setting on web page and it would timeout while it watchdog rebooted.I'm trying to connect via hostname but mainly IP as more reliable but both just timeout on webpage, it's odd. If I default unit it works fine, it seems to be as soon as select 4wire and reboot.If I reboot the esp it just briefly pings once but high latency.I do IT as a job so I'm competent in that regard, Im gonna keep trying somethings and let you know.

DeanozUK, 2024-07-04:
[image]

DeanozUK, 2024-07-04:
Just got this when i managed to get a web page briefly

Reconnecting to COM3
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

         Connected!
rf cal sector: 1020
freq trace enable 0
rf[112] : 00
SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386

Start
Millis: 92 @ line: 42
startWiFi() @ millis: 279
WiFi > Using WiFiManager Config Portal
*WM:
*WM: AutoConnect
fpm close 1
mode : sta(ec:fa:bc:b9:b2:ba)
add if0
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Using last saved values, should be faster
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with Matrix_IoT, channel 6
dhcp client start...
ip:10.10.30.97,mask:255.255.255.0,gw:10.10.30.1
got IP: 10.10.30.97
start NTP
WS IRamheap 24944
[WS-Server] Websocket Version: 2.4.2
[WS-Server] Server Started.
IRamheap 23616
startmqtt
Failed to read mqtt.json. Using defaults.
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 10.10.30.97
*WM: freeing allocated params!
End of setup()
Millis: 3425 @ line: 74
22088
pm open,type:2 0
File size: 6736
HTTP > file sent: /index.html.gz (6736 bytes)
File size: 3088
HTTP > file sent: /main.css.gz (3088 bytes)
File size: 562
HTTP > file sent: /darkmode.js.gz (562 bytes)
File size: 6736
HTTP > file sent: /index.html.gz (6736 bytes)
File size: 3088
HTTP > file sent: /main.css.gz (3088 bytes)
File size: 562
HTTP > file sent: /darkmode.js.gz (562 bytes)
File size: 2953
HTTP > file sent: /visualapproach.png (2953 bytes)
[WS-Server][0] new client from 10.10.30.134
[WS-Server][0][handleHeader] RX: GET / HTTP/1.1
[WS-Server][0][handleHeader] RX: Host: 10.10.30.97:81
[... 61 lines cut ...]
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 8
[WS][0][handleWebsocket] mask: 1 payloadLen: 2
[WS][0][handleWebsocketWaitFor] size: 6 cWsRXsize: 2
[WS][0][handleWebsocketWaitFor][readCb] size: 6 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 8
[WS][0][handleWebsocket] mask: 1 payloadLen: 2
[WS][0][handleWebsocket] get ask for close. Code: 1001

[WS][0][handleWebsocket] clientDisconnect code: 1000
[WS][0][sendFrame] ------- send message frame -------
[WS][0][sendFrame] fin: 1 opCode: 8 mask: 0 length: 2 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[write] n: 4 t: 42883
[WS][0][sendFrame] sending Frame Done (3147us).
[WS-Server][0] client disconnected.
[WS-Server][0] Disconnect client
File size: 4625
HTTP > file sent: /config.html.gz (4625 bytes)
File size: 3088
HTTP > file sent: /main.css.gz (3088 bytes)
File size: 562
HTTP > file sent: /darkmode.js.gz (562 bytes)

visualapproach, 2024-07-04:
The 4-wire models uses more more memory, leaving less for webserver, fileserver etc. It seems that reveals a bug in one or more of them. The strange thing is that it usually works when enabling debugging to serial. The 8266 webserver is allegedly bad at handling several simultaneous connections which may be why it works better with debug output, since it takes a little bit time to do the prints. But it also takes more memory which would make it worse. I struggled with this and implemented a second heap to gain a few kB but in the latest versions I reverted back to get more speed which I thought I needed for a new model. It worked fine on 6-wire models after reducing the number of page loads. Obviously not true for your model. 
It's possible that a single page load (without css, favicon, manifests and such) would work but I don't know for sure.

DeanozUK, 2024-07-04:
After a bit more testing, from default all is good. I select  4wire cio and reboot and all is good. I then select v2b and reboot, still all good. Lastly I selected the first 4wire option on the page and bam, lots of timeouts, high latency and web page is very intermittent. I haven't tried the other options yet except mine but I'm thinking it will have same result.

It used to work on the old firmware couple yrs back but did used to reboot once every day or 2. I'm guessing there's a lot more to this new all in 1 firmware that's affecting the 4wire.

All these tests I've been doing are on the bench, not connected to pump.

I noticed when 6wire default is selected the monitor picks up loads of traffic constantly spamming, unlike the 4wire. This is normal I presume.

DeanozUK, 2024-07-05:
> It's possible that a single page load (without css, favicon, manifests and such) would work but I don't know for sure.

Can you knockup a barebone test firmware so I can try for you? 
I take it a D1 mini will have the same issue as based off same chipset?

visualapproach, 2024-07-05:
D1 mini is the same chip so it would apply the same. I am wrestling with this issue and can't really make a barebone version because it only appears with everything stuffed in there. Thanks for the offer, I'll be in touch when I need help!

DeanozUK, 2024-07-06:
Hi VA,

Right, Ive been doing some more testing with the last firmware and the new one you did today. I hope I've captured as much info as possible to help you with this issue, so please ignore anything you already know.

1st the other firmware where I said it was constantly dropping off 95% pings and unable to load a web page. I decided to hook it up to the pump and it behaved fairly different to my surprise, the pings where alot more reliable (maybe it wanted to see data from DSP?) now pings were 10% dropping and web page would keep dropping but the pump was happy. I experienced very little E13s if i left it alone. when i would say spam the temp up button it would drop and there was chance it dropped long enough that it would E13 but no watchdog reboot anymore. I say it was approx 90% reliable.

2nd, the new firmware 1400 today was uploaded and as you said I had some issues getting to fully load the webpage with all the graphics but eventually I got it. after this it does seem to only drop say 5% pings but better than before. The other issue however is that the signal coming out of the ESP is more unreliable now, causing alot more E13s even if I boot system and don't touch anything, it will always error within 3mins. control from app seems better but the signal drops are the issue causing the E13 now and not the watchdog.

Below is some links to 5 videos I did to hope show what I mean. I have an oscilloscope set up on the 5v side of the LLC (Blue trace is DSP signal going into ESP and the Red trace is the Signal coming out of the ESP to the CIO)

Is this normal that the DSP signal is 1v>5v and the ESP is generating 0v>5v. (note, I've tried the resistors and the ESP just doesn't output at all, not sure if too much pull so removed them).

Link to 5x videos
[https://photos.app.goo.gl/Na4vdFmKrieDBbP37](url)

DeanozUK, 2024-07-06:
Ive just been monitoring the waveforms and Ive noticed a pattern. Every now and then the DSP waveform jumps from 1-5v to 2.3v-5v and this causes the ESP to stop outputting. 

Not sure if helps but looks like an issue Ive finally catch in the act. this Ive seen accross all versions, would explain why with resistors the ESP wouldnt output.

2x screenshots attached. Blue is DSP sig. Red is output of ESP
[image]
[image]

visualapproach, 2024-07-07:
Good morning, I've read your posts @DeanozUK and we have 2 separate issues.
1 - web pages not loading correctly.
2 - Signal problems in your setup.

1 It seems to be caused by too little free memory available. I say "seems" here because I can't see any out of memory errors when debugging. The ESP does not fail gracefully by design as I understand it. So trial and error. In the last update I made, I got enough free memory to make it work in most cases. Now I have just found that I can kick out the wifimanager library and free up several more kB, so I hope this issue will disappear in next release.

2 It's weird that the low states is changing with time! I mean, the pull resistors don't change value. Measurement issue? Does it behave the same when the ESP is not in curcuit? If the voltage doesn't cross the LOW threshold for the ESP it will not react/answer and the DSP will go E13. I'll think about it.

visualapproach, 2024-07-07:
@DeanozUK How long delay can the dsp take before panicking would you say?

visualapproach, 2024-07-07:
[Code.zip](https://github.com/user-attachments/files/16118692/Code.zip)

See if this code works without E13.
First - enter wifi credentials on the ESP network UI page. Otherwise you need to use my undocumented new AP :-)
Then open this extracted folder in platformio and upload the new firmware.

Max time between transmissions is 2000 ms even if we miss a reception or more.

Note - this is an experimental version with a different hostname. Only a thing if you use layzspa.local instead of IP.

DeanozUK, 2024-07-07:
Hi VA,

I agree that there is 2 issues here.

After more testing I did indeed put the pump back to default wiring and did some capturing and the wave forms are indeed 0v-5v and stable. This leads me to believe its the input impedance. So I tried a 10k pull down and same result, tried 1k and that just reduced the upper to 4v so I removed them. 

Then looked into the ESP pin outs on datasheets and rewired and configured the pins as custom as reading up it says 1,2,5 and 6 are best for data and not used for anythg else.
DSP TX D4 changed to D1
DSP RX D3 changed to D6
CIO TX D5
CIO RX D2
This to my surprise this helped as now the DSP input is reading a better 0.6v-5v and seems much more stable. I left it running all night and changed some settings on the pump and all seems good. However I did get few issues when changing via webpage but that leads us onto the other issue (point 1 in ur reply) 
anyways with ESP busy/watchdog etc. But its a good start.

> @DeanozUK How long delay can the dsp take before panicking would you say?

Testing shows approx >2-3secs. So max times between TX 2000ms should be good.

I will get on with installing this release and let you know.

visualapproach, 2024-07-08:
Hi! There is a new version in development branch now. It is pretty much the same version you had, but with some general improvements.

deanZZZZZ, 2024-07-08:
2 hour without E13 :)

visualapproach, 2024-07-08:
Crossing my fingers 🤞🏻

visualapproach, 2024-07-08:
@deanZZZZZ @DeanozUK same person?

deanZZZZZ, 2024-07-08:
> @deanZZZZZ @DeanozUK same person?

hehe no

DeanozUK, 2024-07-08:
Hi VA,Haha, we are different people 😉 I've been doing some testing and it's much better but I have had some e13 issues but it is deffo better. I also had to connect to testssid and Wireshark the IP as it lost the configured, but I got it. I've just got in from work and it's alarming e13 so need to do some checks to try and pinpoint it. I've got some waveforms from first tests last night that showed a 5.5s gap then 3rapid bursts of data (poss trying to catchup) but need to test more.Shall I just update to latest release or carry on with this one as mostly the same?Dean

deanZZZZZ, 2024-07-08:
[image]

visualapproach, 2024-07-08:
Note there is a new way to enter the wifi credentials at first install. It's described in the manual.

visualapproach, 2024-07-08:
> <img alt="image" width="330" src="https://private-user-images.githubusercontent.com/95451596/346639742-7341ca52-4ace-4994-87b3-e463f3458afb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA0NTU3NDQsIm5iZiI6MTcyMDQ1NTQ0NCwicGF0aCI6Ii85NTQ1MTU5Ni8zNDY2Mzk3NDItNzM0MWNhNTItNGFjZS00OTk0LTg3YjMtZTQ2M2YzNDU4YWZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA4VDE2MTcyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjNjFhN2E4NGU5YzRhOWFjOGFmZmVlNThkNDdjODhlZmRjYmMxMTVlM2Q4MGU3N2RhY2Q2NDI1OGE4YjczNzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4OyhX1kO9Oz_q0cGCAxqEAvnHbKcOs2UEfzG1xlbt9w">

Go go go!

deanZZZZZ, 2024-07-09:
didn't make it through the evening :(
[image]

visualapproach, 2024-07-09:
Do you know if something special happened 23:20? Restart of router or scheduled events on the esp?

visualapproach, 2024-07-09:
What model are you on?

deanZZZZZ, 2024-07-09:
no scheduled events on the esp
for the router. I wouldn't know...nothing is set for restart
blue LL with caps and resistor (100nf on 3.3v and 100nf 5v)

[image]

visualapproach, 2024-07-09:
I can confirm SW WDT when restarting my router. Will look at it

deanZZZZZ, 2024-07-09:
I tested it.. every time I "kick" from wifi the watchdog starts

visualapproach, 2024-07-09:
Yes, but I think it's fixable.

DeanozUK, 2024-07-09:
I also can confirm that I had 1 WD kick and on checking on my router it did disconnect and cause E13
[image]
[image]

Other than a few difficulties getting it online its been very stable, even with scheduled tasks on ESP in the night. Ive been testing this morning and still good atm :)

Now with the 4wire getting much loving can I make some observations about my model (BS54138 Hawaii Hydrojet) that are different on the ESP.

PUMP - Can use jets + bubble together.                     ESP - toggles other off and vice versa
PUMP - Can use Filter pump + bubbles together.       ESP - toggles other off and vice versa
PUMP - Bubbles + Filter pump +1 stage heater             ESP - toggle other off not sure on heater

I've always had issues with my 4wire when I started 3yrs ago, and even gave up for abit and now back working on it with you, Its seems much more stable well done VA for all your hard work!

DeanozUK, 2024-07-09:
Right, Ive had another WD reset but this one is interesting and not sure related, see what you think.

I had some schedules setup, and the last one of the day was turn pump off @19:00 but it seems it WD rebooted 3seconds after, and confirmed drop off wifi too. see screenshots

[image]

[image]

If only the ESP would crash and reboot inside 2secs it wouldn't cause E13, then I could live with it :D

visualapproach, 2024-07-09:
I fixed the WDT reset on Wi-Fi reconnection now. At least I get no reboot when I restart my routers... Hope it work for you guys to.
You are welcome to test the updated dev branch.

visualapproach, 2024-07-09:
> I also can confirm that I had 1 WD kick and on checking on my router it did disconnect and cause E13 [...]

I'll process this post later. Had to prioritize the WDT issue. Thanks

DeanozUK, 2024-07-09:
> I'll process this post later. Had to prioritize the WDT issue. Thanks

Not a problem, would you rather I create a feature request to keep it separate from this issue?

DeanozUK, 2024-07-09:
> I fixed the WDT reset on Wi-Fi reconnection now. At least I get no reboot when I restart my routers... Hope it work for you guys to. You are welcome to test the updated dev branch.

I will install it now and come back to you tomoz night to see how it goes. 👍

visualapproach, 2024-07-09:
> > I'll process this post later. Had to prioritize the WDT issue. Thanks
> 
> Not a problem, would you rather I create a feature request to keep it separate from this issue?

Nah, not necessary, I just copied to my onenote .-)

DeanozUK, 2024-07-11:
Still looking good here 👍 I will keep testing it over weekend, I will also test with default v2b pins to confirm stability.

I Have had a strange odd thing happen 3 times were the heater option kept turning off on me. Cant pin point it but It could be because I had open on 2 browsers?

deanZZZZZ, 2024-07-11:
development_v4

{"BOOTINFO":"Software/System restart 2024-07-11 06:57:02"}
{"BOOTINFO":"Software Watchdog 2024-07-11 07:01:38"}
{"BOOTINFO":"Software Watchdog 2024-07-11 07:04:20"}

visualapproach, 2024-07-11:
@deanZZZZZ you on the dev version?
Did you get the restarts without restarting your WiFi router?

deanZZZZZ, 2024-07-11:
yes
{"BOOTINFO":"External System 2024-07-10 10:28:31"}
{"BOOTINFO":"Software Watchdog 2024-07-10 12:46:06"}
{"BOOTINFO":"Software Watchdog 2024-07-10 12:56:30"}
{"BOOTINFO":"Software Watchdog 2024-07-10 12:57:34"}
{"BOOTINFO":"Software Watchdog 2024-07-10 12:59:11"}
{"BOOTINFO":"Software Watchdog 2024-07-10 12:59:43"}
{"BOOTINFO":"External System 2024-07-10 13:00:22"}
{"BOOTINFO":"Software Watchdog 2024-07-10 13:26:05"}
{"BOOTINFO":"Software Watchdog 2024-07-10 13:28:40"}
{"BOOTINFO":"Software Watchdog 2024-07-10 20:58:20"}
{"BOOTINFO":"Software Watchdog 2024-07-10 23:29:10"}
{"BOOTINFO":"Software Watchdog 2024-07-11 06:34:21"}
{"BOOTINFO":"Exception 2024-07-11 06:55:50"}
{"BOOTINFO":"Software/System restart 2024-07-11 06:57:02"}
{"BOOTINFO":"Software Watchdog 2024-07-11 07:01:38"}
{"BOOTINFO":"Software Watchdog 2024-07-11 07:04:20"}
{"BOOTINFO":"Software/System restart 2024-07-11 07:07:21"}
{"BOOTINFO":"External System 2024-07-11 07:10:42"}

visualapproach, 2024-07-11:
Ok. I'll see if I can do anything about it

DeanozUK, 2024-07-11:
Just a note on seeing deanzzzzz reply, I just remoted in  on my router I forced a disconnect from wifi for the ESP and blocked it from reconnecting for 1min then unblocked it and it reconnected fine. Checked esp and no WD reboot. It just reconnected with no issues.

visualapproach, 2024-07-11:
Yes, @deanZZZZZ just to rule out some things, does it make a difference if you make a clean build (platformio command) and also restart you router?

visualapproach, 2024-07-11:
> > I'll process this post later. Had to prioritize the WDT issue. Thanks
> 
> Not a problem, would you rather I create a feature request to keep it separate from this issue?

You should be able to just switch to model NO54173. It has the features you describe and should work.

deanZZZZZ, 2024-07-12:
> Yes, @deanZZZZZ just to rule out some things, does it make a difference if you make a clean build (platformio command) and also restart you router?

now I tried full clean
disconnected the wifi 3x and it works fine :)
... I'll let you know how it goes ...

deanZZZZZ, 2024-07-13:
{"BOOTINFO":"External System 2024-07-12 11:36:23"}
{"BOOTINFO":"Software Watchdog 2024-07-12 23:39:05"}
{"BOOTINFO":"Software Watchdog 2024-07-12 23:41:50"}

visualapproach, 2024-07-13:
@deanZZZZZ ok. In development branch you could turn on logging to file by editing platformio.ini and upload:
```
    ; -DBWC_DEBUGGING=BWC_DEBUG_OUTPUT_OFF
    ; -DBWC_DEBUGGING=BWC_DEBUG_OUTPUT_SERIAL
    -DBWC_DEBUGGING=BWC_DEBUG_OUTPUT_FILE
```
This will print the debug text to file and in the long run fill up the flash memory but it should be fine to run until you get a sw wdt. It is not completely finished yet but maybe you can see what is happening before the reset.

Or wait until I have more time to look at it.

DeanozUK, 2024-07-13:
Ive had some sw wdt resets too now, not sure why all of a sudden.

{"BOOTINFO":"Software/System restart 2024-07-09 22:04:08"}
{"BOOTINFO":"Software Watchdog 2024-07-11 18:24:17"}
{"BOOTINFO":"Software/System restart 2024-07-11 21:17:06"}
{"BOOTINFO":"Software Watchdog 2024-07-12 07:53:19"}
{"BOOTINFO":"Software Watchdog 2024-07-13 15:58:23"}
{"BOOTINFO":"Power On 2024-07-13 17:22:30"}
{"BOOTINFO":"Software Watchdog 2024-07-13 19:55:56"}

I will looking into it tomoz when I get time.

Note: I tried the 54173 options and they do work better on web gui but I noticed that the buttons just beeped when pressed on pump and didn't do anythg. Ive switched back to 54138 again now.

visualapproach, 2024-07-15:
> Note: I tried the 54173 options and they do work better on web gui but I noticed that the buttons just beeped when pressed on pump and didn't do anythg. Ive switched back to 54138 again now.

@DeanozUK that is very strange! The only difference is that 173 allows more states, like air and jets on at the same time. Did you swap on both the DSP and CIO setting?

visualapproach, 2024-07-15:
@deanZZZZZ @DeanozUK 
I have uploaded a new version to development branch. See if it helps with the resets. I frequently give back CPU to the OS, which would in theory help, if the problem is that user code takes too long. Unless I see the crash message it's really difficult to know exactly which part of the code is taking too much time. But if it works it works. There is also a potential fix to a "restore states" bug, for your info.

deanZZZZZ, 2024-07-16:
[image]
Survived the night without restarting :)

[33 more replies]

closed 2025-02-09 by github-actions[bot]