visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #747: Can I confirm the sounds the display makes?!

chunkysteveo, 2024-07-05:
**Hardware (please complete the following information):**
 - Pump S100102
 - 6 wires
 - PCB V1 with 510 Ohm resistors between lines
 - Level shifter type Blue one

**Software (please complete the following information):**
 - Version - latest

Can I confirm what the sounds the display is supposed to make?

I now have a "Paris 2021" with an S100102 Pump - 6 wire. Before adding the WiFi module, the display would give a loud BEEP on all button presses. Now with the new setup, I get a much quieter set of random "funny/weird" sounds - like chirping, ringing, going-up, going-down etc. They are nowhere near as loud either.

I found this video which also confirmed same on an S100102 pump - https://photos.app.goo.gl/JFBNdfnxYtAXdS4X8 - same quirky noises, and same low volume. (https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/533#issuecomment-1593207517)

If this is the way, then OK. If it's supposed to be just BEEPS and just as loud - then do I have a fault on my board/joints?

I also don't hear any start up melody or tune that others have described.

Thanks - Steve

visualapproach, 2024-07-05:
The sound, even if at low volume, is intended. You can play a melody by sending the appropriate command and filename from spa config page.  Like furelise.mel

visualapproach, 2024-07-05:
When I played with the sound I even got it to play real wav files, but then the volume was so low that it was unusable. And it took too much resources also from the poor esp

visualapproach, 2024-07-05:
If you would want the old sound back - hypothetical of course - you can solder wire #6 between cables instead of to the header on the pcb

chunkysteveo, 2024-07-05:
Ahhh - I thought that is how it's setup, so the ESP manages the buzzer because the CIO pin is n/c.

Should there be a melody on startup by default or is that all custom now with the above command stuff?

Thnaks

visualapproach, 2024-07-05:
No there is no starting melody anymore, sorry ;)

chunkysteveo, 2024-07-05:
> No there is no starting melody anymore, sorry ;)

Cool. That's good to know I've done it all right! I may add a manual internal switch to connect the buzzer back to the CIO... 

Will also order the new V2 PCB so I can correctly add these resistors. 

Thanks again for the awesome project!!!! Upgrade to latest firmware and from 4 wire to new 6 wire was a great experience.

closed 2024-07-05 by chunkysteveo

chunkysteveo, 2024-07-05:
Just played with the melody command, very cool indeed @visualapproach !

So it's my assumption that the volume is low as the 5V from the ESP board PIN isn't enough to drive the buzzer, is that right? In an original state - is the DSP getting a higher voltage from the CIO cable to trigger the buzzer at a higher volume?

visualapproach, 2024-07-05:
If I only knew I'd be happy to do something about it. I know there are electronic gurus here that could fix it if they wanted to...

chunkysteveo, 2024-07-05:
> If I only knew I'd be happy to do something about it. I know there are electronic gurus here that could fix it if they wanted to...

I'm guessing we'd need the ESP pin to drive a transistor that's hooked up to a higher voltage line... not an easy way to do it without some major hacking and extra lines etc.... I think?!

visualapproach, 2024-07-09:
One should measure the voltage on the audio line with a o-scope... I don't have that, nor that pump.