visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #748: Hardware configuration view [BUG]

n3xus1, 2024-07-06:
Hardware configuration view info doesn't show in screen

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Hardware configuration
2. hover over ? for ambient temp sensor, sensor pins, power levels

**Screenshots**
[image]

**Hardware (please complete the following information):**
 - Pump model S100104
 - 6 wires
 - PCB type D1 Mini

**Software (please complete the following information):**
 - Version 4.4.0 beta

n3xus1, 2024-07-07:
look like ambient temperature was fixed but the power levels still looks off
[image]

visualapproach, 2024-07-08:
Should be fixed in latest dev :)

closed 2024-07-08 by n3xus1