visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #756: Air + heating reports wrong power-usage

chsw, 2024-07-12:
**Hardware (please complete the following information):**
 - Pump model: S100101
 - 4 or 6 wires: 6 wires
 - PCB type Wemos D1 mini
 - Level shifter type,

**Software (please complete the following information):**
 - Version 4.4.1

**Your message**
On this pump (at least the one aimed for the Swedish market) only one of the heaters is active when both the heating and air bubbles are active at the same time. But the power-usage reported is both heat 1 and 2.
According to an external power meter, the heating seems to use about 850w when active at the same time as the bubbles.

visualapproach, 2024-07-15:
Ok, thanks for the info. I think I thought about it when adding the power meters, but decided no one would notice 🫤 
Might have a look later. Thx

visualapproach, 2024-07-15:
Ok, try dev branch and see if it is working now.

chsw, 2024-07-17:
I tried the dev branch. It seems to report a more realistic power usage when using both air and bubbles, but it appears as it affected the energy usage reported (it looks like it only report the correct energy consumed when heating is active).

[image]

The first drop in the graph showing consumed energy correlates to installing the new firmware.

Unfortunately I also seem to have lost the ability to connect my module to my wifi. As soon as I connect to the layzspa-hotspot and tries to load the page, it reboots. It doesn't allow me to upload a new firmware either, so I haven't been able to verify if this issue is related to the changes made to the ap manager or if it is an hardware issue related to the fact that I run a d1 mini without level-converters.

visualapproach, 2024-07-17:
If it worked before I don't think it's a hardware problem.

visualapproach, 2024-07-18:
The reboot problem is fixed in todays dev version

visualapproach, 2024-07-18:
Reagarding the energy graph. Can you double check this? And compare total kwh vs daily. Not sure what your graph represents.
Thx

chsw, 2024-07-18:
I managed to get it to connect to my wifi this morning, and upload the latest version from dev.
Unfortunately its now "bricked" and I am unable to start the pump at all!
Probably need to physically remove the module and flash it...

visualapproach, 2024-07-18:
Does it not start the access point?

visualapproach, 2024-07-18:
If you uploaded the file system also, you need to do all the settings again.

chsw, 2024-07-18:
I only  uploaded a new firmware (last 5 commits to dev).
No Ap and no response what so ever on the display.

chsw, 2024-07-18:
I have tried to powercycle, but still nothing. I tried the same firmware on a spare esp, and I seem to work there. 
I cant really access the module inside the pump right now though.

chsw, 2024-07-18:
Ok, so if Ap  manager is disabled and it have trouble connecting, it will not forward any data on the lines?
I tried to press power-lock-timer-power to reset the wifi, but that didn't really do anything. 
Seems like I really should enable AP Manager again :)

chsw, 2024-07-18:
About the issue with energy usage:

[image]

Energy is the energy entity and today the today entity, as shown in Home assistant using mqtt integration.
Today seem to be unaffected, although the 75Kwh usage today is very strange. But since it had trouble this morning connecting, It might have to do with that. 

The (total) energy seem to have two modes. One value which increases when the heater is on (probably the real value). And another value (which also increases over time, but is much smaller) when heater is off.

visualapproach, 2024-07-18:
> Ok, so if Ap manager is disabled and it have trouble connecting, it will not forward any data on the lines? I tried to press power-lock-timer-power to reset the wifi, but that didn't really do anything. Seems like I really should enable AP Manager again :)

I updated the network config page now. It should be more clear what the checkboxes mean.
Basically - do not disable "soft AP".

You should also update to the latest dev firmware, because it will now remember the SSID/pswd even after you upload the file image.

visualapproach, 2024-07-18:
> About the issue with energy usage: [...]

testing

visualapproach, 2024-07-18:
confirmed. WIP

visualapproach, 2024-07-18:
Fixed (dev branch)

closed 2024-09-24 by github-actions[bot]