visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #757: [BUG]

wittyaurelien, 2024-07-12:
**Describe the bug**
When there is a power outage, and the "Restore last states on startup" checkbox is checked, the SPA automatically restarts heating at 40°C, regardless of the temperature previously selected before the power outage.

**To Reproduce**
Only power outage, not on "restart ESP" 

**Hardware:**
 - Pump model : 6 wire, 2021 (air)
 - 6 wires
 - PCB type : V2B

**Software :**
 - Version : 2024-06-17-1334

visualapproach, 2024-07-13:
Thx I'll look at it

visualapproach, 2024-07-15:
@wittyaurelien Maybe its' fixed in development branch now. I found a bug that saved current states before restoring them, which ofc destroyed the previous setting. That bug should be fixed. Let's hope that was it.

wittyaurelien, 2024-07-15:
After several tests of the version present in the development branch, it seems good! Thanks

visualapproach, 2024-07-15:
Thanks for trying it out!

closed 2024-07-22 by wittyaurelien