visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #758: [BUG] Connection Timed Out - Connecting to Lay-Z-Spa AP After Flash

mossc001, 2024-07-14:
**Describe the bug**
After erasing the flash of a v4.3.1 device and uploading both the system and program via VS Code for v4.4.1, it's very slow to connect to the Lay-Z-Spa AP, then once connected via 192.168.4.1, you get connection timed out.

**To Reproduce**
Steps to reproduce the behaviour:
1. General > Full Clean
2. Platform > Erase Flash
3. General > Build (v4.4.1)
4. General > Upload
5. Platform > Build Filesystem Image (v4.4.1)
6. Platform > Upload Filesystem Image
7. Connect to Lay-Z-Spa AP with a device
8. Go to 192.168.4.1; receives "Connection Timed Out"

**Expected behaviour**
You would expect to hit the WiFi configuration manager. 

**Hardware (please complete the following information):**
 - ESP8266 NodeMCU V3 Micro USB

**Additional context**
If I clean/erase and go back to v4.3.1, the speed of connection is rapid and the WiFi Manager connects instantly.

visualapproach, 2024-07-14:
Yes, you are entering the wrong adress.

_An Access Point is created called "Lay-Z-Spa Module". Look for a new Wi-Fi hot spot on your phone. When connecting to that SSID you need to enter the password layzspam0dule. Then open a browser and enter address http://192.168.4.2/wifi.html. Fill in your credentials and save. Wait a few seconds and restart the ESP._

mossc001, 2024-07-14:
@visualapproach , I was just about to reply with that :) I can now see the device has a router of 192.168.4.1 but to access the device, I now need to go to http://192.168.4.2/wifi.html. My mistake!

visualapproach, 2024-07-14:
Easy mistake. A bit funny address maybe but it's more easy to remember than Shelly's ...33 ;)

closed 2024-07-14 by mossc001