visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #760: > Hello ! The spa display dont work anymore since several days..., all is ok from wifi. Previously all works perfectly...On the spa display when I press buttons there is no longueur bip. And if I press a buttons on thé spa display nothing work. Maybe I have accidentaly tick something wrong in the menu ? Thanks by advance !

karls238, 2024-07-15:
Hello ! I have check this week, all seems ok, no corrosion, soldering looks good, I have disconnect and reconnecte red board, but nothing change,
If I remove the board all IS ok from spa display, I have also reser thé board but nothing change, the display buttons are Always inactive...
> Hello ! The spa display dont work anymore since several days..., all is ok from wifi. Previously all works perfectly...On the spa display when I press buttons there is no longueur bip. And if I press a buttons on thé spa display nothing work. Maybe I have accidentaly tick something wrong in the menu ? Thanks by advance !

Hello, few questions. did it ever work? Has it been wet or cable become loose? The soldering looks good still? No corrosion? 
Have you checked your pump config hasn’t become lost and you need to re-enable?
Finally, can you lift and resit the red circuit board (whilst it’s all plugged in) once it’s back in press the RST button and let me know if anything changes

_Originally posted by @metaljay in https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/615#discussioncomment-9915518_

metaljay, 2024-07-15:
Did you solder this yourself? Could you post some photos of it please?
Have you checked the spa buttons are enabled from the wifi config?

visualapproach, 2024-07-15:
Edited the topic to something short. Here is the OP from @karls238 :
> Hello ! The spa display dont work anymore since several days..., all is ok from wifi. Previously all works perfectly...On the spa display when I press buttons there is no longueur bip. And if I [...]

closed 2024-09-21 by github-actions[bot]