visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #767: Wireless AP for setup not present[BUG]

n3xus1, 2024-07-18:
**Describe the bug**
When setting up device for the first time the AP is not enabled to setup (Lay-Z-Spa Module##### does not appear)

**To Reproduce**
erase flash and install latest release or beta

**Expected behavior**
AP should appear Lay-Z-Spa Module#####
If you flash version 4.4.2 the AP appears, if you upgrade from 4.4.2 once Wifi is setup you can access the GUI on you local network.

Hardware (please complete the following information):

    Pump model S100104
    6 wires
    PCB type D1 Mini

**Software (please complete the following information):**
 - Version 4.4.3 or development_v4

visualapproach, 2024-07-19:
Thx. Fixed in 4.4.4

n3xus1, 2024-07-19:
awesome, thanks again for the quick fix
works great now :)

closed 2024-07-19 by n3xus1