visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #770: MQTT password not save upon restart[BUG]

n3xus1, 2024-07-19:
**Describe the bug**
when a restart of ESP is done the mqtt password is lost and upon restart mqtt is not connected. You need to reset the password and save for it to connect.

**To Reproduce**
Steps to reproduce the behavior:
1. set up mqtt
2. restart ESP from drop down menu
3. main page shows mqtt not connected

**Expected behavior**
I would assume that you should be able to restart the ESP and retain settings like user names and passwords

**Hardware (please complete the following information):**
- Pump model S100104
- 6 wires
- PCB type D1 Mini

**Software (please complete the following information):**
 - Version: 2024-07-19-0816

visualapproach, 2024-07-19:
You got to give it a couple of minutes before it reconnects. It remembers between reboots

n3xus1, 2024-07-19:
you are correct, I was just too impatient

closed 2024-07-19 by n3xus1