visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #771: [Help] RCD Breaker tripping

metaljay, 2024-07-20:
Slightly odd request for help, ive made up a 6 wire board, tested as normal on my pump everything works fine. Reset it and flashed latest update then shipped it to someone in the states, after a few emails back and forth he has said its tripping the RCD fuse on the plug.

He is able to plug in via USB and the board works fine including configuring it etc, but as soon as he plugs back into the pump it trips the fuse.

I have asked for photos of the board as I know he has played with the JST cables, but even then, Im not sure what's causing the trip.

my only thought would be the live and ground are touching somehow (hence asking to see the cables) but other than that im at a loss.
Any thoughts before I tell him to bin it and ill send him another?
He has a S100105 pump and 4.4.2 firmware.

visualapproach, 2024-07-21:
A gcf should only trip if current escapes into earth and not returning through neutral. At least I think it's the way it's supposed to work in eu. Sounds unlike that would be triggered by the device unless it touches earth

metaljay, 2024-07-22:
Just came to update to say it now works. The board must of been touching something in the pump. Now it’s isolated it no longer shorts against metal…

closed 2024-07-22 by metaljay