visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #772: Lost network connection[BUG]

n3xus1, 2024-07-21:
**Describe the bug**
After running latest software I lost network access after about 24 hrs
Unable to connect to local wifi AP
I am able to connect to device AP and all settings appear to be saved, when i try and save WiFi SSID and password and restart ESP, device never connects

tomorrow I'm going to pull the device and reload older firmware as I'm unable to upload over WiFi at the moment, i tried connected to the AP and IP 192.168.4.2 but it failed

Hardware (please complete the following information):

    Pump model S100104
    6 wires
    PCB type D1 Mini

Software (please complete the following information):

    Version: 2024-07-19-0816

visualapproach, 2024-07-21:
I don't know if "fast roaming" works with the esp. Try turning it off if you have it set.

n3xus1, 2024-07-21:
I don't believe this is a fast roaming issue as the AP don't have an option and from what I can tell don't support it.
I think what is happening is the AP's are on different channels and the ESP is hanging on to the one AP even it is rebooted.
I did some googling and it appears a lot of people have similar issues with this and fast roaming and some have gotten around it by adding some code to search and select the AP with the same SSID with the highest signal strength then connect.
I looked at the code base but I'm really rusty as i have not coded much in the last 15 years and most of my micro stuff was 20+ years ago.

visualapproach, 2024-07-22:
Ok, I'll keep it in mind. Hope you can work around it for the moment.

n3xus1, 2024-07-22:
Yea it's not really and issue at the moment, I discovered it as an AP was unplugged and it wouldn't connect. Once I plugged it in all was good again.

closed 2024-09-28 by github-actions[bot]