visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #777: [BUG]

acolod, 2024-07-29:
**Describe the bug**
MQTT is showing duplicate entities in HA with slightly different names. "Matching pairs" seem to all be in the same state/have the same value.

**Screenshots**
[image]

**Hardware (please complete the following information):**
 - S100105
 - 6 Wire
 - D1 Mini

**Software (please complete the following information):**
 - Version 4.4.5

**Additional context**
It's been this way at least for the last month - tried upgrading firmware to most current but still an issue (although the names read better now).  It looks like I have two entities which seemingly are linked in some way. If I toggle a switch, for example, both change state.  I tried deleting from HA and re-adding/configuring but still persists with 107 entities. 

It doesn't effect functionality, but curious if anyone has experienced the same/has a fix. Could be my HA configuration but not seeing any reason it would be.

visualapproach, 2024-07-29:
Delete the whole device in HA and restart the ESP.

acolod, 2024-07-29:
Thanks for the quick reply and easy fix. I hadn't restarted last time I deleted it, seems to have done the trick!

closed 2024-07-29 by acolod