visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #779: Can't connect to D1 Mini v3 WiFi after installing firmware & filesystem image

redfusion83, 2024-07-30:
**Hardware (please complete the following information):**
 - S100101
 - 6 wires
 - PCB type: None
 - Level shifter type: None
 - Board: D1 Mini v3

**Software (please complete the following information):**
 - 4.4.5

I followed the instructions, compiled and uploaded the firmware and filesystem image. After rebooting I can see the wifi network I need to connect to. However, all attempts to do so have failed (tried with Android and iPhone). Now, I've had problems with faulty chips in the past, so maybe this is the same problem. However, maybe there is something I can do to diagnose the cause and get it working?

visualapproach, 2024-07-30:
Did you go to http://192.168.4.2?

redfusion83, 2024-07-31:
> Did you go to http://192.168.4.2?

Not possible, since I cannot connect to the access point ("Connection failed").

n3xus1, 2024-08-08:
> > Did you go to http://192.168.4.2?
> 
> 
> 
> Not possible, since I cannot connect to the access point ("Connection failed").

Did you enter the AP password that's in the manual when connecting and make sure it was spelled correctly?
I'd try deleting aka forgetting the AP and reconnecting. Then make sure you are connecting to the AP using only http and not https

redfusion83, 2024-08-10:
Yep, password is correct (otherwise the connection is rejected and there is a message that authentification has failed).
The connections to the AP is very unstable, it connects and disconnects all the time, making it impossible to open the configuration page in the browser. I tried it on different chips (d1 mini v2, d1 mini v3, nodemcu 1.0) and with different phones (Android & iPhone), so it seems unlikely that there is something wrong with the hardware. Still strange that it seems like I'm the only one with this problem...

However, I was able to configure the esp by using a laptop. The connection was still kind of weak, but strong enough to set everything up.

Feno2000, 2024-08-18:
getting the same issue with the d1 mini pro. finds the AP but the password from the manual just doesn't work, tried copying and pasting it from the manual and from the source code and it just doesn't connect. I have the d1 mini and it works fine, although the wifi range on the mini is crap hence going with the pro and external aerial.

n3xus1, 2024-08-20:
I'd try connecting to the D1 using no wires connected or soldered to the board and power it via USB near your AP.
If you have more than one AP make sure to connect to it and use it near the one that it will always use. There is issues with roaming between multiply AP with the same SSID, currently the software doesn't look for the strongest signal then connect it always connects to the same AP and channel

visualapproach, 2024-08-20:
It works on the `d1_mini` board. Cannot try the PRO since I don't have it. Does it work if you change the board to `d1_mini_pro`?

visualapproach, 2024-08-22:
Try the latest dev branch! Turns out that in some cases the ESP tries to reconnect all the time and by doing so it switches wifi channels and since the AP uses the same antenna it will behave badly. So I modified the code to only try to reconnect to station every minute. This made the AP much more stable. It will take up to 2 minutes though to reconnect to station after you've set the credentials.

visualapproach, 2024-08-22:
@Feno2000 If your 'bricked' minis have ch340 usb chip, and you are using windows, the windows drivers doesn't work. Google 'ch340 driver' and download from 'wemos.cc' and install.

Feno2000, 2024-08-22:
i do have that one, the device goes to "Windows has stopped this device because it has reported problems. (Code 43)".
ive bought a usb to ttl bypass and even that doesn't help, cant get tot he board to reflash. i just cant get to a point where the device shows on the com ports. it did originally obviously as i did get it flashed but has since been downhill. i think its best to start with a fresh board and see how i get on as i cant confirm any solid findings now.

closed 2024-10-29 by github-actions[bot]