visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #780: [OTHER]

allanstevens, 2024-07-31:
**Hardware (please complete the following information):**
 - Pump model #54123 Eggshape
 - 6 wires
 - PCB type V2B
 - Level shifter type, not sure what this means.

**Software (please complete the following information):**
 - Version 3.1.2

**Your message**
Hi first up, what a wonderful device.  Thank you.

It worked perfectly at first, but after a day or so, it started slowing down. For example, heating up would result in long, continuous beeps instead of the normal sound effect. A power cycle temporarily fixed the issue, but now it's happening right from startup. Do you have any ideas on how to troubleshoot this? Thank you.

**http://layzspa.local/info/**

Stack size:               236 
Free Dram Heap:           30776 
Min Dram Heap:            28352 
Max free Dram block size: 29456 

Free Iram Heap:           12104 
Max free Iram block size: 12104 

Core version:             3.1.2 
CPU fq:                   160 MHz
Cycle count:              1822539755 
Free cont stack:          1148 
Sketch size:              587296 
Free sketch space:        2555904 

**http://layzspa.local/inputs/**

Edges received on pin D1: 2415
Edges received on pin D2: 20231
Edges received on pin D5: 827
Edges received on pin D6: 1
Edges received on pin D4: 1
Edges received on pin D3: 1
Edges received on pin D7: 1
On 6-w pump the highest number is CLK, next is DATA and third is CS. On 4-wires the highest is CIO or DSP TX to ESP.

allanstevens, 2024-07-31:
It seems to have speeded up again now. 

Not sure if it's related, but I did the following things:

- Deleted bootlog.txt
- Deleted hwtest jpgs
- Turned off debug mode.  Not sure if debug mode was on, item was bought from ebay preconfigured.

Could the above have fixed the issue?  I'll post again if it starts to slow again.

Thanks.

visualapproach, 2024-07-31:
Hi! Thanks for the kind words. The version you referred to is the core version (arduino framework or esp idf, don't remember). If you look at the bottom of your main page it says firmware: and a date/time.
The huge bootlog suggests you have had a lot of reboots. Don't know if the filesize is the direct issue but if you didn't reboot on purpose thousands of times there may be a problem with something.

allanstevens, 2024-08-01:
Thanks for you reply,

Still working great and the bootlog didn't grow overnight 🤞🤞

Firmware: 2024-06-10-2016
Model: PRE2021
RSSI: -76

**Current bootlog**
{"BOOTINFO":"Software/System restart 2024-07-31 12:03:03"}
{"BOOTINFO":"Software/System restart 2024-07-31 12:03:03"}
{"BOOTINFO":"Software/System restart 2024-07-31 12:03:03"}
{"BOOTINFO":"Software/System restart 2024-07-31 12:03:03"}
{"BOOTINFO":"Software/System restart 2024-07-31 12:03:03"}
{"BOOTINFO":"Software/System restart 2024-07-31 12:03:03"}

visualapproach, 2024-08-01:
Oh, I think I know why your bootlog expanded so much. Every time the NTP is reconnected it writes a new line in the file. That's a bug that is fixed in later versions. Along with other improvements. Don't fix things that ain't broken, but you may want to update to latest version.
You can click on 'releases' on github to see what's changed.

allanstevens, 2024-08-01:
Thank you. I'll update it when I get a minute or two.

closed 2024-10-09 by github-actions[bot]