visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #784: [BUG]  Deprecated option used in mqtt publish action call

acolod, 2024-08-05:
**Describe the bug**
Updated HA and integrations, now receiving this warning:

Deprecated option used in mqtt publish action call
This stops working in version 2025.2.0. Please address before upgrading.
Deprecated payload_template option used in MQTT publish action call to topic layzspa/command from payload template {"CMD":15,"VALUE":27, "XTIME":0, "INTERVAL":0}. Use the payload option instead. In automations templates are supported natively. Update the automation or script to use the payload option instead and restart Home Assistant to fix this issue.

Warning - 8/3/2024

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Hardware (please complete the following information):**

- S100105
- 6 Wire
- D1 Mini

**Software (please complete the following information):**
 - V4.5.5

metaljay, 2024-08-08:
Just to add I don’t get this and I use MQTT & HA

visualapproach, 2024-08-08:
I don't see anything about this in the docs. Thanks for reporting.

n3xus1, 2024-08-08:
I also received this warning today after updating to the latest core version of home assistant 2024.8.0

Skeletorjus, 2024-08-08:
Just replace `payload_template:` with `payload:` in your automations/script to remove the warning.

PaleFlyer, 2024-08-17:
> Just replace `payload_template:` with `payload:` in your automations/script to remove the warning.

I tried that for when I push my local Tempest temperature into the ESP, and it had worse alarms.

jamesonuk, 2024-10-09:
Thank you for saving my sanity....
I had put this down to what was being published to MQTT but obviously (and had I read the actual warning) it is what I am sending to MQTT.    It is setting the ambient temperature.

The docs on page 24 just need to be amended from `payload_template` to `payload` but not a bug in the code.

closed 2024-12-16 by github-actions[bot]