visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #788: [FEATURE RQ] - Allow MQTT FQDN

mossc001, 2024-08-20:
**Describe the solution you'd like**
It would be great if the MQTT IP address boxes could be changed for a field to allow a FQDN whether that be pointing internally or externally.

**Describe alternatives you've considered**
No other alternatives for IP or FQDN

visualapproach, 2024-08-20:
I think it can be done relatively easy when I get time. Thx.

mossc001, 2024-08-21:
Hi @visualapproach , thanks. I did try a PoC but couldn't get it to work; probably something obvious:

Input Field:
`				<tr>
					<td>IP Address:</td>
					<td>
						<input type="text" id="mqttIpAddress1" style="width:150px">
					</td>
				</tr>`

Get:
`		{
			var json = JSON.parse(req.responseText);
			console.log(json);
			document.getElementById('enableMqtt').checked = json.enableMqtt;
			document.getElementById('mqttIpAddress1').value = json.mqttIpAddress;
			document.getElementById('mqttPort').value = json.mqttPort;
			document.getElementById('mqttUsername').value = json.mqttUsername;
			document.getElementById('mqttPassword').value = json.mqttPassword;
			document.getElementById('mqttClientId').value = json.mqttClientId;
			document.getElementById('mqttBaseTopic').value = json.mqttBaseTopic;
			document.getElementById('mqttTelemetryInterval').value = json.mqttTelemetryInterval;
		}`

Set:
`	var json = {
		'enableMqtt':(document.getElementById('enableMqtt').checked),
		'mqttIpAddress':(document.getElementById('mqttIpAddress1').value),
		'mqttPort':parseInt(document.getElementById('mqttPort').value),
		'mqttUsername':(document.getElementById('mqttUsername').value),
		'mqttPassword':(document.getElementById('mqttPassword').value),
		'mqttClientId':(document.getElementById('mqttClientId').value),
		'mqttBaseTopic':(document.getElementById('mqttBaseTopic').value),
		'mqttTelemetryInterval':(document.getElementById('mqttTelemetryInterval').value)
	};`

visualapproach, 2024-08-22:
You can test the dev branch now. You need to update both the firmware and the mqtt.html file on the flashdrive.

mossc001, 2024-08-22:
@visualapproach , tested and working but the character limit 'maxlength="31"' is a bit of a restriction. If I remove the limit, re-upload mqtt.html and input a longer FQDN, it crashes the ESP8266. Thoughts?

visualapproach, 2024-08-22:
Ok, yeah I was wondering where to put the limit. How long DN do you need?

mossc001, 2024-08-22:
At the moment I'm testing it with 52 characters.

visualapproach, 2024-08-22:
I tried on the bench here with 63 chars and it did not crash. Do you have extremely long fields like user name and passwords? There is a chance that the total number of chars exceed the reserved space.
Out of curiosity - why do you use a FQDN? Have you set up a DNS in your LAN or do you use a broker on the internet?

mossc001, 2024-08-23:
Hi @visualapproach , thanks for checking!

I've tested with an internal FQDN which I use for ease with an internal DNS on my LAN because it saves me having to update the IPs of 60 MQTT devices if I move VMs for either testing or updates. It seems to work with my internal FQDN.

When I come to use an internet broker, it keeps going into reboot loop for example:
e8a729cb0d024866867f3c473f40af2c.s1.eu.hivemq.cloud:8883
Username/Password are not extremely long.

visualapproach, 2024-08-23:
Did you put the port number in the host string?

mossc001, 2024-08-23:
> Did you put the port number in the host string?

No.

visualapproach, 2024-08-23:
Ok, I tried that address and it didn't crash. Only timed out. I managed to connect to another open broker through FQDN though.

mossc001, 2024-08-23:
I used another open broker to run some tests and found the below from test.mosquitto.org:
1. 1883 : MQTT, unencrypted, unauthenticated - Success
2. 1884 : MQTT, unencrypted, authenticated - Success
3. 8883 : MQTT, encrypted, unauthenticated - Crash on MQTT Save, CONNECTION_TIMEOUT, then reboots every 25-30 seconds
4. 8885 : MQTT, encrypted, authenticated - Crash on MQTT Save, CONNECTION_TIMEOUT, then reboots every 25-30 seconds

mossc001, 2024-08-23:
> Okay, good report. I think we have to live with that. Encryption is too much for the poor 8266.

Yes, agreed, not for ESP8266.

closed 2024-08-24 by mossc001