visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #790: [BUG] d1_mini_pro errors on directly connected pins.

mwitkow, 2024-08-23:
**Describe the bug**
I'm trying to get the software running on a d1_mini_pro (as it has a detachable WiFi). I tried it now on two boards to see if it is not a hardware failure.  I actually even made a custom PCB for it:
[image]

But now I'm troubleshooting why it doesn't work, and I'm finding that there are `hwtest` errors even on a direct connection between the pins of the board

```                                                                                                                                                                                                                                                                                                                                                                                                                                                                Start test. Seq begins with HIGH, then alters.

Sending on D4, receiving on D7
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D3, receiving on D6
-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0 // 50 errors out of 100
Sending on D2, receiving on D5
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D7, receiving on D4
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D6, receiving on D3
-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0 // 50 errors out of 100
Sending on D5, receiving on D2
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
End of test!
"1" or "0" indicates ERROR, depending on test state. "-" is good.
Switching cio pins 5s HIGH -> 5s LOW -> input
then DSP pins 5s HIGH -> 5s LOW -> input, repeating
Disconnect cables then reset chip when done!
```

**To Reproduce**
Steps to reproduce the behavior:
1. Get a d1_mini_pro 
2. Change `board = d1_mini` in `platformio.ini` (otherwise i get `device reports readiness to read but returned no data`) Build. Upload. Build filesystem. Flash. 
3. Change HW config to:

[image]

4. Connect pins directly using jumper cables.
5. [image]

6. Run ``hw_test` see errors.
7. See error

**Expected behavior**
hw test passing since the pins are connected directly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - Pump model, for example 54123
 - 4 or 6 wires
 - PCB type
 - Level shifter type,

**Software (please complete the following information):**
 - Version: latest `master` checkout.

mwitkow, 2024-08-23:
For context, I tried also setting `board = d1_mini_pro`, but filesystem reliably fails on:
```
el_gzip_data([".pio/build/nodemcuv2/littlefs.bin"], ["datazip"])
Clearing zipped files
Looking for upload port...
Auto-detected: /dev/cu.usbserial-023TNOBQ
Uploading .pio/build/nodemcuv2/littlefs.bin
esptool.py v3.0
Serial port /dev/cu.usbserial-023TNOBQ
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:e7:29:68:15:09
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 4292845568 bytes to 4243433...
Writing at 0x00200000... (0 %)
Writing at 0x00204000... (0 %)
Writing at 0x00208000... (1 %)

A fatal error occurred: Failed to write compressed data to flash after seq 2 (result was C400)
*** [uploadfs] Error 2
```

mwitkow, 2024-08-23:
Interestingly, the same firmware build and the same cables work on a `d1_mini` (non pro)

```                                                                                                                                                                                                                                                                                                                                                                                                                                                                          Start test. Seq begins with HIGH, then alters.

Sending on D4, receiving on D7
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D3, receiving on D6
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D2, receiving on D5
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D7, receiving on D4
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D6, receiving on D3
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D5, receiving on D2
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
End of test!
"1" or "0" indicates ERROR, depending on test state. "-" is good.
Switching cio pins 5s HIGH -> 5s LOW -> input
then DSP pins 5s HIGH -> 5s LOW -> input, repeating
Disconnect cables then reset chip when done!
```

I searched the internet to see whether there is anything about D1 Mini vs D1 Mini Pro differences (e.g. when one of the ports is treated specially, but no dice.

visualapproach, 2024-08-23:
Maybe one of the pins is kaputt

mwitkow, 2024-08-24:
These are with 2 different D1 Mini Pro board, exactly the same pair `D3, D6` fails. Same pins and cables on the D1 Mini (non pro)

intelx3, 2024-09-08:
Hey.

Did you found solution how to upload filesystem in to flash when `board = d1_mini_pro` selected ?
I'm facing same problem :( 

`A fatal error occurred: Failed to write compressed data to flash after seq 2 (result was C400)`

My lay-z-spa works with d1 mini but wifi range is terrible
So i'm also looking to use d1 mini pro - but with no luck

Pozdrawiam Dawid :)

intelx3, 2024-09-08:
Ok. I found solution how to flash correctly wemos d1 mini pro
Just add `board_build.ldscript = eagle.flash.8m7m.ld` in `platform.ini`
Without this compiler make a flash file `littlefs.bin` with size 4GB ! :) (there is bug in size calculation code).

But after that programm still not working :)

Hwtest pass ok

Start test. Seq begins with HIGH, then alters.

Sending on D4, receiving on D7
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D3, receiving on D6
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D2, receiving on D5
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D7, receiving on D4
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D6, receiving on D3
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
Sending on D5, receiving on D2
---------------------------------------------------------------------------------------------------- // 0 errors out of 100
End of test!
"1" or "0" indicates ERROR, depending on test state. "-" is good.
Switching cio pins 5s HIGH -> 5s LOW -> input
then DSP pins 5s HIGH -> 5s LOW -> input, repeating
Disconnect cables then reset chip when done!

but there is nothing on the panel screen - no lights and no start beep.

i attached file witch /inputs
[input log.zip](https://github.com/user-attachments/files/16923103/input.log.zip)

Any ideas ?

visualapproach, 2024-10-18:
@intelx3 thanks for posting your board solution.
Idea - use a level converter!

closed 2024-12-26 by github-actions[bot]