visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #796: Question about calibration feature

Neuroquila-n8fall, 2024-09-06:
First of all: what a great project! I thought about getting the wifi pump unit but this is more like it :) 

I've got a general question about how this feature works. I already looked through all information but couldn't quite understand under which conditions the calibration learns and what its main purpose is.

So for starters my spa is indoors and the room temperature is actively reported to the module. The room temperature is hovering around 23°C +/- 0.5°C right now.
The water temperature (setpoint) is 35°C.
Usually the temperature is dropping by 1°C every 04:30 and this is where the heater kicks in. It then runs for around 30 minutes and reaches the target temperature again.

Now from what I've learned is that in order for the calibration to work, the pump has to run for at least 5 minutes, the heater must not be turned on and the water temperature must not be equal to the ambient temperature.

1. what is the "ready time" saying when the temperature is at setpoint but displaying it's going to be "ready" in 5 hours?
2. What is the "ideal" condition for calibration? In my case it will take several hours for a change in water temperature, does that even matter? Should I just let the pump run with a target temperature with say 30° to make sure that the heater won't kick in in between?

Thanks in advance for clarification ;)

visualapproach, 2024-09-07:
Hi and thx!

First, if you only want to run the heater at particular times you can do it like you do, by lowering the target temperature. Or you just switch off the heater and let the pump run and the water will cool off to whatever temperature it ends up being when you turn on the heater again.

Regarding the calibration, the ESP clocks how long it takes the water to cool at a specific temp difference between the water and ambient air. To get a good estimate you need to have some difference to start with. Preferably more than 5 degrees. You just have to keep the pump going without the heater until the tickbox is checked in spa config page. Or you can time it yourself and enter the values manually.

If the esp is calibrated it can track the real water temperature for quite a long time without the pump going. The display will easily show 5 degrees less than the actual water temp in the tub if it is cold outside. Or inside, wherever the tub is located.
This estimated "real" temperature is called virtual temperature. The virtual temperature will be reset to the measured temperature* after the pump is run for 5 minutes. At this time the temp in the tub is considered equal to the temp in the hose where it is measured.

*to within a degree

Time to ready is based on virtual temperature vs ambient and target temperature. So even if measured temp is the same as target, it will show the time from virtual temp to get to target.

Sounds complicated maybe, but ask again if there is something unclear

Neuroquila-n8fall, 2024-09-10:
...thanks for the explanation!

So what I observed is that the calibration "ticked" when the temperature did not drop. That surprised me since how should it calculate the values then? What I observe is the following:
[image]
"Poolstate" means: 1 = cycling, 4 = heating
The ambient temperature was at 21°C straight the whole time.

Clearly it can't make any sense out of the learned values. Did I just hit a bug, maybe?

visualapproach, 2024-09-10:
The ambient temp doesn't need to change. It actually better if it's a flat line. It measures the time for the water temperature to drop a degree. Which means if you start at 38 the clock starts when it flips to 37 and stops when it drops to 36.

Neuroquila-n8fall, 2024-09-10:
Now I get the principle behind it... 
So I just reset everything, turned on the pump and disabled the ambient temperature synchro so it stays constant. Now I only need to wait for like 12 hours for the result hehe...

visualapproach, 2024-09-10:
Yup 👍🏼

Neuroquila-n8fall, 2024-09-11:
Alright. It's now 1° off but that's hardly an issue. The virtual temperature curve is off by half an hour of heat-up time, so to speak. If that bugs me hard enough I will look into the code and try a different method... until then thanks for your explanations :)

closed 2024-09-11 by Neuroquila-n8fall