visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #799: [FEATURE RQ]

Gifford47, 2024-09-15:
**Is your feature request related to a problem? Please describe.**
It's only a feature request.
It would be nice to show the current time in the display. It could be alternate between time and temperature.

**Describe the solution you'd like**
Show the current time without three digits. We can separate the hours from the minutes with a dot. The time and temperature shown in the display could be changed every X seconds.

**Describe alternatives you've considered**
A separate clock :)

**Additional context**

visualapproach, 2024-09-15:
Hi and Thanks for your FR 🙂 
There is no dots available unfortunately. Unless you paint it on the display. In theory you could schedule a 'print text' command at 1:00 with the text "1 00" and have it repeat every 12/24  hours. Do it again for 02:00 and so on. Then do all of it again for every minute 01, 02 and so on until the device crashes due to out of memory. 😂
If you have a smart hub you can schedule text messages, like showing the time.
But I take it you want the ESP to do it. I hear you.

Gifford47, 2024-09-16:
Do you mean, that we cannot address the dots in the display or that some displays might not have extra dots in the display (f.e. the older egg pumps)? because the newer ones have [displays](https://static.wixstatic.com/media/48b729_d92385da73dc42f885e9fba63e568c3b~mv2.jpg/v1/fill/w_480,h_480,al_c,q_80,usm_0.66_1.00_0.01,enc_auto/48b729_d92385da73dc42f885e9fba63e568c3b~mv2.jpg) with dots.
Is the esp already nearly 100% busy with the current system tasks?

visualapproach, 2024-09-16:
You are right, the dots exists in the display, but they are not addressable. Probably not connected.
Well, the ESP is for the most part not super busy. The main loop runs at a few kHz. But some activities requires a lot of memory so with that in mind I'll say it is on the limit. Your FR would probably be doable if made in a efficient way. I mean it can already show text so it's not that much of a deal as long as you don't choke it with long strings or print too often.

visualapproach, 2024-09-16:
In latest dev branch you can now do this:
[image]
or this, but keep in mind: the time it takes to display the text should NOT be longer than the repeat interval, or you will run out of memory pretty soon.
[image]

Gifford47, 2024-09-16:
very nice idea, thanks for the fast implementation! looking forward to see it in master, but for now I´m testing 😄

closed 2024-11-24 by github-actions[bot]