visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #800: OTA update to Version 4.4.5 made my heater unusable.

codennifty, 2024-09-27:
**Hardware (please complete the following information):**
 - Pump model, S100101
 - 6 wires
 - d1_mini

**Software (please complete the following information):**
 - Current Version 4.3

I am trying to upgrade from 4.3 to 4.4.5 however when I did the ota update I couldn't change the spa type in the web page and the buttons did not respond and were disabled.

visualapproach, 2024-09-27:
You don't write very much to work with, but did you try updating via usb? Why can't you change hardware settings?

codennifty, 2024-09-28:
I'll try to clarify. So basically I tried to update my firmware over wifi following the guide.  I was able to successfully upload the latest version but after I did that I lost physical control over the heater. In the settings it reset my hardware configuration settings but when I tried to update them they did not save. In the spa config all the buttons were disabled and I couldn't reenable them. The only thing that worked was re uploading the software I previously was one. Thanks for you help.

codennifty, 2024-10-10:
I was able to fix by re flashing via usb.

closed 2024-10-10 by codennifty