visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #803: [FEATURE RQ]

Gifford47, 2024-10-06:
**Is your feature request related to a problem? Please describe.**
Feature request:
Toggle brightness after pressing Power button

**Describe the solution you'd like**
Like original firmware, the power button toggles the brightness of the display between 0 and the last user setting.
It would be also great to have a new switch entity in Homeassistent for the power button.

**Describe alternatives you've considered**

**Additional context**

visualapproach, 2024-10-06:
Function to turn power off/on from command queue in config.html is added in dev branch. Is it to dim the power LED or what's the reason for this rq?
No autodiscovered button is added to HA yet. You can add it manually until further ny sending cmd 24 over mqtt if you know how to do that.

Gifford47, 2024-10-06:
Yes, it's to dim the display. Would be useful for me to dim the display manually.
It is possible to add the mqtt discovery in the next release?

visualapproach, 2024-10-06:
Ok, yes I think so

closed 2024-10-13 by visualapproach