visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #809: [BUG]MQTT to Home Assistant no longer working on 4.4.6

PaleFlyer, 2024-10-31:
**Describe the bug**
Updated to 4.4.6, and noticed HA was no longer getting/sending data to the controller

**To Reproduce**
Update controller, and reboot

**Expected behavior**
Communication to exist

**Screenshots**
[image]
[image]
I've tried a variety of configs on the MQTT page, this is the latest. I have verified my password is correct using another MQTT device.

**Hardware (please complete the following information):**
[image]
using the "red" level shifter

**Software (please complete the following information):**
 - Version 4.4.6

**Additional context**
Add any other context about the problem here.

PaleFlyer, 2024-10-31:
Mosquitto isn't even seeing it, unless it somehow is coming in on an IP I don't have in my network.

Mosquitto is showing a LOT of hits from a 172 address. My 2 VLANS are 192 and 10. And the ESP is on the 192 network.

patatman, 2024-11-01:
Just to verify, I'm running the latest version 4.4.6 and I don't have any issues.
MQTT is working fine. So this seems to be a local problem.
Are you still having issues? Maybe try power-cycling the entire unit?
[image]

PaleFlyer, 2024-11-01:
> Just to verify, I'm running the latest version 4.4.6 and I don't have any issues. MQTT is working fine. So this seems to be a local problem. Are you still having issues? Maybe try power-cycling the entire unit? <img alt="image" width="254" src="https://private-user-images.githubusercontent.com/15031001/382263836-e2a9bd22-ab9c-400c-a5b8-e79f65d7d15d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA0NjcxOTQsIm5iZiI6MTczMDQ2Njg5NCwicGF0aCI6Ii8xNTAzMTAwMS8zODIyNjM4MzYtZTJhOWJkMjItYWI5Yy00MDBjLWE1YjgtZTc5ZjY1ZDdkMTVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAxVDEzMTQ1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdjMmRkOTAzZDI1NzcxZDNjNjNkYjkzMTNiN2RkM2M1YjI2MDU2MWY0N2M2ZWIxZjZhY2ZmZjlkNzBjMGNmZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JzvRd3Zx6A8ZMkvXoSb2bsX2YmgS3Vo2tQbAvy5UkIo">

I did try soft-rebooting the device, but I have not tried pulling the power.

visualapproach, 2024-11-01:
You need to update the file(system) also:
[image]

PaleFlyer, 2024-11-01:
Clearly VSCode, PlatformIO, and/or the ESP hates me. I can't push the FileSystem OTA... So I'm gonna have to crack the case open again... So I guess I need to find the waterproof USB I had gotten for this kind of thing, and install that finally.
[image]

visualapproach, 2024-11-02:
Don't use "upload fs OTA". Use "upload fs"

PaleFlyer, 2024-11-02:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/809/2452970036@
> github.com>

Tried that to, said it couldn't grab the serial port for the network upload

PaleFlyer, 2024-11-06:
For some reason the upload non-OTA worked this time...

visualapproach, 2024-11-06:
Nice

PaleFlyer, 2024-11-06:
Getting a "Connection_Unauthorized" now, but I have a feeling that is due to my server, as it's been showing some signs of issues for a while now, and just rebooted HomeAssistant as I can't open the logs as the API call is failing in HomeAssistant....

PaleFlyer, 2024-11-06:
Okay, for no apparent reason, the user I have on all my other MQTT devices just doesn't like the hot tub, but... I got it working.

Is there any way to WRITE the time to the timers?

IE with blowing out the file system, my counters on the filter, and the chlorine are all saying it's been over 2000 days...

The Chlorine and the filter rinse I need to do today normally, but the wash/replace is not a "today" item...

visualapproach, 2024-11-10:
> Is there any way to WRITE the time to the timers?

No, but the next update you can just upload individual files instead of the whole file system. By choosing the UPLOAD FILE in the UI menu, and upload them one at a time. A bit of a hassle, but the saved data will be retained. Or you can download all the saved settings and put it in the data folder before uploading the FS.

Moris101010, 2024-11-15:
I also have a problem with MQTT. I can enter a password consisting of 32 characters, but in HA I have a much longer one. How can I enter a long password?
There was no problem on the old version.

visualapproach, 2024-11-15:
> I also have a problem with MQTT. I can enter a password consisting of 32 characters, but in HA I have a much longer one. How can I enter a long password?
> 
> There was no problem on the old version.

You can edit line 69 in data/mqtt.html and upload the file to the esp from the UI menu. There is a max total message length though so it will probably crash if all your fields are excessively long. But go ahead and try a larger maxlength value!

closed 2025-01-22 by github-actions[bot]