visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #813: [BUG] Unresponsive erratic behaviour

john270, 2024-12-15:
Hi,
I've had a PCBv2 device running successfully in my pre2001 6 wire hot tub for a while, however at the moment the web portal is only responsive for a few minutes after powering on and then - although remaining connected - the web buttons cease to be responsive and / or the pump itself will randomly turn on/off, change target temp etc.
Any thoughts on this?
I generally control it via HA over MQTT but have tested it without any MQTT and the issue persists.
I've looked at the 'inputs' page and this is pasted below.  Every time I refresh the page the content is different, however I'm unsure what it represents?
Software is up-to-date.

Thanks 

```
wait<br>micros, gpio registers
866013544,E0035035
866013733,E0037035
866020153,E0033035
866022081,E0037035
866035129,E0033035
866035182,E0033015
866035187,E0033005
866035207,E0033015
866035228,E0033035
866035232,E0033025
866035252,E0033035
866035285,E0033005
866035306,E0033015
866035330,E0033005
866035351,E0033015
866035375,E0033005
866035404,E0033015
866035428,E0033005
866035449,E0033015
866035473,E0033025
866035506,E0033035
866035526,E0033015
866035531,E0033005
866035551,E0033015
866035584,E0033035
866035608,E0033025
866035629,E0033035
866035653,E0033025
866035674,E0033035
866035706,E0033025
866035727,E0033035
866035751,E0033025
866035772,E0033035
866035804,E0033025
866035825,E0033035
866035849,E0033025
866035886,E0033035
866035919,E0033025
866035939,E0033035
866035964,E0033025
866035984,E0033035
866036061,E0033025
866036086,E0033035
866036118,E0033025
866036139,E0033035
866036163,E0033025
866036184,E0033035
866036204,E0033025
866036237,E0033035
866036261,E0033025
866036282,E0033035
866036306,E0033025
866036335,E0033035
866036359,E0033025
866036380,E0033035
866036404,E0033025
866036433,E0033035
866036482,E0037035
866050153,E0033035
[... 942 lines cut ...]
866245799,E0033025
866245820,E0033035
866245844,E0033025
866245881,E0033035
866245913,E0033025
866245934,E0033035
866245958,E0033025
866245979,E0033035
866246060,E0033025
866246080,E0033035
866246113,E0033025
866246134,E0033035
866246158,E0033025
866246179,E0033035
866246203,E0033025
866246232,E0033035
866246256,E0033025
866246277,E0033035
866246301,E0033025
866246330,E0033035
866246354,E0033025
866246375,E0033035
866246399,E0033025
Cut and paste all above. Zip and post on forum for help.
```

visualapproach, 2024-12-15:
Hi,

It's just a log of all pin states. Whenever a pin changes state there will be a timestamp in microseconds, and the pin register state.
So
@ 866013544 us the pin register is E0035035 which in binary is 1110 0000 0000 0011 01**0**1 0000 0011 0101
@ 866013733 us the pin register is E0037035 which in binary is 1110 0000 0000 0011 01**1**1 0000 0011 0101
we can see that the bold bit has flipped.
It is just a way to see that the relevant pins is active as they are supposed to. If for instance the clock signal were shorted to ground, there would be no activity (bit flips) on that bit.
Now, what bit is what GPIO... yeah, you need to check which GPIO number corresponds to which D#. When you know the GPIO # you count from right to left - ...bit4, bit3, bit2, bit1, bit0 where bit 0 = GPIO 0.

Not particularly user friendly

john270, 2024-12-16:
So you don't think those logs are related to the erratic and unresponsive behaviour?
I'm really confused as if do the hwtest it works without any errors, but in the pump (possibly coincidentally when the temperature gets sub 10°C) it generally goes wacky and I have to leave it unplugged

visualapproach, 2024-12-16:
Sounds like corrosion. Cold temperatures shrinkage on solder joints. Try resoldering

john270, 2024-12-19:
Thanks - I think you are right about the corrosion.  I took the board out today and there was notable rust on the pins of the centre component (less or none on the ESP which would explain why the web portal was working fine).
I'll close the issue now