visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #814: [BUG]

MattSpeke, 2025-01-17:
**Describe the bug**
Trying to upload the development fireware following the same instructions as the manual. Error when uploading code to platformio when press arrow to upload code into the program.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
 Executing task: C:\Users\mspeke\.platformio\penv\Scripts\platformio.exe run --target upload 

Processing nodemcuv2 (platform: espressif8266@^4; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.2.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 49 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.21.2
|-- WebSockets @ 2.6.1
|-- PubSubClient @ 2.8.0
|-- ESPAsyncTCP @ 1.2.2
|-- EspSoftwareSerial @ 8.2.0
|-- DallasTemperature @ 4.0.1
|-- smarthomes
|-- BWC_unified
|-- ArduinoOTA @ 1.0
|-- ESP8266WebServer @ 1.0
|-- ESP8266WiFi @ 1.0
|-- LittleFS @ 0.1.0
|-- OneWire @ 2.3.8
|-- Ticker @ 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Archiving .pio\build\nodemcuv2\lib381\libSPI.a
Archiving .pio\build\nodemcuv2\lib2dc\libEthernet.a
Compiling .pio\build\nodemcuv2\liba15\WebSockets\WebSocketsClient.cpp.o
Compiling .pio\build\nodemcuv2\liba15\WebSockets\WebSocketsServer.cpp.o
Compiling .pio\build\nodemcuv2\liba15\WebSockets\libb64\cdecode.c.o
Compiling .pio\build\nodemcuv2\liba15\WebSockets\libb64\cencode.c.o
Compiling .pio\build\nodemcuv2\liba15\WebSockets\libsha1\libsha1.c.o
Compiling .pio\build\nodemcuv2\libf00\PubSubClient\PubSubClient.cpp.o
Compiling .pio\build\nodemcuv2\libaac\ghostl\ghostl\FastScheduler.cpp.o
Compiling .pio\build\nodemcuv2\lib2d8\EspSoftwareSerial\SoftwareSerial.cpp.o
Compiling .pio\build\nodemcuv2\lib04a\OneWire\OneWire.cpp.o
Compiling .pio\build\nodemcuv2\lib9a0\DallasTemperature\DallasTemperature.cpp.o
Compiling .pio\build\nodemcuv2\libb25\cio\CIO_4W.cpp.o
In file included from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Strings/StringAdapters.hpp:7,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Memory/MemoryPool.hpp:10,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Variant/JsonVariantConst.hpp:10,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:9,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Array/JsonArray.hpp:7,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson.hpp:29,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson.h:9,
                 from lib\BWC_unified/bwc.h:12,
                 from src\main.h:28,
                 from src\main.cpp:1:
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Polyfills/type_traits.hpp:7:10: fatal error: type_traits/conditional.hpp: No such file or directory
    7 | #include "type_traits/conditional.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\nodemcuv2\src\main.cpp.o] Error 1
.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
  523 |     return (rc == expectedLength);
      |             ~~~^~~~~~~~~~~~~~~~~
=================================================================== [FAILED] Took 5.38 seconds ===================================================================

 *  The terminal process "C:\Users\mspeke\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

**Hardware (please complete the following information):**
 - Pump model, for example 54123
 - 6 wire 
 - PCB type v2b
 - Level shifter type, red

**Software (please complete the following information):**
 - Version

**Additional context**
Add any other context about the problem here.

visualapproach, 2025-01-18:
Works here after full clean. Are you on a mac?

MattSpeke, 2025-01-19:
I am using  a windows 11 laptop.

Tried a full clean by pressing full clean on the menu. Now get this error:

[image]

Tried to open the folder again after the software was restarted and get

[image]

I tried again from the start and got this when trying to upload the code folder

[image]

I am not a coder. I am just trying to follow your instructions from the manual on how to upload the firmware to the ESP

MattSpeke, 2025-01-19:
With the original firmware I can upload by pressing the arrow, then Build filesystem image, then Upload filesystem image. There are no errors.

visualapproach, 2025-01-19:
Okay, is it the same on master and development branch? (Select branch on main github page)

MattSpeke, 2025-01-19:
> <visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/814/2601014320@
> github.com>

Master branch is fine. It’s the development branch I get the errors on.

visualapproach, 2025-01-20:
Okay I'll see if I can figure it out.

visualapproach, 2025-01-21:
I downloaded the project zip file, extracted all and the opened the folder ".../code" from vscode. Works ok for me.

closed 2025-03-30 by github-actions[bot]